esphome/issues #643: fatal error: esphomelib/application.h: No such file or directory

souptonuts, 2019-09-02:
**Operating environment/Installation:**

Hass.io Add-on: esphomeyaml-edge vdev
Development Version! Manage and program ESP8266/ESP32 microcontrollers through YAML configuration files
armhf / null / HA 0.97.2 / SU 183 / stable
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 NodeMCU
**Affected component:**

**Description of problem:**
My first time using ESPHomeyaml-edge add on.  I have ESP8266 board connected with USB to RP3.  When I try to upload to device using /dev/ttyUSB0CP2102 USB to UART bridge the compile fails with the following error:

Compiling .pioenvs/garage/src/main.cpp.o
src/main.cpp:3:36: fatal error: esphomelib/application.h: No such file or directory
#include "esphomelib/application.h"

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphomeyaml:
  name: garage
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'my_router'
  password: 'top_secret'

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphomeyaml/garage
Processing garage (platform: espressif8266@1.8.0; framework: arduino; board: nodemcuv2)
--------------------------------------------------------------------------------
LibraryManager: Installing id=305 @ 1.1.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing esphomelib
git version 2.17.1
Cloning into '/config/esphomeyaml/garage/.piolibdeps/_tmp_installing-P8IHYK-package'...
LibraryManager: Installing id=3837 @ 5.13.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.13.0-dev #254d059
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESPAsyncTCP> 1.1.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/garage/src/main.cpp.o
Generating LD script .pioenvs/garage/ld/eagle.app.v6.common.ld
Compiling .pioenvs/garage/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pioenvs/garage/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
src/main.cpp:3:36: fatal error: esphomelib/application.h: No such file or directory
#include "esphomelib/application.h"
^
compilation terminated.
*** [.pioenvs/garage/src/main.cpp.o] Error 1
========================== [ERROR] Took 15.11 seconds ==========================
```

**Additional information and things you've tried:**

Using the Homeassistant configurator web tool, I looked at the Hassio filesystem and do see application.h located here:
/config/esphomeyaml/garage/.piolibdeps/esphome-core/src/esphome/application.h

brandond, 2019-09-02:
eaphomelib and esphomeyaml were merged into just esphome ages ago. Have you tried updating?

souptonuts, 2019-09-02:
The problem was an out-of-date add on in Homeassistant.  I was unaware that the add-on link had changed.  I uninstalled the esphomeyaml add-on and migrated to the new esphome repository.  No longer seeing error on the build.

closed 2019-09-02 by souptonuts