esphome/issues #644: Python console script - TypeError: __init__() got an unexpected keyword argument 'required'

Hsxsky, 2019-09-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

centos 
Python 2.7.5
voluptuous-0.11.7

**ESP (ESP32/ESP8266, Board/Sonoff):**

Not relevant in this case.

**Affected component:**

Python console script
https://esphome.io/guides/getting_started_command_line.html

**Description of problem:**

Error in the script.
TypeError: __init__() got an unexpected keyword argument 'required'

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

Not relevant in this case.

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

[a@centhome /]# esphome
Traceback (most recent call last):
  File "/usr/bin/esphome", line 6, in <module>
    from esphome.__main__ import main
  File "/usr/lib/python2.7/site-packages/esphome/__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "/usr/lib/python2.7/site-packages/esphome/writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "/usr/lib/python2.7/site-packages/esphome/config.py", line 15, in <module>
    from esphome import core, core_config, yaml_util
  File "/usr/lib/python2.7/site-packages/esphome/core_config.py", line 6, in <module>
    import esphome.config_validation as cv
  File "/usr/lib/python2.7/site-packages/esphome/config_validation.py", line 1242, in <module>
    Optional(CONF_COMMAND_TOPIC): All(requires_component('mqtt'), subscribe_topic),
  File "/usr/lib/python2.7/site-packages/esphome/voluptuous_schema.py", line 198, in extend
    ret = super(_Schema, self).extend(schema, extra=extra)
  File "/usr/lib/python2.7/site-packages/voluptuous/schema_builder.py", line 782, in extend
    return result_cls(result, required=result_required, extra=result_

```

**Additional information and things you've tried:**

voluptuous-0.11.5 version is ok.

kluszczyn, 2019-09-02:
I have faced the same issue when installing esphome latest or beta version (```pip install -U esphome``` or  ```pip install --pre -U esphome```. Upgrading to latest beta via ```pip install https://github.com/esphome/esphome/archive/dev.zip``` removed the error.

closed 2019-09-03 by Hsxsky

lord-carlos, 2019-09-04:
I had to do this as well.