esphome/issues #645: Ethernet (LAN8720) interface starts, connects (static only) but doesn't work

nicolasbisi, 2019-09-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hassio 0.98.1
The same behaviour is happening on release and dev branch

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 nodemcu
LAN8720 Ethernet adaptor

**Affected component:**
https://esphome.io/components/ethernet.html

**Description of problem:**
If static IP is not set, connection keeps failing in a loop to connect.
"Connecting via ethernet failed! Re-connecting..."

With static IP set, the same message appears about 600 times and then it connects (the problem was addressed and supposedly fixed here https://github.com/esphome/issues/issues/53)
but even saying connected, I still can't ping/find on the dashboard (green light)/ add on hassio manually or even access with the web_server
**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32_test
  platform: ESP32
  board: nodemcu-32s

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode:  GPIO16_OUT
  phy_addr: 1
  
  manual_ip:
      static_ip: 192.168.0.209
      gateway: 192.168.0.1
      subnet: 255.255.255.0
    
# Enable logging
logger:
  level: VERBOSE
  
# Enable Home Assistant API
api:

ota:

web_server:
  port: 80

```

**Logs (if applicable):**

```
[20:55:14][I][logger:156]: Log initialized
[20:55:14][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[20:55:14][I][app:028]: Running through setup()...
[20:55:14][V][app:029]: Sorting components by setup priority...
[20:55:14][C][ethernet:025]: Setting up Ethernet...
[20:55:29][W][ethernet:041]: Connecting via ethernet failed! Re-connecting...
[20:55:29][W][ethernet:041]: Connecting via ethernet failed! Re-connecting...
[20:55:29][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 20 - ETH_START
[20:55:29][V][ethernet:107]: [Ethernet event] ETH started (num=20)
[20:55:29][W][ethernet:041]: Connecting via ethernet failed! Re-connecting...
Repeat 600+ times Connecting via ethernet failed! Re-connecting...
[20:55:33][W][ethernet:041]: Connecting via ethernet failed! Re-connecting...[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 24 - ETH_GOT_IP
[20:55:33][D][WiFiGeneric.cpp:421] _eventCallback(): ETH IP: 192.168.0.209, MASK: 255.255.255.0, GW: 192.168.0.1
[20:55:33]:[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 24 - ETH_GOT_IP
[20:55:33][D][WiFiGeneric.cpp:421] _eventCallback(): ETH IP: 192.168.0.209, MASK: 255.255.255.0, GW: 192.168.0.1
[20:55:33][V][ethernet:107]: [Ethernet event] ETH Got IP (num=24)
[20:55:33]
[20:55:33][I][ethernet:052]: Connected via Ethernet!
[20:55:33][C][ethernet:220]:   IP Address: 192.168.0.209
[20:55:33][C][ethernet:221]:   Hostname: 'esp32_test'
[20:55:33][C][ethernet:222]:   Subnet: 255.255.255.0
[20:55:33][C][ethernet:223]:   Gateway: 192.168.0.1
[20:55:33][C][ethernet:226]:   DNS1: 0.0.0.0
[20:55:33][C][ethernet:228]:   DNS2: 0.0.0.0
[20:55:33][C][ethernet:231]:   MAC Address: REDACTED
[20:55:33][C][ethernet:232]:   Is Full Duplex: YES
[20:55:33][C][ethernet:233]:   Link Up: YES
[20:55:33][C][ethernet:234]:   Link Speed: 100
[20:55:33][C][web_server:061]: Setting up web server...
[20:55:33][C][ota:029]: Over-The-Air Updates:
[20:55:33][C][ota:030]:   Address: 192.168.0.209:3232
[20:55:33][C][api:022]: Setting up Home Assistant API server...
[20:55:33][I][app:058]: setup() finished successfully!
[20:55:34][I][app:101]: esphome version 1.14.0-dev compiled on Sep  2 2019, 20:44:15
[20:55:34][C][logger:175]: Logger:
[20:55:34][C][logger:176]:   Level: VERBOSE
[20:55:34][C][logger:177]:   Log Baud Rate: 115200
[20:55:34][C][logger:178]:   Hardware UART: UART0
[20:55:34][C][ethernet:066]: Ethernet:
[20:55:34][C][ethernet:220]:   IP Address: 192.168.0.209
[20:55:34][C][ethernet:221]:   Hostname: 'esp32_test'
[20:55:34][C][ethernet:222]:   Subnet: 255.255.255.0
[20:55:34][C][ethernet:223]:   Gateway: 192.168.0.1
[20:55:34][C][ethernet:226]:   DNS1: 0.0.0.0
[20:55:34][C][ethernet:228]:   DNS2: 0.0.0.0
[20:55:34][C][ethernet:231]:   MAC Address: REDACTED
[20:55:34][C][ethernet:232]:   Is Full Duplex: YES
[20:55:34][C][ethernet:233]:   Link Up: YES
[20:55:34][C][ethernet:234]:   Link Speed: 100
[20:55:34][C][ethernet:069]:   MDC Pin: 23
[20:55:34][C][ethernet:070]:   MDIO Pin: 18
[20:55:34][C][ethernet:071]:   Type: LAN8720
[20:55:34][C][web_server:117]: Web Server:
[20:55:34][C][web_server:118]:   Address: 192.168.0.209:80
[20:55:34][C][ota:029]: Over-The-Air Updates:
[20:55:34][C][ota:030]:   Address: 192.168.0.209:3232
[20:55:34][C][api:095]: API Server:
[20:55:34][C][api:096]:   Address: 192.168.0.209:6053
[20:57:14][I][ota:046]: Boot seems successful, resetting boot loop counter.
```

**Additional information and things you've tried:**

redskinhu, 2019-09-03:
same issue

nicolasbisi, 2019-09-04:
Hum, good to know that I'm not alone and the problem is reproducible.
Just to give more information from what I tested so far:

I have tried changing phy_addr. Anything else than 1 returns the same error:
```[16:11:49][W][ethernet:041]: Connecting via ethernet failed! Re-connecting...
[16:11:50][V][:000]: E (15999) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0xffff
[16:11:50]
[16:11:51][V][:000]: E (17000) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0xffff
[16:11:51]
[16:11:51][V][:000]: E (17001) emac: Initialise PHY device Timeout
[16:11:51]
[16:11:51][E][ethernet:125]: ETH enable error: -1
[16:11:51][E][component:092]: Component was marked as failed.
```
 clockmode "works" with either GPIO17_OUT or GPIO16_OUT. GPIO0_IN/OUT fails

I have tried to flash using a raspberry pi3+ connecting directly on the nodeMCU board USB and also using a TTL board with 5v. Both gave me the same result.

nicolasbisi, 2019-09-07:
I found [this](https://translate.google.com/translate?hl=en&sl=auto&tl=en&u=http%3A%2F%2Fblog3.narimatsu.net%2Farticle%2F180309a). I'm not sure if it's helpfull at all. But every place I look, it seems that they had to do hardware modification.

minsuke, 2019-09-13:
I want to know how to connect hardware pinmap with esp32, I bought LAN8720 module, and I wanto to connect with wired connection. can you share your connection diagram ?

nicolasbisi, 2019-09-13:
Hey @minsuke, I have 4 wires connected to the LAN8720 module.
\+ 3.3
ground
esp32 GPIO023 connected to mdc_pin
esp32 GPIO18 connected to mdio_pin

The code I used was this:
```
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode:  GPIO16_OUT
  phy_addr: 1
```

nicolasbisi, 2019-10-11:
Nobody else has any ideas?

minsuke, 2019-11-13:
> Nobody else has any ideas?

Hello nicolasbisi ? 

I tried but I have no connection 

but I'm very happy that I can try connect with module not a special board (POE System) 

I wish this module can be supported someday. thx again

nicolasbisi, 2020-01-11:
Hi @OttoWinter 
Since there's no solution for this still, shouldn't we remove it from the documentation to prevent other people from buying the ethernet board?
Cheers

mig77angel, 2020-04-11:
Is there no progress on this?

closed 2020-12-05 by stale[bot]