esphome/issues #647: "getting started" guide fails on step 2 (Windows 10 )

timdonovanuk, 2019-09-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Python 2.7 on Windows 10

**Description of problem:**

Following the getting started guide.

1. Step 1 (pip install esphome): works
2. Step 2 (esphome livingroom.yaml wizard): fails

**Logs (if applicable):**

```
C:\Users\user>esphome livingroom.yaml wizard
Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Python27\Scripts\esphome.exe\__main__.py", line 5, in <module>
  File "c:\python27\lib\site-packages\esphome\__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "c:\python27\lib\site-packages\esphome\writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "c:\python27\lib\site-packages\esphome\config.py", line 15, in <module>
    from esphome import core, core_config, yaml_util
  File "c:\python27\lib\site-packages\esphome\core_config.py", line 6, in <module>
    import esphome.config_validation as cv
  File "c:\python27\lib\site-packages\esphome\config_validation.py", line 1242, in <module>
    Optional(CONF_COMMAND_TOPIC): All(requires_component('mqtt'), subscribe_topic),
  File "c:\python27\lib\site-packages\esphome\voluptuous_schema.py", line 198, in extend
    ret = super(_Schema, self).extend(schema, extra=extra)
  File "c:\python27\lib\site-packages\voluptuous\schema_builder.py", line 782, in extend
    return result_cls(result, required=result_required, extra=result_extra)
TypeError: __init__() got an unexpected keyword argument 'required'

```

**Additional information and things you've tried:**

timdonovanuk, 2019-09-02:
Fixed by doing `pip install voluptuous==0.11.5.`

therafman, 2019-09-03:
> Fixed by doing `pip install voluptuous==0.11.5.`

Had the same issue. Thank you!

OttoWinter, 2019-09-04:
Duplicate of https://github.com/esphome/issues/issues/621

closed 2019-09-04 by OttoWinter