esphome/issues #648: [web server] no status shown for sensors

froido, 2019-09-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Hi,

there is a problem with showing the states on the web server. There are no states shown in the web frontend, even though they are updating in the serial console. I tried this with a couple of sensors and binaries.

Config: Adafruit ESP32 Feather as standalone webserver

### YAML-File:
```
esphome:
  name: pump
  platform: ESP32
  board: featheresp32

web_server:
  port: 80

wifi:
  ap:
    ssid: "esp32-pump"
    password: "esp32-pump!"
  reboot_timeout: 0s
  
# Enable logging
logger:

sensor:
  - platform: esp32_hall
    name: "ESP32 Hall Sensor"
    update_interval: 2s
    id: bla

```

### Serial-Log:
[image]

### Web-Frontend:
[image]

OttoWinter, 2019-09-04:
The problem is that the JS files fail to load because you have no internet connection, see the js/css_url options in the web_server config to point it to a link hosted on a machine in your own network

closed 2019-09-04 by OttoWinter

brandond, 2019-09-04:
I know using the device in AP mode is intended to be temporary, but for folks that use it that way exclusively it might be nice to be able to have the option to serve the CSS and JS off the device?

froido, 2019-09-05:
I agree with Brandnon. Is there a way to include the css/js files in the code to use esp wifi/internet independently? Should I open this as a pull request?

brandond, 2019-09-05:
If you come up with a patch to do that, then yes absolutely put in a PR.