esphome/issues #650: Icon's not displaying correctly in hass.io add-on after updating to HA 0.98.3

shbatm, 2019-09-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Ubuntu -- Rock Pi 4b

**ESP (ESP32/ESP8266, Board/Sonoff):**

Several, not applicable to this issue.
**Affected component:**

Hass.io Add-on Page

**Description of problem:**
Material Design Icons are not displaying in the add-on page after the latest update to home assistant. Tried clearing the cache and tried in both Chrome and Edge, same thing.
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
N/A

```

**Logs (if applicable):**

```
N/A

```

**Additional information and things you've tried:**

randybb, 2019-09-10:
Yea, it is missing `MaterialIcons-Regular.woff` / `MaterialIcons-Regular.woff2` in /usr/lib/python3.7/site-packages/esphome/dashboard/static/fonts.
You can fix it temporary by downloading `https://github.com/google/material-design-icons/raw/master/iconfont/MaterialIcons-Regular.woff` and `https://github.com/google/material-design-icons/raw/master/iconfont/MaterialIcons-Regular.woff2` to mentioned directory, just change prefix /usr/ based on your system (could be in your home, in case you have installed it via `pip install --user ...`, or in any other based on your virtual environment).

Anonym-tsk, 2019-09-15:
Already fixed in https://github.com/esphome/esphome/pull/703

shbatm, 2019-09-22:
That brings me to another question... if this is already pushed to the `dev` channel, why are the font files still not showing up when I rebuild my hass.io dev container?

I watch the logs run through downloading the latest `dev.zip` from GH and running setup, but the changes still aren't there.

<details>
  <summary>Log from Hass.io Container (on dev branch)</summary>
  
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing...
-----------------------------------------------------------

 Hass.io Add-on: ESPHome (dev)
 Development Version! Manage and program ESP8266/ESP32 microcontrollers through YAML configuration files
-----------------------------------------------------------
 Add-on version: dev
 You are running the latest version of this add-on.
 System: Ubuntu 18.04.3 LTS  (aarch64 / qemuarm-64)
 Home Assistant version: 0.99.2
 Supervisor version: 188
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing...
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing...
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing...
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing...
[01:03:14] INFO: Installing esphome version 'dev' (https://github.com/esphome/esphome/archive/dev.zip)...
Collecting https://github.com/esphome/esphome/archive/dev.zip
  Downloading https://github.com/esphome/esphome/archive/dev.zip
Requirement already up-to-date: PyYAML<6,>=5.1 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: colorlog>=4.0.2 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: esptool<3,>=2.6 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: ifaddr<1,>=0.1.6 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: paho-mqtt<2,>=1.4 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Collecting platformio>=3.6.5 (from esphome==1.14.0.dev0)
  Downloading https://files.pythonhosted.org/packages/5d/59/2ba54e3f2ae8435997acbd0d0ff31993a7041a8b11515e0b5cc846a71ca4/platformio-4.0.3.tar.gz (140kB)
Requirement already up-to-date: protobuf<3.8,>=3.7 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: pyserial<4,>=3.4 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: pytz>=2019.1 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: tornado<6,>=5.1.1 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: tzlocal>=1.5.1 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: voluptuous<0.12,>=0.11.5 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: typing>=3.6.6 in /usr/local/lib/python2.7/dist-packages (from esphome==1.14.0.dev0)
Requirement already up-to-date: ecdsa in /usr/local/lib/python2.7/dist-packages (from esptool<3,>=2.6->esphome==1.14.0.dev0)
Requirement already up-to-date: pyaes in /usr/local/lib/python2.7/dist-packages (from esptool<3,>=2.6->esphome==1.14.0.dev0)
Requirement already up-to-date: ipaddress in /usr/local/lib/python2.7/dist-packages (from ifaddr<1,>=0.1.6->esphome==1.14.0.dev0)
Collecting bottle<0.13 (from platformio>=3.6.5->esphome==1.14.0.dev0)
  Downloading https://files.pythonhosted.org/packages/c4/a5/6bf41779860e9b526772e1b3b31a65a22bd97535572988d16028c5ab617d/bottle-0.12.17.tar.gz (72kB)
Collecting click<8,>=5 (from platformio>=3.6.5->esphome==1.14.0.dev0)
  Downloading https://files.pythonhosted.org/packages/fa/37/45185cb5abbc30d7257104c434fe0b07e5a195a6847506c074527aa599ec/Click-7.0-py2.py3-none-any.whl (81kB)
Requirement already up-to-date: colorama in /usr/local/lib/python2.7/dist-packages (from platformio>=3.6.5->esphome==1.14.0.dev0)
Requirement already up-to-date: requests<3,>=2.4.0 in /usr/local/lib/python2.7/dist-packages (from platformio>=3.6.5->esphome==1.14.0.dev0)
Collecting semantic_version<3,>=2.8.1 (from platformio>=3.6.5->esphome==1.14.0.dev0)
  Downloading https://files.pythonhosted.org/packages/0f/3b/8fee26649a86c71df159ed0ae7ac5f9ac38829bccd8a7404e116f903929b/semantic_version-2.8.2-py2.py3-none-any.whl
Collecting tabulate<1,>=0.8.3 (from platformio>=3.6.5->esphome==1.14.0.dev0)
  Downloading https://files.pythonhosted.org/packages/c2/fd/202954b3f0eb896c53b7b6f07390851b1fd2ca84aa95880d7ae4f434c4ac/tabulate-0.8.3.tar.gz (46kB)
Collecting setuptools (from protobuf<3.8,>=3.7->esphome==1.14.0.dev0)
  Downloading https://files.pythonhosted.org/packages/b2/86/095d2f7829badc207c893dd4ac767e871f6cd547145df797ea26baea4e2e/setuptools-41.2.0-py2.py3-none-any.whl (576kB)
[... cut ...]
Successfully installed bottle-0.12.17 certifi-2019.9.11 click-7.0 esphome-1.14.0.dev0 platformio-4.0.3 semantic-version-2.8.2 setuptools-41.2.0 tabulate-0.8.3 urllib3-1.25.5
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing...
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[01:03:55] INFO: Waiting for dashboard to come up...
[01:03:55] INFO: Starting ESPHome dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[01:03:56] INFO: Starting NGINX...
```
</details>

brandond, 2019-09-22:
They probably need to be added to the manifest.

shbatm, 2019-09-22:
I was digging all over for where those includes were--didn't even think to check `MANIFEST.in`, duh.  

Sent a proposed PR for adding `*.woff*` (for both `.woff`/`.woff2` and `LICENSE`) to `MANIFEST.in`.

closed 2019-09-23 by glmnet

jamesrshaffer, 2019-09-24:
This seems to have been resolved. I installed HA 0.99.2, uninstalled, then reinstalled esphome:dev and the problem went away!