esphome/issues #652: CVE-2019-12586 CVE-2019-12587 CVE-2019-12588 bugs

Peter4825, 2019-09-05:
Hi,

Several severe security bugs have been found in the ESP32/ESP8266 chipset/libraries, which are also present in the arduino SDK.  As far as I am aware, ESPHome uses the arduino SDK and is therefore also vurneable.  I'm opening this bug to make sure that ESPHome is aware of these flaws, and that this hopefully will result in a new release that contains clear language in the release notes so we know that our ESPHome firmware are exploit free again.

For more information about these bugs:
https://matheus-garbelini.github.io/home/post/esp32-esp8266-eap-crash/
https://matheus-garbelini.github.io/home/post/zero-pmk-installation/
https://matheus-garbelini.github.io/home/post/esp8266-beacon-frame-crash/

or find a summary at https://hackaday.com/2019/09/05/esp8266-and-esp32-wifi-hacked/

**Operating environment/Installation (Hass.io/Docker/pip/etc.): All

**ESP (ESP32/ESP8266, Board/Sonoff): All

**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
Not relevant...
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-09-06:
AFAIK there is not yet an Arduino release that resolves all of these issues. Once that is available through PlatformIO it can be used here.

OttoWinter, 2019-09-06:
Yeah I read through the report when it first appeared on hacker news.

But I would consider it low(ish) severity because:

 - First two CVEs require a connection to a WPA2-Enterprise network, which is not supported by ESPHome atm (so no problem)
 - Third is a denial of service attack type, which is of course a concern, but considering a (similar) DoS is always possible by spamming the WiFi channel with deauth packets. 

As @brandond said, this will probably take some time to get through the dependency chain (ESP8266: NONOS-SDK -> Arduino -> Platformio -> ESPHome, ESP32: ESP-IDF -> Arduino -> Platformio -> ESPHome).

But then there's another issue: The latest major versions of the base-sdk/arduino layer are in some cases quite buggy (especially with ESP8266), and I'm not sure if the folks there will backport those changes to older major versions as patch releases.

I guess this is just another reason for forking the Arduino/Platformio layers so that we are in control of those release cycles as well and backport fixes (like this really annoying one https://github.com/esphome/esphome/blob/dev/esphome/core/esphal.cpp#L249) ourselves 🙄

OttoWinter, 2019-11-05:
The default ESP SDK versions have been updated in 1.14.0

closed 2019-11-05 by OttoWinter