esphome/issues #653: Errors with component SIM800L

tiagofreire-pt, 2019-09-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32-WROOM-32

**Affected component:**
SIM800L

**Description of problem:**
Errors are generated using this component: https://github.com/esphome/esphome/pull/522

Also, there is no documentation available in http://esphome-io.

**Problem-relevant YAML-configuration entries:**
```yaml
uart:
  tx_pin: GPIO27
  rx_pin: GPIO14
  baud_rate: 9600
  software_buffer_size: 512

sim800l:
  id: sms
  on_sms_received:
    lambda: |-
      ESP_LOGD("main", "Received sms from %s: %s", sender.c_str(), message.c_str());

binary_sensor:
  platform: gpio
  name: "Send sms"
  pin: GPIO4
  on_press:
    lambda: |-
      id(sms).send_sms("+351916277170", "Hello there from ESPHome");

```

**Logs (if applicable):**

```
INFO Reading configuration...
INFO Detected timezone 'WET' with UTC offset 0 and daylight savings time from 03/29/20 01:00:00 to 10/25/20 02:00:00
Failed config

uart: 
  - [source config/miflora.yaml:104]
    tx_pin: GPIO27
    rx_pin: GPIO14
    baud_rate: 9600
    
    [software_buffer_size] is an invalid option for [0]. Please check the indentation.
    software_buffer_size: 512 [source config/miflora.yaml:107]
sim800l: [source config/miflora.yaml:110]
  
  Component not found: sim800l.
  id: sms
  on_sms_received: 
    lambda: ESP_LOGD("main", "Received sms from %s: %s", sender.c_str(), message.c_str());

```

glmnet, 2019-09-06:
This component is in DEV branch only (not released) but I'm happy to help you getting it to work.
If you are on Hassio you should be using the dev branch I guess, the `Component not found: sim800l` seems to indicate you are using a version older than v1.14

The docs are prerelease too, they can be found here: https://next.esphome.io/components/sim800l.html

tiagofreire-pt, 2019-09-17:
Could you provide the how to install esphome-DEV on Manjaro, please? I'm on my PC for extra compiling speed and performance.

glmnet, 2019-09-17:
I don't know manjaro, check if this helps https://esphome.io/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version

closed 2019-10-21 by OttoWinter