esphome/issues #654: RF Switch from Home Assistant

AquaMCU, 2019-09-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I am using ESPhome 1.13.6. I want to replace my current Home Assistant setup with a ESPhome stand alone solution.

**ESP (ESP32/ESP8266, Board/Sonoff):**

D1_mini with a RF Transmitter connected to D0.

**Affected component:**

remote_transmitter

**Description of problem:**
I try to migrate the following Home Assistant configuration into ESP home. I can see, that the module transmits (Using a frequency scanner on 433MHz) but the switch does not open

HOME ASSISTANT CONFIG:
      Gartentor:
        pulselength: 370
        code_on: 21
        code_off: 20
        signal_repetitions: 10
        protocol: 1

**Problem-relevant YAML-configuration entries:**
```yaml
remote_transmitter:
  pin: D0
  carrier_duty_percent: 100%

switch:
#  - platform: gpio
#    name: "TM"
#    pin: D0
#    id: TM

  - platform: template
    name: door
    id: door
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
#          code: '00010100000101000001010000010100'
#          protocol: 
#            pulse_length: 396
#            sync: [1,31]
#            zero: [1,3]
#            one: [3,1]
#            inverted: false
#          repeat:
#            times: 10
#            wait_time: 50ms

           code: '00010100'
           protocol: 1
           repeat: 
             times: 10
             wait_time: 50ms

```

**Logs (if applicable):**

```
[D][switch:029]: 'door' Toggling ON.
[D][remote_transmitter:066]: Sending remote code...
```

**Additional information and things you've tried:**
I tried to change and tweak all parameters including the commented out ones. Can you please help me with the configuration?

manju-rn, 2019-09-10:
Have you tried to use the Remote Receiver component (in a seperate setup) and test the signal generated?  Unless you do that, there is no gurantee that your setup is working or even connected properly.

AquaMCU, 2019-09-10:
I have tried that, but somehow I did not receive any codes. I think the sender and receiver have an intereference when they are on the same board. I ordered a second 8266 and I will try this today....

manju-rn, 2019-09-10:
Do you have any other RF transmitter (from toys or otherwise) to test if just the receiving function is working fine?

glmnet, 2019-09-10:
> I have tried that, but somehow I did not receive any codes. I think the sender and receiver have an intereference when they are on the same board. I ordered a second 8266 and I will try this today....

I not 100% sure with RF but I guess it is the same as with IR (Infra red), when the ESP is sending the code the CPU can only do that and thus ignores anything it receives from the sensor, so that will never work sending and receiving on the same board.

AquaMCU, 2019-09-10:
Hey. Thanks for all the reply. I have bought a second D1 mini and there the receiver works. So I found the following code and it works ;)

turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          protocol: 1 
          code: '000000000000000000010101'
          repeat:
            times: 20
            wait_time: 500ms

BTW: ESPhome rocks! I have changed all my systems to it. I have installed the dashboard on my RPI, so I can develop without installing the tools on my notebook. 

Perfect work to the team!

closed 2019-09-10 by AquaMCU