esphome/issues #656: IR transmitter repeat wait_time stopeed working

sisaenkov, 2019-09-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, esphome 1.13.6.

**ESP (ESP32/ESP8266, Board/Sonoff):**

D1_mini with a IR Transmitter connected to D4.

**Affected component:**

remote_transmitter

**Description of problem:**
After upgrading from 1.12.x to 1.13.x stopped working **wait_time** in repeat section.

**Problem-relevant YAML-configuration entries:**
```yaml
remote_transmitter:
  pin: 4
  carrier_duty_percent: 50%

switch:
  - platform: template
    id: bedroomtvkodi
    name: Bedroom TV Kodi
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E0807F
        repeat:
          times: 3
          wait_time: 1s
```

**Logs (if applicable):**

```
[14:47:08][D][switch:021]: 'Bedroom TV Kodi' Turning ON.
[14:47:08][D][remote_transmitter:066]: Sending remote code...
```

**Additional information and things you've tried:**
It looks like this parameter is ignored and all IR-commands are sent without any delay.

brandond, 2019-09-08:
IIRC it was using the wrong time unit, but that has been fixed. Have you tried the dev branch?

sisaenkov, 2019-09-08:
> IIRC it was using the wrong time unit, but that has been fixed. Have you tried the dev branch?

@brandond thank you. Works as intended with dev branch.

closed 2019-09-08 by sisaenkov