esphome/issues #657: esp8266_pwm ignored

alexyao2015, 2019-09-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 [Feit Bulb](https://blakadder.github.io/templates/feit_rgb_a19.html)
**Affected component:**

esp8266_pwm

**Description of problem:**
esp8266_pwm seems like it is being ignored and instead only a binary output is being used. Turning on and off has a transition but there is no transition and acts just like a switch instead of light. Brightness shows up in HA but is ignored (only maximum brightness turns on the light).

Cleared build and tried a different system. Tasmota PWM works fine so it does not seem to be a hardware issue.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: light2
  platform: ESP8266
  board: esp01_1m

ota:
  password: "redacted"
    
wifi:
  ssid: "redacted"
  password: "redacted"
  
  manual_ip:
    static_ip: 192.168.1.30
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  
  ap:
   ssid: "ESPhome light2"
  
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "redacted"

output:
  - platform: esp8266_pwm
    pin: GPIO5
    id: cold_white_gpio
    frequency: 1000 Hz
    inverted: False
    min_power: 0
    max_power: 1
    
  - platform: esp8266_pwm
    pin: GPIO12
    id: warm_white_gpio
    frequency: 1000 Hz
    inverted: False
    min_power: 0
    max_power: 1
    
light:
  - platform: cwww
    id: white_led
    name: "white_led"
    cold_white: cold_white_gpio
    warm_white: warm_white_gpio
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2000 K
    gamma_correct: 0.8
    default_transition_length: 0.4s

web_server:
  port: 80

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[18:38:34][D][light:261]: 'white_led' Setting:
[18:38:34][D][light:270]:   State: ON
[18:38:34][D][light:300]:   Transition Length: 0.4s
[18:38:37][D][light:261]: 'white_led' Setting:
[18:38:37][D][light:274]:   Brightness: 67%
[18:38:37][D][light:300]:   Transition Length: 0.4s
[18:38:37][D][light:261]: 'white_led' Setting:
[18:38:37][D][light:274]:   Brightness: 100%
[18:38:37][D][light:300]:   Transition Length: 0.4s
[18:38:39][D][light:261]: 'white_led' Setting:
[18:38:39][D][light:270]:   State: OFF
[18:38:39][D][light:300]:   Transition Length: 0.4s
```

**Additional information and things you've tried:**
Tried changing the light to a monochromatic light and still had the same issue. Logs clearly show the brightness changing but visibly there is no change. 

This exact same config previously worked. It suddenly stopped working today. Strange because there was no recent release or patch. Same results across 2 different bulbs/same model, and same with firmware compiled on different machine.

glmnet, 2019-09-09:
> It suddenly stopped working today. 

Can you expand a little here? Did you upload any change so it stopped working? 

Which version is it running?

alexyao2015, 2019-09-09:
All I did was reupload the code again. Made no changes to it at all. I'm running v1.13.6.

It's really strange because I thought I had changed something in the code so I found old backups of the code that previously worked and those also failed to work. I then tried compiling on a different machine altogether and still had the same issues.

alexyao2015, 2019-09-15:
I tried another bulb with the same code, and its working fine. Is it possible that something hardware wise broke? It doesn't make much sense since the pwm is software based?

alexyao2015, 2019-09-15:
Just powercycled the other bulb after reflashing and its also working now? I have no explanation for why its working but it is now.

OttoWinter, 2019-10-18:
Yeah this is due to the arduino framework and interactions with the ESP8266 internals.

Tasmota uses a different PWM mechanism than ESPHome (in both cases, PWM uses some low level stuff like non-maskable interrupts). 

Some of the PWM stuff doesn't get cleared when doing "just" a software reset. A power cycle (hard reset) does fix PWM though.

Will add a note about it in the docs.

closed 2019-10-18 by OttoWinter