esphome/issues #661: remote_transmitter component with NodeMCU (ESP8266) not transmitting correct signals

svippe, 2019-09-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome 1.13.6
Home Assistant
etc.
in Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

NODEMCU ESP8266
**Affected component:**

remote_transmitter
switch
GPIO
IR diode
**Description of problem:**
Trying to setup an ESPHome node to act as a remote to my Sony TV but the ESPHome transmitter function will not send the correct signal as seen in the log, tried both `remote_transmitter.transmit_raw` and `remote_transmitter.transmit_sony` with the corresponding codes.

Have tried the hardware on another build from mdhiggins called ESP8266-HTTP -IR-Blaster compiled using the Arduino IDE with success. There I could both send and receive the right codes, it also decoded the raw code correctly in to Sony remote codes. So there are no hardware issues but seems to be related to how the `remote_transmitter.transmit` function works.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: node1
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "xxxxx"
  password: "xxxxx"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: xxxxx
    # Set this to the IP address of the router. Often ends with .1
    gateway: xxxxx
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: xxxxx
    
web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxxxxxxxx"

ota:
  password: "xxxxxxxxxx"

#remote_receiver:
#  pin:
#    number: D5
#    inverted: true
#    mode: INPUT_PULLUP
#  dump: raw

remote_transmitter:
  pin: D2
  carrier_duty_percent: 50%
  
switch:
  - platform: template
    name: "Sony Mute Button"
    id: sony_mute_button
    turn_on_action:
      - remote_transmitter.transmit_raw:
         code: [2336, -666, 1136, -666, 535, -666, 1136, -666, 536, -666, 1136, -666, 536, -666, 535, -666, 1136, -670, 532, -667, 534, -666, 536, -666, 535]
         carrier_frequency: 38khz
         repeat: 3
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[13:25:59][D][remote.raw:041]: Received Raw: 2293, -706, 1096, -706, 495, -706, 1099, -703, 495, -708, 1094, -707, 495, -706, 495, -707, 1095, -707, 494, -707, 496, -705, 498, -704, 494
[13:25:59][D][remote.raw:041]: Received Raw: 2291, -706, 1096, -706, 495, -706, 1096, -706, 495, -707, 1095, -708, 494, -708, 493, -707, 1096, -706, 495, -707, 494, -706, 495, -707, 496
[13:26:09][D][switch:029]: 'Sony Mute Button' Toggling ON.
[13:26:09][D][remote_transmitter:066]: Sending remote code...
[13:26:09][D][remote.raw:041]: Received Raw: 238
[13:26:09][D][remote.raw:041]: Received Raw: 235
[13:26:09][D][remote.raw:041]: Received Raw: 236
[13:26:11][D][switch:029]: 'Sony Mute Button' Toggling ON.
[13:26:11][D][remote_transmitter:066]: Sending remote code...
[13:26:11][D][remote.raw:041]: Received Raw: 234
[13:26:11][D][remote.raw:041]: Received Raw: 235
[13:26:11][D][remote.raw:041]: Received Raw: 238
[13:26:12][D][switch:029]: 'Sony Mute Button' Toggling ON.
[13:26:12][D][remote_transmitter:066]: Sending remote code...
[13:26:12][D][remote.raw:041]: Received Raw: 235
[13:26:12][D][remote.raw:041]: Received Raw: 235
[13:26:12][D][remote.raw:041]: Received Raw: 237
```

**Additional information and things you've tried:**
Tried on other ESP8266 IR libraries to see if there could be an issue with HW or how I've wired it but that works so I have pinpointed the issue possibly to `remote_transmitter.transmit`.

To exclude the issue being with HomeAssistant I've also setup the Web server and toggled the send command from there with the same output as in the log above.

Output from ESP8266-HTTP-IR-Blaster. Transmitted from the IR diode to the IR photoresistor on the same test board (1cm between components):

[image]

glmnet, 2019-09-10:
I appreciate this report.
There is a known bug https://github.com/esphome/issues/issues/485 related to repeat times on IR transmitter.
Can you perform another test without repeat?

svippe, 2019-09-10:
Thanks for the quick reply!

I've tried without repeat and attached the output. Still the same short raw response...

```
[14:00:06][D][remote.raw:041]: Received Raw: 2367, -637, 1165, -636, 566, -637, 1189, -612, 565, -637, 1166, -637, 564, -637, 589, -613, 1166, -636, 565, -637, 565, -636, 565, -637, 565
[14:00:06][D][remote.raw:041]: Received Raw: 2367, -637, 1166, -637, 564, -637, 1165, -638, 564, -636, 1166, -638, 564, -614, 587, -637, 1189, -614, 564, -637, 589, -612, 565, -637, 589
[14:00:12][D][api:573]: Client 'Home Assistant 0.97.2 (192.168.1.9)' connected successfully!
[14:00:13][D][switch:029]: 'Sony Mute Button' Toggling ON.
[14:00:13][D][remote_transmitter:066]: Sending remote code...
[14:00:13][D][remote.raw:041]: Received Raw: 201
[14:00:17][D][switch:029]: 'Sony Mute Button' Toggling ON.
[14:00:17][D][remote_transmitter:066]: Sending remote code...
[14:00:17][D][remote.raw:041]: Received Raw: 203
[14:00:20][D][switch:029]: 'Sony Mute Button' Toggling ON.
[14:00:20][D][remote_transmitter:066]: Sending remote code...
[14:00:20][D][remote.raw:041]: Received Raw: 199
```

glmnet, 2019-09-10:
Ahh wait, are you testing transmitting and receiving on the same node? That won't work

svippe, 2019-09-10:
But it works to run both receiver and transmitter in the other build (ESP8266-HTTP -IR-Blaster)? 

I will setup another node separately as a receiver to see if it works.

glmnet, 2019-09-10:
I see, I don't know about that library, but that for sure will not work with ESPHome.
Another thing you can try is to add a -10000 to the end of the raw code, that will effectively add a "wait time" in between code (re)transmissions in case you need the repeat: 3

svippe, 2019-09-10:
Ok, so I've tried setting up another NodeMCU and it can read the raw code from the transmitter which is similar to the received raw code from the TV remote. The problem is it doesnt work with the TV which it did with other library.

Transmitter node to the right and receiver node to the left. The raw output is both from transmitter and TV remote and doesnt seem to be that different.
[image]

What else could be wrong?

glmnet, 2019-09-10:
Looks like repeat 3 is still enabled ?
Please double confirm you try (to the tv) with no repeat and also the last -10000 trick

svippe, 2019-09-10:
No, repeat is not on, it's just me spamming the toggle button but I have also tried activating the code/toggling at much lower speed.

I've retried the code, ensured it's without repeat and with and without the -10000 trick.

OttoWinter, 2019-10-22:
Closing as @svippe found a solution to the problem - as @glmnet noted repeat was broken in the last version but that has been fixed in the latest beta.

closed 2019-10-22 by OttoWinter