esphome/issues #666: TypeError: __init__() got an unexpected keyword argument 'required'

LandonStewart, 2019-09-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

macOS using esphome in a virtualenv with python3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

Cannot run esphome at all.

**Description of problem:**
Just trying to execute esphome in general.  Fails with `TypeError: __init__() got an unexpected keyword argument 'required'`

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: cam
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "YYYYYYYYYY"
  password: "XXXXXXXXX"

esp32_camera:
  name: ESP32CAM1
  external_clock:
    pin: GPIO27
    frequency: 20MHz
  i2c_pins:
    sda: GPIO25
    scl: GPIO23
  data_pins: [GPIO17, GPIO35, GPIO34, GPIO5, GPIO39, GPIO18, GPIO36, GPIO19]
  vsync_pin: GPIO22
  href_pin: GPIO26
  pixel_clock_pin: GPIO21
  reset_pin: GPIO15
  resolution: 640x480
  jpeg_quality: 10

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "XXXXXXXXX"

ota:
  password: "XXXXXXXXX"
```

**Logs (if applicable):**

```
landons-imac:esphome lstewart$ virtualenv -p /usr/local/bin/python3 venv
Running virtualenv with interpreter /usr/local/bin/python3
Using base prefix '/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7'
New python executable in /Users/lstewart/esphome/venv/bin/python3.7
Also creating executable in /Users/lstewart/esphome/venv/bin/python
Installing setuptools, pip, wheel...
done.
Activating virtualenv "esphome"...
(esphome)landons-imac:esphome lstewart$ pip install esphome
Collecting esphome
  Using cached https://files.pythonhosted.org/packages/df/bd/a96abf8fb626b21793c42d5750a1503520d33a11fc4947408845fa7a4a24/esphome-1.13.6-py2.py3-none-any.whl
Collecting ifaddr<1,>=0.1.6 (from esphome)
Collecting esptool<3,>=2.6 (from esphome)
Collecting pyserial<4,>=3.4 (from esphome)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting colorlog>=4.0.2 (from esphome)
  Using cached https://files.pythonhosted.org/packages/68/4d/892728b0c14547224f0ac40884e722a3d00cb54e7a146aea0b3186806c9e/colorlog-4.0.2-py2.py3-none-any.whl
Collecting tornado<6,>=5.1.1 (from esphome)
Collecting voluptuous<0.12,>=0.11.5 (from esphome)
Collecting PyYAML<6,>=5.1 (from esphome)
Collecting tzlocal>=1.5.1 (from esphome)
  Using cached https://files.pythonhosted.org/packages/ef/99/53bd1ac9349262f59c1c421d8fcc2559ae8a5eeffed9202684756b648d33/tzlocal-2.0.0-py2.py3-none-any.whl
Collecting protobuf<3.8,>=3.7 (from esphome)
  Using cached https://files.pythonhosted.org/packages/d7/08/a316e74d41a1f3d606dfc3b71cc068354e8b9a0232f46bc098aa50b37116/protobuf-3.7.1-cp37-cp37m-macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl
Collecting paho-mqtt<2,>=1.4 (from esphome)
Collecting platformio>=3.6.5 (from esphome)
Collecting ecdsa (from esptool<3,>=2.6->esphome)
  Using cached https://files.pythonhosted.org/packages/23/a8/8aa68e70959e1287da9154e5164bb8bd5dd7025e41ae54e8d177b8d165c9/ecdsa-0.13.2-py2.py3-none-any.whl
Collecting pyaes (from esptool<3,>=2.6->esphome)
Collecting pytz (from tzlocal>=1.5.1->esphome)
  Using cached https://files.pythonhosted.org/packages/87/76/46d697698a143e05f77bec5a526bf4e56a0be61d63425b68f4ba553b51f2/pytz-2019.2-py2.py3-none-any.whl
Collecting six>=1.9 (from protobuf<3.8,>=3.7->esphome)
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Requirement already satisfied: setuptools in ./venv/lib/python3.7/site-packages (from protobuf<3.8,>=3.7->esphome) (41.2.0)
Collecting click<8,>=5 (from platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/fa/37/45185cb5abbc30d7257104c434fe0b07e5a195a6847506c074527aa599ec/Click-7.0-py2.py3-none-any.whl
Collecting tabulate<1,>=0.8.3 (from platformio>=3.6.5->esphome)
Collecting requests<3,>=2.4.0 (from platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/51/bd/23c926cd341ea6b7dd0b2a00aba99ae0f828be89d72b2190f27c11d4b7fb/requests-2.22.0-py2.py3-none-any.whl
Collecting colorama (from platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/4f/a6/728666f39bfff1719fc94c481890b2106837da9318031f71a8424b662e12/colorama-0.4.1-py2.py3-none-any.whl
Collecting semantic-version<3,>=2.8.1 (from platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/0f/3b/8fee26649a86c71df159ed0ae7ac5f9ac38829bccd8a7404e116f903929b/semantic_version-2.8.2-py2.py3-none-any.whl
Collecting bottle<0.13 (from platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/69/d1/efdd0a5584169cdf791d726264089ce5d96846a8978c44ac6e13ae234327/bottle-0.12.17-py3-none-any.whl
Collecting idna<2.9,>=2.5 (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
Collecting certifi>=2017.4.17 (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/18/b0/8146a4f8dd402f60744fa380bc73ca47303cccf8b9190fd16a827281eac2/certifi-2019.9.11-py2.py3-none-any.whl
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/e6/60/247f23a7121ae632d62811ba7f273d0e58972d75e58a94d329d51550a47d/urllib3-1.25.3-py2.py3-none-any.whl
Installing collected packages: ifaddr, ecdsa, pyaes, pyserial, esptool, colorlog, tornado, voluptuous, PyYAML, pytz, tzlocal, six, protobuf, paho-mqtt, click, tabulate, idna, chardet, certifi, urllib3, requests, colorama, semantic-version, bottle, platformio, esphome
Successfully installed PyYAML-5.1.2 bottle-0.12.17 certifi-2019.9.11 chardet-3.0.4 click-7.0 colorama-0.4.1 colorlog-4.0.2 ecdsa-0.13.2 esphome-1.13.6 esptool-2.7 idna-2.8 ifaddr-0.1.6 paho-mqtt-1.4.0 platformio-4.0.3 protobuf-3.7.1 pyaes-1.6.1 pyserial-3.4 pytz-2019.2 requests-2.22.0 semantic-version-2.8.2 six-1.12.0 tabulate-0.8.3 tornado-5.1.1 tzlocal-2.0.0 urllib3-1.25.3 voluptuous-0.11.7
(esphome)landons-imac:esphome lstewart$ esphome
Traceback (most recent call last):
  File "/Users/lstewart/esphome/venv/bin/esphome", line 6, in <module>
    from esphome.__main__ import main
  File "/Users/lstewart/esphome/venv/lib/python3.7/site-packages/esphome/__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "/Users/lstewart/esphome/venv/lib/python3.7/site-packages/esphome/writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "/Users/lstewart/esphome/venv/lib/python3.7/site-packages/esphome/config.py", line 15, in <module>
    from esphome import core, core_config, yaml_util
  File "/Users/lstewart/esphome/venv/lib/python3.7/site-packages/esphome/core_config.py", line 6, in <module>
    import esphome.config_validation as cv
  File "/Users/lstewart/esphome/venv/lib/python3.7/site-packages/esphome/config_validation.py", line 1242, in <module>
    Optional(CONF_COMMAND_TOPIC): All(requires_component('mqtt'), subscribe_topic),
  File "/Users/lstewart/esphome/venv/lib/python3.7/site-packages/esphome/voluptuous_schema.py", line 198, in extend
    ret = super(_Schema, self).extend(schema, extra=extra)
  File "/Users/lstewart/esphome/venv/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 782, in extend
    return result_cls(result, required=result_required, extra=result_extra)
TypeError: __init__() got an unexpected keyword argument 'required'
(esphome)landons-imac:esphome lstewart$ 
```

**Additional information and things you've tried:**
I tried using python2.7.  I even reinstalled python3 with brew and recreated the virtualenv as you see above.

LandonStewart, 2019-09-13:
Nevermind.  It works with the dev version.

```
pip install https://github.com/esphome/esphome/archive/dev.zip
```

closed 2019-09-13 by LandonStewart

solarkraft, 2019-10-04:
It does work with the dev version. But will the problem with the pip package ever be resolved? Otherwise maybe it shouldn't be recommended in the getting started guide.

glmnet, 2019-10-04:
If you want to run 1.13 you'll have to use an older voluptuous as stated [here](https://github.com/esphome/issues/issues/679#issuecomment-534224875)