esphome/issues #669: Cover modules: if you send multiple actions MQTT seems to crash.

neographikal, 2019-09-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome 1.14.0-dev using MQTT
HomeAssistant (MQTT) 0.98.5

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

Covers in general

I have 5 covers/screens connected to an ESP32 through 10 relays. I started out using the timebased cover, but since that didn't work well (position reporting is screwed it seems, going one way works but then ESPHome/HA seems to forget where it was, so you can't go the other way)) with HA over MQTT I tried the template version. That works okay-ish.

In both cases however, if I close or open multiple screens at once HA seems to lose connection with the ESPHome. All the buttons get greyed out in HA. Logging from ESPHome also suggests it loses connection:

```
[09:37:53][D][cover:072]: 'Master bedroom left template' - Setting
[09:37:53][D][cover:082]:   Command: CLOSE
[09:37:53][D][switch:025]: 'Screensfront GPIO17' Turning OFF.
[09:37:53][D][switch:021]: 'Screensfront GPIO16' Turning ON.
[09:37:53][D][switch:045]: 'Screensfront GPIO16': Sending state OFF
[09:37:53][D][cover:152]: 'Master bedroom left template' - Publishing:
**[09:37:53][I][mqtt:202]: MQTT Connected!**
[09:37:53][D][esp32_ble_tracker:470]: Found device 3E:B4:15:4E:D3:8D RSSI=-91
[09:37:53][D][esp32_ble_tracker:491]:   Address Type: RANDOM
[09:37:54][D][cover:072]: 'Master bedroom right template' - Setting
[09:37:54][D][cover:082]:   Command: CLOSE
[09:37:54][D][switch:025]: 'Screensfront GPIO15' Turning OFF.
[09:37:54][D][switch:021]: 'Screensfront GPIO14' Turning ON.
[09:37:54][D][switch:045]: 'Screensfront GPIO14': Sending state OFF
[09:37:54][D][cover:152]: 'Master bedroom right template' - Publishing:
[09:37:54][D][cover:160]:   State: CLOSED
**[09:37:58][I][mqtt:202]: MQTT Connected!**
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: screensfront
  platform: ESP32
  board: nodemcu-32s

<<: !include common.yaml

esp32_ble_tracker:
  scan_interval: 30s

switch:
  - platform: gpio
    pin: 13
    inverted: yes
    id: "gpio13"
    name: "Screensfront GPIO13"
    restore_mode: "ALWAYS_OFF"
#   interlock: ""
  - platform: gpio
    pin: 33
    inverted: yes
    id: "gpio33"
    name: "Screensfront GPIO33"
    restore_mode: "ALWAYS_OFF"

cover:
  - platform: template
    name: "Bay window left template"
    optimistic: true
    open_action:
      - switch.turn_off: gpio19
      - switch.turn_on: gpio21
      - delay: 40s
      - switch.turn_off: gpio21
    #open_duration: 37s
    close_action:
      - switch.turn_off: gpio21
      - switch.turn_on: gpio19
      - delay: 40s
      - switch.turn_off: gpio19
    #close_duration: 37s
    stop_action:
      - switch.turn_off: gpio21
      - switch.turn_off: gpio19
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
[image]

OttoWinter, 2019-10-21:
Hi, sorry for the late reply. That's weird, that should not happen (especially because the MQTT messages themselves are very short, much shorter than the discovery messages even). I suspect it might have something to do with the electronics - maybe the spike in current when the motors turn on is browning out the ESP.

To debug this, it would require logs form the device over USB while the issue is happening.

neographikal, 2019-10-29:
Thank, but I tried it with a 2A power supply as well, same result. If I send the commands 5-10s apart there is no problem, if I do it <2s apart it crashes. I don't see why that would trigger a voltage drop.

I'll prepare another ESP32 to confirm.

closed 2019-11-04 by no-response[bot]