esphome/issues #670: warning [display.nextion:041]: Waiting for ACK timed out!

Electronlibre2012, 2019-09-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESP8266 Wenos D1 Mini and Nextion 3.5" NX4832T035_11
**ESP (ESP32/ESP8266, Board/Sonoff):**
EspHome 1.13.6

**Affected component:**
https://esphome.io/components/display/nextion.html?highlight=nextion

**Description of problem:**
I have a continuous warning, sometimes for all id entity of my nection, and the api disconnected and reconnect, because too much ACK timed out : 30 per minute : like this exemple :
```
[15:06:51][W][display.nextion:041]: Waiting for ACK timed out!
[15:06:51][W][display.nextion:088]: Sending command 'conso.txt=4' failed because no ACK was received
```
i want to post back my experience becaus i found what is wrong in my config : 

I DUPLICATED AN ENTITY ID FROM A ESP32-CAM ALREADY PRESENT IN MY CONFIGURATION ESPHOME, à "switch" to restart the wifi of this camera, and after i change the name for this sxitch :
``` 
switch:
  - platform: restart
    name: "Restart Nextion35" --> i make a copy-paste of the same switch in a esp32-cam node of EspHome
```
after change this mistake, no more "Waiting for ACK timed out!",

if it could help others...

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

sermayoral, 2019-09-18:
Please close de issue if it is solved

glmnet, 2019-09-18:
I'm closing as OP said it's solved

closed 2019-09-18 by glmnet