esphome/issues #671: Captive Portal is not working

awarecan, 2019-09-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip, dev branch

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, nodemcu-32s

**Affected component:**

[WiFi](https://esphome.io/components/wifi.html)

**Description of problem:**
Try on new captive portal feature, it looks not working on ESP32.
I installed esphome from dev branch in order to enable the `captive_portal` component.
I also have to switch to dev branch of `platform-espressif32` to use  ESP-IDF v3.3 (so that https://github.com/espressif/esp-idf/issues/2915 can be fixed)

I have tried connect the soft-AP via Android Phone (Android P) and on Chrome browser on Mac OS, the captive portal never shows up. I just keep get "No Internet" error. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: m_proto
  platform: ESP32
  board: nodemcu-32s
  arduino_version: dev

wifi:
  ap:
    ssid: M Setup

logger:
  level: DEBUG

api:

ota:

captive_portal:
```

**Logs (if applicable):**

```
INFO Running:  platformio run -d m_proto -t upload --upload-port /dev/cu.SLAB_USBtoUART
Processing m_proto (board: nodemcu-32s; framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/stage)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 (Stage) 1.10.0-beta.1 #839ae90 > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), tool-mkspiffs 2.230.0 (2.30), framework-arduinoespressif32 06a399b, tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Hash library in registry
Warning! Library `{'requirements': None, 'name': 'Hash'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Hash'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESP Async WebServer> 1.2.2
|   |-- <AsyncTCP> 1.0.3
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <Update> 1.0
Linking .pioenvs/m_proto/firmware.elf
Building .pioenvs/m_proto/firmware.bin
Retrieving maximum program size .pioenvs/m_proto/firmware.elf
Checking size .pioenvs/m_proto/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  12.9% (used 42188 bytes from 327680 bytes)
PROGRAM: [=====     ]  52.7% (used 862858 bytes from 1638400 bytes)
esptool.py v2.6
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/cu.SLAB_USBtoUART
Uploading .pioenvs/m_proto/firmware.bin
esptool.py v2.6
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
MAC: 30:ae:a4:74:ff:84
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15568 bytes to 10188...
Wrote 15568 bytes (10188 compressed) at 0x00001000 in 0.9 seconds (effective 137.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 1427.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 7573.0 kbit/s)...
Hash of data verified.
Compressed 862976 bytes to 483074...
Wrote 862976 bytes (483074 compressed) at 0x00010000 in 42.7 seconds (effective 161.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
============================================================ [SUCCESS] Took 53.87 seconds ============================================================
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.SLAB_USBtoUART with baud rate 115200
[01:23:34][I][logger:156]: Log initialized
[01:23:34][C][ota:364]: There have been 2 suspected unsuccessful boot attempts.
[01:23:34][I][app:028]: Running through setup()...
[01:23:34][C][wifi:033]: Setting up WiFi...
[01:23:34][C][wifi:147]: Setting up AP...
[01:23:34][C][wifi:149]:   AP SSID: 'M Setup'
[01:23:34][C][wifi:150]:   AP Password: ''
[01:23:34][C][wifi:159]:   IP Address: 192.168.4.1
[01:23:34][C][ota:029]: Over-The-Air Updates:
[01:23:34][C][ota:030]:   Address: m_proto.local:3232
[01:23:34][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[01:23:34][C][api:022]: Setting up Home Assistant API server...
[01:23:34][I][app:058]: setup() finished successfully!
[01:23:34][I][app:101]: esphome version 1.14.0-dev compiled on Sep 17 2019, 01:22:33
[01:23:34][C][logger:175]: Logger:
[01:23:34][C][logger:176]:   Level: DEBUG
[01:23:34][C][logger:177]:   Log Baud Rate: 115200
[01:23:34][C][logger:178]:   Hardware UART: UART0
[01:23:34][C][wifi:399]: WiFi:
[01:23:34][C][wifi:267]:   SSID: ''
[01:23:34][C][wifi:268]:   IP Address: 0.0.0.0
[01:23:34][C][wifi:270]:   BSSID: 00:00:00:00:00:00
[01:23:34][C][wifi:271]:   Hostname: 'm_proto'
[01:23:34][C][wifi:275]:   Signal strength: 0 dB ▂▄▆█
[01:23:34][C][wifi:279]:   Channel: 1
[01:23:34][C][wifi:280]:   Subnet: 0.0.0.0
[01:23:34][C][wifi:281]:   Gateway: 0.0.0.0
[01:23:34][C][wifi:282]:   DNS1: 0.0.0.0
[01:23:34][C][wifi:283]:   DNS2: 0.0.0.0
[01:23:34][C][ota:029]: Over-The-Air Updates:
[01:23:34][C][ota:030]:   Address: m_proto.local:3232
[01:23:34][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[01:23:34][C][api:095]: API Server:
[01:23:34][C][api:096]:   Address: m_proto.local:6053
[01:23:53]dhcps: send_offer>>udp_sendto result 0
[01:25:34][I][ota:046]: Boot seems successful, resetting boot loop counter.
```

**Additional information and things you've tried:**
If I use "stable" version esp32, I will get following error mesage when I tried to connect the AP
```
[00:31:33][D][:000]: E (42261) event: mismatch or invalid event, id=63
[00:31:33]
[00:31:33][D][:000]: E (42262) event: default event handler failed!
[00:31:33]
```

awarecan, 2019-09-17:
More information:

ping 192.168.4.1 OK

```
> nslookup google
;; connection timed out; no servers could be reached
```

The hotspot detect query sent from my laptop when I connect to soft-AP
[image]

Response
[image]

OttoWinter, 2019-10-18:
Hmm, looked into this and can reproduce. I'm getting these lines in the verbose log:

```
[12:18:59][VV][wifi_esp8266:365]: Event: AP receive Probe Request MAC=... RSSI=-42
[12:18:59]wifi evt: 7
[12:19:01]ERROR: send_offer (error -13)
```

And this looks related https://github.com/esp8266/Arduino/issues/6114

This will need to be fixed before 1.14 is shipped

OttoWinter, 2019-10-18:
Ah ok I misread, this is about ESP32 🤦

I think the issue here is a different thing: The captive portal is never enabled because the device doed not have any STA config. I guess the behavior can be changed to always enable captive portal in that case.

closed 2019-10-18 by OttoWinter