esphome/issues #672: Rotary Encoder "Guru Meditation Error" - ESP32 tried on 1.13.6 and 1.14.0-dev

justfra, 2019-09-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU-32S (ESP32)

**Affected component:**

Rotary Encoder: [https://esphome.io/components/sensor/rotary_encoder.html](url)

**Description of problem:**
The ESP32 crashes with "Guru Meditation Error". I know there was a known bug similar to my issue which had been fixed but even though I'm using the latest versions of ESPHome (tried 1.13.6 and 1.14.0-dev) I still face the problem.
If I turn the knob very slow most of the times it works but I have to be super slow otherwise it crashes.
If I do not attach the encoder to any output it works although even if I attached the output to some other input (not the encoder) it works.
I'm using an LPD3806 encoder.

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: smart_dimmerable_lights
  platform: ESP32
  board: nodemcu-32s
    
wifi:
  ssid: "******"
  password: "******"
# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: rotary_encoder
    name: "Encoder Channel 1"
    resolution: 2
    pin_a:
      number: GPIO22
      mode: INPUT_PULLUP
    pin_b:
      number: GPIO23
      mode: INPUT_PULLUP
    filters:
      - or:
        - debounce: 0.1s
        - delta: 10
      - lambda: |-
          if (x < 0.0) return 0.0;
          if (x > 255.0) return 255.0;
          return x;
    on_value:
      then:
        - light.turn_on:
            id: light1
            brightness: !lambda |-
              return x / 255;
        - delay: 100ms
    
output:
  - platform: ledc
    pin: GPIO2
    id: gpio_2

light:
[... 181 lines cut ...]
[0;35m[C][wifi:282]:   DNS1: 0.0.0.0[0m
[0;35m[C][wifi:283]:   DNS2: 0.0.0.0[0m
[0;36m[D][sensor:092]: 'Encoder Channel 1': Sending state 1.00000 steps with 0 decimals of accuracy[0m
[0;36m[D][light:261]: 'Led Channel 1' Setting:[0m
[0;36m[D][light:274]:   Brightness: 0%[0m
[0;36m[D][light:300]:   Transition Length: 1.0s[0m
[0;36m[D][sensor:092]: 'Encoder Channel 1': Sending state 13.00000 steps with 0 decimals of accuracy[0m
[0;36m[D][light:261]: 'Led Channel 1' Setting:[0m
[0;36m[D][light:274]:   Brightness: 5%[0m
[0;36m[D][light:300]:   Transition Length: 1.0s[0m
[0;35m[C][ledc.output:032]: LEDC Output:[0m
[0;35m[C][ledc.output:033]:   PinGPIO2 (Mode: OUTPUT)[0m
[0;35m[C][ledc.output:034]:   LEDC Channel: 0[0m
[0;35m[C][ledc.output:035]:   Frequency: 1000.0 Hz[0m
[0;35m[C][ledc.output:036]:   Bit Depth: 15[0m
[0;36m[D][sensor:092]: 'Encoder Channel 1': Sending state 33.00000 steps with 0 decimals of accuracy[0m
[0;36m[D][light:261]: 'Led Channel 1' Setting:[0m
[0;36m[D][light:274]:   Brightness: 13%[0m
[0;36m[D][light:300]:   Transition Length: 1.0s[0m
**Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4015e590: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4015e590: bad00bad bad00bad bad00bad**
```

justfra, 2019-09-22:
Tried version 1.13.2 where issue https://github.com/esphome/esphome/pull/580 was fixed although I experience the same problem..

justfra, 2019-09-23:
I think this might be related the implementation of the encoder sensor using interrupts. I have also tried another type of encoder (KY-040, while usually I had been using a LPD3806) although the same exact problem occurs. I'm not sure how this could be working for other people.

In my yaml file if I change the following
```
    on_value:
      then:
        - light.turn_on:
            id: light1
            brightness: !lambda |-
              return x / 255;
```
to:
```
    on_value:
      then:
        - delay: 10ms
```

It doesn't crash. That would make me think the issue could be related to updating the light brightness, although I also tried changing the rotary_encoder.cpp implementation not to use interrupts and it works that way so I'm still more inclined to believe the issue has to do with the interrupts.

axiom00, 2019-11-02:
@justfra did you ever solve the issue? I am running into the same error. Even on the latest 1.14.0.

justfra, 2019-11-03:
@axiom00 unfortunately not. I gave up on using ESPHome for this project..

axiom00, 2019-11-05:
Updated to 1.14.2 today and the esp32 is panicking less. If I turn the encoder slow (1 click/second) it behaves normally and change the brightness of the LED. At medium speed (2 clicks/second) on 1.13 would easily cause it to crash. On 1.14.2 at medium speed it is still stable. Faster than medium speed would cause it to crash and reboot.

@OttoWinter Here's the serial log of the crash. Let me know if there's anything else that would help debugging the crash. 
```
[18:10:01][D][sensor:092]: 'Rotary Encoder': Sending state -24.00000 steps with 0 decimals of accuracy
[18:10:01][D][light:264]: 'ql_quad_01 ouput1' Setting:
[18:10:01][D][light:277]:   Brightness: 100%
[18:10:01][D][light:303]:   Transition Length: 2.0s
[18:10:01]Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
[18:10:01]Core 1 register dump:
[18:10:01]PC      : 0x40080fa1  PS      : 0x00060034  A0      : 0x80081210  A1      : 0x3ffbe790  
[18:10:01]A2      : 0x3ffb8d30  A3      : 0x3f403aee  A4      : 0x00060023  A5      : 0x3ffbe790  
[18:10:01]A6      : 0x3ffbe7d8  A7      : 0x00000001  A8      : 0x3f403ad4  A9      : 0x0000bad0  
[18:10:01]A10     : 0x00000000  A11     : 0x3ffb19b0  A12     : 0x8008a60b  A13     : 0x3ffbe760  
[18:10:01]A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x0000001e  EXCCAUSE: 0x00000007  
[18:10:01]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[18:10:01]Core 1 was running in ISR context:
[18:10:01]EPC1    : 0x40062235  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40080fa1
[18:10:01]
[18:10:01]Backtrace: 0x40080fa1:0x3ffbe790 0x4008120d:0x3ffbe7b0 0x40084a09:0x3ffbe7d0 0x40062232:0x3ffb18f0 0x40092613:0x3ffb1910 0x4009264a:0x3ffb1940 0x400926e9:0x3ffb1970 0x40092b7e:0x3ffb1990 0x40087069:0x3ffb19b0 0x400870da:0x3ffb19d0 0x400fad13:0x3ffb19f0 0x400fb0a1:0x3ffb1a10 0x400f8f4a:0x3ffb1a30 0x400f946e:0x3ffb1ab0 0x400f8851:0x3ffb1b20 0x400deab8:0x3ffb1b60 0x400dec83:0x3ffb1bb0 0x400d89d7:0x3ffb1bd0 0x400df766:0x3ffb1c40 0x400df7b5:0x3ffb1cd0 0x4015b8a5:0x3ffb1d00 0x400df82c:0x3ffb1d20 0x4015b891:0x3ffb1d50 0x400df819:0x3ffb1d70 0x4015b8a5:0x3ffb1da0 0x400df7e9:0x3ffb1dc0 0x400da641:0x3ffb1de0 0x400da6ca:0x3ffb1e10 0x400da2ce:0x3ffb1e70 0x400da345:0x3ffb1e90 0x400da2e6:0x3ffb1ec0 0x400da2d3:0x3ffb1ef0 0x400da335:0x3ffb1f10 0x40081133:0x3ffb1f30 0x400dd135:0x3ffb1f60 0x400df696:0x3ffb1f90 0x400e5801:0x3ffb1fb0 0x40088e15:0x3ffb1fd0
[18:10:01]
[18:10:01]Rebooting...
```

OttoWinter, 2019-11-05:
Ah I see what's happening now.

All the functions are correctly placed in iram0. However, the global lookup table is not and resides in flash, triggering the cache disabled issue.

closed 2019-11-05 by OttoWinter

axiom00, 2019-11-06:
@OttoWinter I tested the latest dev branch and it's working perfectly. Thanks for fixing it!