esphome/issues #674: Remote Receiver does't detect RC5 codes

Santobert, 2019-09-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 NodeMCU v2

**Affected component:**

Remote Receiver
https://esphome.io/components/remote_receiver.html

**Description of problem:**
The Remote Receiver component doesn't detect RC5 code. The same procedure worked will with another remote and NEC codes. I'll post some examples below.

**Problem-relevant YAML-configuration entries:**
```yaml
remote_receiver:
  pin:
    number: D5
    inverted: True
  dump: all
```

**Logs (if applicable):**

Press On/Off
```
{"Protocol":"RC5","Bits":12,"Data":0x4C}
```
```
[08:19:13][D][remote.raw:041]: Received Raw: 1207, -590, 1838, -843, 920, -864, 900, -884, 885, -905, 874, -1802, 1766, -903, 879, -1801, 869, -911, 1766, -910, 872
[08:19:13][D][remote.raw:041]: Received Raw: 1229, -554, 1194, -605, 1839, -840, 924, -860, 903, -883, 883, -1798, 1771, -900, 883, -1793, 880, -903, 1771, -909, 876
[08:19:14][D][remote.raw:041]: Received Raw: 1215, -570, 1858, -838, 930, -855, 912, -874, 890, -897, 882, -1799, 1769, -904, 880, -1793, 884, -909, 1763, -912, 873
[08:19:15][D][remote.raw:041]: Received Raw: 1221, -563, 1187, -614, 1833, -845, 916, -868, 900, -884, 881, -1797, 1769, -914, 872, -1796, 878, -909, 1768, -905, 876
[08:19:16][D][remote.raw:041]: Received Raw: 1220, -564, 1860, -834, 931, -853, 908, -876, 891, -895, 881, -1797, 1767, -910, 875, -1795, 880, -904, 1772, -908, 874
```
Press Mute
```
{"Protocol":"RC5","Bits":12,"Data":0x4D}
```
```
[08:22:24][D][remote.raw:041]: Received Raw: 974, -820, 1815, -862, 902, -882, 885, -907, 873, -912, 875, -1796, 1771, -907, 875, -1797, 878, -907, 1769, -1797, 880
[08:22:24][D][remote.raw:041]: Received Raw: 997, -805, 954, -830, 1809, -869, 892, -892, 885, -905, 878, -1798, 1769, -908, 874, -1797, 875, -911, 1763, -1800, 881
[08:22:25][D][remote.raw:041]: Received Raw: 979, -816, 1819, -858, 902, -882, 887, -901, 881, -904, 879, -1796, 1769, -908, 877, -1796, 878, -903, 1771, -1802, 872
[08:22:25][D][remote.raw:041]: Received Raw: 971, -821, 943, -841, 1790, -886, 882, -909, 877, -904, 881, -1796, 1768, -911, 871, -1798, 880, -902, 1769, -1802, 874
[08:22:26][D][remote.raw:041]: Received Raw: 984, -815, 1824, -853, 914, -871, 892, -893, 882, -905, 880, -1797, 1767, -909, 874, -1800, 880, -907, 1766, -1798, 881
```

**Additional information and things you've tried:**
I've recorded some key press events with esphome and with sonoff tasmota. The given json payloads worked with tasmota. The given raw codes were recorded with esphome.

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#ir-remote

Santobert, 2019-10-22:
Is someone able to fix this?

LukasK13, 2020-04-23:
I can confirm this issue. Any update here?

closed 2020-06-18 by glmnet