esphome/issues #676: Panel entry for independent Docker deployment?

Nornode, 2019-09-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker (Independent containers)
**ESP (ESP32/ESP8266, Board/Sonoff):**

All,
**Affected component:**

Dashboard & HomeAssistant

**Description of problem:**

Hi, this is awesome software!
Thanks!

Would you have a preferred way of integrating this into the panel (left menu in Homeassistant)
for us that don't run HASS.io?
I'm running the HA & ESPhome in containers and would still like to have the feel with the left-hand menu.

How should I implement a custom panel entry when I have HA accessible externally on https://ha.***.net and would still like to be able to access ESPhome dashboard which is hosted on 192...*:6052 ?

Thanks!

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Valcob, 2019-09-25:
I guess all you need is just to expose the 6052 port when u r starting the esphome docker container
`docker run -p 6052:6052 -v "${PWD}":/config -it esphome/esphome`
And then just do it the same way as any other panel integration by ip:port

OttoWinter, 2019-10-20:
Exactly, you need expose it through a different port (for example 6052) and also expose that port on your router.

closed 2019-10-20 by OttoWinter