esphome/issues #677: Can not Compiling on hassio

vipk31, 2019-09-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**ESP8266

**Affected component:**

**Description of problem:**
can not Compiling 
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: lalaa
  platform: ESP8266
  board: esp01_1m
  build_path: lalaa
  platformio_options: {}
  esp8266_restore_from_flash: false
  arduino_version: espressif8266@1.8.0
  includes: []
  libraries: []
  board_flash_mode: dout
wifi:
  fast_connect: false
  domain: .local
  power_save_mode: NONE
  reboot_timeout: 5min
  networks:
  - password: yyyyyyyy
    ssid: xxxxxx
  use_address: lalaa.local
logger:
  logs: {}
  esp8266_store_log_strings_in_flash: true
  level: DEBUG
  baud_rate: 115200
  tx_buffer_size: 512
  hardware_uart: UART0
api:
  password: '123654'
  reboot_timeout: 5min
  port: 6053
ota:
  password: '123654'
  safe_mode: true
  port: 8266

```

**Logs (if applicable):**

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/lalaa
Processing lalaa (framework: arduino; platform: espressif8266@1.8.0; board: esp01_1m)
--------------------------------------------------------------------------------
LibraryManager: Installing id=305 @ 1.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Compiling /data/lalaa/.pioenvs/lalaa/src/esphome/components/api/api_message.cpp.o
Compiling /data/lalaa/.pioenvs/lalaa/src/esphome/components/api/api_server.cpp.o
Compiling /data/lalaa/.pioenvs/lalaa/src/esphome/components/api/basic_messages.cpp.o
Compiling /data/lalaa/.pioenvs/lalaa/src/esphome/components/api/command_messages.cpp.o
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/lalaa/.pioenvs/lalaa/src/esphome/components/api/api_message.cpp.o] Error 4
========================== [ERROR] Took 17.85 seconds ==========================

```

**Additional information and things you've tried:**

alexyao2015, 2019-09-22:
Having similar issues. Can't compile anything after clearing the build cache. Get the same error every single time

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test
Processing test (framework: arduino; platform: espressif8266@1.8.0; board: esp01_1m)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <FastLED> 3.2.0
|   |-- <EspSoftwareSerial> 3.4.1
|-- <ESPAsyncTCP> 1.2.1
|   |-- <ESP8266WiFi> 1.0
Linking /data/test/.pioenvs/test/firmware.elf
/data/test/.pioenvs/test/liba68/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient7_s_recvEPvP7tcp_pcbP4pbufl+0x4): multiple definition of `AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long)'
/data/test/.pioenvs/test/lib87e/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient7_s_recvEPvP7tcp_pcbP4pbufl+0x10): first defined here
/data/test/.pioenvs/test/liba68/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::_s_sent(void*, tcp_pcb*, unsigned short)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient7_s_sentEPvP7tcp_pcbt+0x4): multiple definition of `AsyncClient::_s_sent(void*, tcp_pcb*, unsigned short)'
/data/test/.pioenvs/test/lib87e/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient7_s_sentEPvP7tcp_pcbt+0x10): first defined here
/data/test/.pioenvs/test/liba68/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::_s_error(void*, long)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient8_s_errorEPvl+0x4): multiple definition of `AsyncClient::_s_error(void*, long)'
/data/test/.pioenvs/test/lib87e/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient8_s_errorEPvl+0xc): first defined here
/data/test/.pioenvs/test/liba68/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::_s_poll(void*, tcp_pcb*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient7_s_pollEPvP7tcp_pcb+0x4): multiple definition of `AsyncClient::_s_poll(void*, tcp_pcb*)'
/data/test/.pioenvs/test/lib87e/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient7_s_pollEPvP7tcp_pcb+0x10): first defined here
/data/test/.pioenvs/test/liba68/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::AsyncClient(tcp_pcb*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClientC2EP7tcp_pcb+0x30): multiple definition of `AsyncClient::AsyncClient(tcp_pcb*)'
/data/test/.pioenvs/test/lib87e/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClientC2EP7tcp_pcb+0x48): first defined here
/data/test/.pioenvs/test/liba68/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::AsyncClient(tcp_pcb*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClientC2EP7tcp_pcb+0x30): multiple definition of `AsyncClient::AsyncClient(tcp_pcb*)'
/data/test/.pioenvs/test/lib87e/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClientC2EP7tcp_pcb+0x48): first defined here
/data/test/.pioenvs/test/liba68/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::_s_connected(void*, void*, long)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient12_s_connectedEPvS0_l+0x4): multiple definition of `AsyncClient::_s_connected(void*, void*, long)'
/data/test/.pioenvs/test/lib87e/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient12_s_connectedEPvS0_l+0x10): first defined here
/data/test/.pioenvs/test/liba68/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::connect(IPAddress, unsigned short)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient7connectE9IPAddresst+0x18): multiple definition of `AsyncClient::connect(IPAddress, unsigned short)'
/data/test/.pioenvs/test/lib87e/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient7connectE9IPAddresst+0x1c): first defined here
/data/test/.pioenvs/test/liba68/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::_s_dns_found(char const*, ip_addr const*, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient12_s_dns_foundEPKcPK7ip_addrPv+0x4): multiple definition of `AsyncClient::_s_dns_found(char const*, ip_addr const*, void*)'
/data/test/.pioenvs/test/lib87e/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient12_s_dns_foundEPKcPK7ip_addrPv+0x4): first defined here
/data/test/.pioenvs/test/liba68/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::connect(char const*, unsigned short)':
[... cut ...]
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x8): multiple definition of `AsyncClient::onTimeout(std::function<void (void*, AsyncClient*, unsigned int)>, void*)'
/data/test/.pioenvs/test/lib87e/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x8): first defined here
collect2: error: ld returned 1 exit status
*** [/data/test/.pioenvs/test/firmware.elf] Error 1
========================== [ERROR] Took 4.26 seconds ==========================
```

zimpower, 2019-09-22:
Looks like I am having identical issue to @alexyao2015 - failing on linking.

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
- homeassistant v0.99.2
- hassio v188
- ESPHOME addin 1.13.6

**ESP (ESP32/ESP8266, Board/Sonoff):**
- ESP8285

**Affected component:**
- Teckin Smart Plug SP23

**Description of problem:**
can not Compiling

**Logs (if applicable):**
```
INFO Reading configuration...
INFO Detected timezone 'GMT' with UTC offset 0 and daylight savings time from 03/29/20 01:00:00 to 10/25/20 02:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/smart_plug_6
Processing smart_plug_6 (framework: arduino; platform: espressif8266@1.8.0; board: esp8285)
--------------------------------------------------------------------------------
LibraryManager: Installing id=3837 @ 5.13.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing id=306 @ 1.1.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing id=305
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=1826
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=305 @ 1.2.0
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8285.html
PLATFORM: Espressif 8266 > Generic ESP8285 Module
HARDWARE: ESP8266 80MHz 80KB RAM (423.98KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/smart_plug_6/.pioenvs/smart_plug_6/src/esphome/components/api/api_message.cpp.o
Compiling /data/smart_plug_6/.pioenvs/smart_plug_6/src/esphome/components/api/api_server.cpp.o
Compiling /data/smart_plug_6/.pioenvs/smart_plug_6/src/esphome/components/api/basic_messages.cpp.o
Compiling /data/smart_plug_6/.pioenvs/smart_plug_6/src/esphome/components/api/command_messages.cpp.o
Compiling /data/smart_plug_6/.pioenvs/smart_plug_6/src/esphome/components/api/list_entities.cpp.o
Compiling /data/smart_plug_6/.pioenvs/smart_plug_6/src/esphome/components/api/service_call_message.cpp.o
Compiling /data/smart_plug_6/.pioenvs/smart_plug_6/src/esphome/components/api/subscribe_logs.cpp.o
[... 346 lines cut ...]
/data/smart_plug_6/.pioenvs/smart_plug_6/lib2ed/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::onPoll(std::function<void (void*, AsyncClient*)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient6onPollESt8functionIFvPvPS_EES1_+0x4): multiple definition of `AsyncClient::onPoll(std::function<void (void*, AsyncClient*)>, void*)'
/data/smart_plug_6/.pioenvs/smart_plug_6/liba31/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient6onPollESt8functionIFvPvPS_EES1_+0x4): first defined here
/data/smart_plug_6/.pioenvs/smart_plug_6/lib2ed/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncServer::onClient(std::function<void (void*, AsyncClient*)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncServer8onClientESt8functionIFvPvP11AsyncClientEES1_+0x4): multiple definition of `AsyncServer::onClient(std::function<void (void*, AsyncClient*)>, void*)'
/data/smart_plug_6/.pioenvs/smart_plug_6/liba31/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncServer8onClientESt8functionIFvPvP11AsyncClientEES1_+0x4): first defined here
/data/smart_plug_6/.pioenvs/smart_plug_6/lib2ed/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::onAck(std::function<void (void*, AsyncClient*, unsigned int, unsigned int)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient5onAckESt8functionIFvPvPS_jjEES1_+0x8): multiple definition of `AsyncClient::onAck(std::function<void (void*, AsyncClient*, unsigned int, unsigned int)>, void*)'
/data/smart_plug_6/.pioenvs/smart_plug_6/liba31/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient5onAckESt8functionIFvPvPS_jjEES1_+0x8): first defined here
/data/smart_plug_6/.pioenvs/smart_plug_6/lib2ed/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::onData(std::function<void (void*, AsyncClient*, void*, unsigned int)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient6onDataESt8functionIFvPvPS_S1_jEES1_+0x8): multiple definition of `AsyncClient::onData(std::function<void (void*, AsyncClient*, void*, unsigned int)>, void*)'
/data/smart_plug_6/.pioenvs/smart_plug_6/liba31/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient6onDataESt8functionIFvPvPS_S1_jEES1_+0x8): first defined here
/data/smart_plug_6/.pioenvs/smart_plug_6/lib2ed/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::onPacket(std::function<void (void*, AsyncClient*, pbuf*)>, void*)':
[... cut ...]
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x8): multiple definition of `AsyncClient::onTimeout(std::function<void (void*, AsyncClient*, unsigned int)>, void*)'
/data/smart_plug_6/.pioenvs/smart_plug_6/liba31/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x8): first defined here
collect2: error: ld returned 1 exit status
*** [/data/smart_plug_6/.pioenvs/smart_plug_6/firmware.elf] Error 1
========================== [ERROR] Took 26.64 seconds ==========================
```
**Additional information and things you've tried:**
- Clean Build
- Reboot

codedmind, 2019-09-22:
The issue appears to be in the linking with two versions of ESPAsyncTCP (1.2.1 and 1.2.0) don't know how to solve it :/

alexyao2015, 2019-09-22:
Looking at the GitHub for ESPAsyncTCP, there were recent commits to the repository which would explain why it stopped working. All new builds would download the new version and have the problem.

zimpower, 2019-09-22:
Thanks - do you know if it is possible to roll back the dependency versions in the yaml or in hassio ?

alexyao2015, 2019-09-22:
Yeah if you just want it to compile for now, you can copy over the build cache from another node to this one. That'll prevent it from downloading the lastest version.

a74451, 2019-09-22:
I have the same problem here, for now, what worked for me was disabling the web server.

comment ou remove this lines.
web_server:
  port: 80

zimpower, 2019-09-22:
Thanks that worked.

glmnet, 2019-09-23:
Closing as seems fixed.

dinkyshaw, 2019-09-23:
I had the same issue and commenting out web_server worked.

alexyao2015, 2019-09-23:
@glmnet 
The issue isn't fixed. There is just a workaround to allow it to temporarily compile by disabling the webserver. The issue still exists since the code cannot compile with the webserver because of a recent change in ESPasyncTCP.

glmnet, 2019-09-23:
Ok reopening it,
It is just that I cannot reproduce it:

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/lalala
Processing lalala (framework: arduino; platform: espressif8266@1.8.0; board: esp01_1m)
--------------------------------------------------------------------------------
LibraryManager: Installing id=305 @ 1.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/api/api_message.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/api/api_server.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/api/basic_messages.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/api/command_messages.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/api/list_entities.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/api/service_call_message.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/api/subscribe_logs.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/api/user_services.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/api/util.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/logger/logger.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/wifi/wifi_component.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/wifi/wifi_component_esp32.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/core/application.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/core/component.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/core/controller.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/core/esphal.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/core/helpers.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/core/log.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/core/preferences.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/esphome/core/util.cpp.o
Compiling /data/lalala/.pioenvs/lalala/src/main.cpp.o
Generating LD script /data/lalala/.pioenvs/lalala/ld/eagle.app.v6.common.ld
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/WiFiClient.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/WiFiClientSecureAxTLS.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/WiFiServer.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libd05/ESP8266WiFi/WiFiUdp.cpp.o
Compiling /data/lalala/.pioenvs/lalala/lib3f7/ESP8266mDNS/ESP8266mDNS.cpp.o
Compiling /data/lalala/.pioenvs/lalala/libb74/ESPAsyncTCP_ID305/AsyncPrinter.cpp.o
[... 50 lines cut ...]
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/libb64/cencode.c.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/libc_replacements.c.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/pgmspace.cpp.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/sntp-lwip2.c.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/spiffs/spiffs_cache.c.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/spiffs/spiffs_check.c.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/spiffs/spiffs_gc.c.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/spiffs/spiffs_hydrogen.c.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/spiffs/spiffs_nucleus.c.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/spiffs_api.cpp.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/spiffs_hal.cpp.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/time.c.o
Compiling /data/lalala/.pioenvs/lalala/FrameworkArduino/uart.c.o
[... cut ...]
Linking /data/lalala/.pioenvs/lalala/firmware.elf
Building /data/lalala/.pioenvs/lalala/firmware.bin
Retrieving maximum program size /data/lalala/.pioenvs/lalala/firmware.elf
Checking size /data/lalala/.pioenvs/lalala/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  39.0% (used 31968 bytes from 81920 bytes)
PROGRAM: [===       ]  30.9% (used 316312 bytes from 1023984 bytes)
========================= [SUCCESS] Took 83.99 seconds =========================
INFO Successfully compiled program.
```

glmnet, 2019-09-23:
Ok, it failed now.

Reproduce steps are:
have a config which uses `web_server` click clean build, click compile

It works if you compile without `web_server` and then compile with `web_server` if you don't click clean in between.
I believe there is no need to use clean build anyway

glmnet, 2019-09-23:
@vipk31 is this your same issue? I don't see you're using `web_server` there.

alexyao2015, 2019-09-23:
Another way to reproduce is to just try adding a new node with `web_server` which is what I did.

alexyao2015, 2019-09-23:
> @vipk31 is this your same issue? I don't see you're using `web_server` there.

I feel like @vipk31 has a different issue than the one that we have since the logs are showing a different error.

alexconrey, 2019-09-23:
I'm also able to confirm that with both `web_server` and/or `mqtt` enabled in configs - compilation fails:

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d kitchen_light_controller
Processing kitchen_light_controller (board: nodemcuv2; framework: arduino; platform: espressif8266@1.8.0)
-------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 1.8.0 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 2.20402.4 (2.4.2), tool-esptool 1.413.0 (4.13)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Update library in registry
Warning! Library `{'name': 'Update', 'requirements': None}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'name': 'Update', 'requirements': None}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.1
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.1
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/api/api_message.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/api/basic_messages.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/api/command_messages.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/api/service_call_message.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/api/subscribe_logs.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/api/util.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/gpio/output/gpio_binary_output.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/custom_mqtt_device.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/mqtt_binary_sensor.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/mqtt_client.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/mqtt_climate.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/mqtt_component.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/mqtt_cover.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/mqtt_fan.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/mqtt_light.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/mqtt_sensor.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/mqtt_switch.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/mqtt/mqtt_text_sensor.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/output/automation.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/output/float_output.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/output/switch/output_switch.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/switch/automation.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/switch/switch.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/wifi/wifi_component_esp32.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/core/application.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/core/component.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/core/controller.cpp.o
Compiling .pioenvs/kitchen_light_controller/src/esphome/core/esphal.cpp.o
[... 313 lines cut ...]
.pioenvs/kitchen_light_controller/lib3c9/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::onDisconnect(std::function<void (void*, AsyncClient*)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient12onDisconnectESt8functionIFvPvPS_EES1_+0x4): multiple definition of `AsyncClient::onDisconnect(std::function<void (void*, AsyncClient*)>, void*)'
[... cut ...]
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x8): multiple definition of `AsyncClient::onTimeout(std::function<void (void*, AsyncClient*, unsigned int)>, void*)'
.pioenvs/kitchen_light_controller/lib575/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x8): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/kitchen_light_controller/firmware.elf] Error 1
============================================================= [FAILED] Took 13.21 seconds ============================================================
```

glmnet, 2019-09-23:
@alexconrey can you create another issue to track this? OP Issue is another thing

brandond, 2019-09-23:
@alexconrey I've hidden your link failures. As @glmnet said, please open a different issue for these as a failure to link is not the same as the compiler segfaulting.

LastedGuide, 2019-10-02:
Same problem here, I am running the docker on rasbian though

jcbevns, 2019-10-05:
Where are you commenting out ```web_server```, its not in my .yaml but I'm guessing its on a build file. Somebody with a path?

LastedGuide, 2019-10-05:
I have the same problem @jcbevns I don't have `web_server` to comment out

patrul, 2019-10-06:
Same error here. 

No problem compiling with ESP32 (no need to comment out webserver), but I cannot make it work with Sonoff (Basic, RF Bridge and Mini)

Everything was working properly on Ubuntu VM but I moved to RPI4

glmnet, 2019-10-06:
Can everybody post information of about the hardware using? e.g. RPI4, etc.

alexyao2015, 2019-10-07:
Nodemcu with Ubuntu vm running hassio

dmitrybabeshko, 2019-10-07:
Same problem. RPI4 4Gb running hassio. Works for   ESP32, but segfault for sonoff basic.

wtrbuit, 2019-10-08:
**Same problem**
- Removing webserver doesn't work
- Clean build before compile doesn't work
- Tested with stable and dev version

**Hardware**
- RPI 3B+
- HassIO in Docker

KarlosHit, 2019-10-09:
I have exactly the same problem with an ESP8266, I get the same error when compiling, tried everything, and nothing ... it doesn't work. If I do not solve it, I will have to leave esphome :(

My error is this:
"Compiling .pioenvs/prueba/src/esphome/components/mqtt/mqtt_binary_sensor.cpp.o
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/prueba/src/esphome/components/mqtt/mqtt_binary_sensor.cpp.o] Error 4"

And my code is this:

esphome:
  name: prueba
  platform: ESP8266
#  board: esp01_1m
  board: esp01
  
wifi:
  ssid: "****"
  password: "****"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "****"

ota:
  password: "****"

web_server:
  port: 80

mqtt:
  broker: 192.168.31.15
  username: "****"
  password: "****"
  
sensor:
  - platform: dht
    pin: 
      number: GPIO0
    model: DHT22
    temperature:
      name: "Temperatura DHT22"
    humidity:
      name: "Humedad Relativa DHT22"
    update_interval: 60s

glmnet, 2019-10-09:
people try `free -m` command the issue might be related to low memory

glmnet, 2019-10-09:
try rebooting to.. another thing to try is [this ](https://stackoverflow.com/a/44288591/229052)

this is not ESPHome bug specific, the error comes in different times of compilation, the answer should be out there, keep reporting

patrul, 2019-10-09:
> people try `free -m` command the issue might be related to low memory

```
$ free -m
              total        used        free      shared  buff/cache   available
Mem:           3906         923         617          17        2365        2885
Swap:            99           0          99
```

Wojcioo, 2019-10-10:
It doesn't work for wemos d1 mini, too

INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/fan
Processing fan (framework: arduino; platform: espressif8266@1.8.0; board: d1_mini)
--------------------------------------------------------------------------------
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/fan
Processing fan (framework: arduino; platform: espressif8266@1.8.0; board: d1_mini)
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/fan/.pioenvs/fan/src/esphome/components/api/basic_messages.cpp.o
Compiling /data/fan/.pioenvs/fan/src/esphome/components/api/list_entities.cpp.o
Compiling /data/fan/.pioenvs/fan/src/esphome/components/api/service_call_message.cpp.o
Compiling /data/fan/.pioenvs/fan/src/esphome/components/api/subscribe_logs.cpp.o
Compiling /data/fan/.pioenvs/fan/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/fan/.pioenvs/fan/src/esphome/components/api/user_services.cpp.o
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/fan/.pioenvs/fan/src/esphome/components/api/user_services.cpp.o] Error 4
Compiling /data/fan/.pioenvs/fan/src/esphome/components/api/util.cpp.o
========================== [ERROR] Took 12.88 seconds ==========================

glmnet, 2019-10-10:
You might also want to increase the swap file size which by default is only 100MB. 

Follow [this](https://wpitchoune.net/tricks/raspberry_pi3_increase_swap_size.html)

LastedGuide, 2019-10-10:
I have the same problem on a Raspberry PI 3 32 bit running hassio on Rasbian with a Wemos d1 mini

Wojcioo, 2019-10-11:
> 
> You might also want to increase the swap file size which by default is only 100MB.
> 

I've changed it to CONF_SWAPSIZE=2000 - it doesn't help.

patrul, 2019-10-11:
> You might also want to increase the swap file size which by default is only 100MB.
> 
> Follow [this](https://wpitchoune.net/tricks/raspberry_pi3_increase_swap_size.html)

Thanks for your advise, it is recommended to adjust to current memory size

It does not work for me. Anyway, I don't know where the problem is, but it doesn't seem related to config file, it is unpredictable. 

After Clean Built, I compiled my ESPHome config file for a Sonoff Basic

```
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/api_message.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/api_server.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/basic_messages.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/command_messages.cpp.o
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/basic_messages.cpp.o] Error 4
========================== [ERROR] Took 13.44 seconds ==========================

```
After that, I compiled again with same result but the error points to different file

```
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/basic_messages.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/list_entities.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/service_call_message.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/subscribe_logs.cpp.o
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/list_entities.cpp.o] Error 4
========================== [ERROR] Took 5.15 seconds ==========================
```

Third try ... I guess if I continue trying different files would be pointed out

```
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/list_entities.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/user_services.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/api/util.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/json/json_util.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/logger/logger.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/restart/restart_switch.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/sensor/automation.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/sensor/filter.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/sensor/sensor.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/status_led/status_led.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/switch/automation.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/switch/switch.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/time/automation.cpp.o
Compiling /data/dormitorio/.pioenvs/dormitorio/src/esphome/components/time/real_time_clock.cpp.o
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/dormitorio/.pioenvs/dormitorio/src/esphome/components/text_sensor/text_sensor.cpp.o] Error 4
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/dormitorio/.pioenvs/dormitorio/src/esphome/components/time/automation.cpp.o] Error 4
========================== [ERROR] Took 15.70 seconds ==========================
```

KarlosHit, 2019-10-12:
The same thing happens to me, I always get the same error, but at different points in the process, it seems to affect different files, but it is really the compilation process that is failing.
I have already done my best. With ESP32 it works fine, but the compilation process for ESP01 does not work.
Honestly, I've been like this for a week, but if it doesn't work, I'll have to resort to other firmware.

[image]

patrul, 2019-10-12:
I’m thinking on building a virtual machine with ESPHome to compile and upload through sonoff device’s web interface

Wojcioo, 2019-10-13:
> building a virtual machine with ESPHome

I've build oracle VM with hassio, and it doesn't work for me:

collect2: error: ld returned 1 exit status
*** [/data/sonoff4ch/.pioenvs/sonoff4ch/firmware.elf] Error 1

Wojcioo, 2019-10-13:
> 
> 
> I’m thinking on building a virtual machine with ESPHome to compile before upload through sonoff device’s web interface

I've build oracle VM with hassio, and it doesn't work for me:

collect2: error: ld returned 1 exit status
*** [/data/sonoff4ch/.pioenvs/sonoff4ch/firmware.elf] Error 1

patrul, 2019-10-13:
Thanks for letting us know

If thar is the case, I don’t know how to proceed

De: Wojcioo
Enviado: domingo, 13 de octubre de 2019 7:40
Para: esphome/issues
CC: patrul; Comment
Asunto: Re: [esphome/issues] Compiler segmentation fault (#677)

I’m thinking on building a virtual machine with ESPHome to compile before upload through sonoff device’s web interface
I've build oracle VM with hassio, and it doesn't work for me:
collect2: error: ld returned 1 exit status
*** [/data/sonoff4ch/.pioenvs/sonoff4ch/firmware.elf] Error 1

KarlosHit, 2019-10-13:
I am losing hope. I can complete an ESP32, but always the same error in ESP01. I've tried everything you recommended, but it doesn't work.
In my previous Raspi card, it worked for me, but I had to reinstall the entire system, and from there, always the same error: **_"xtensa-lx106-elf-g ++: internal compiler error: Segmentation fault (program cc1plus)"._**

It is a very specific error, it is always repeated, and only with ESP01, it has to have a clear cause.

This is my configuration:
- ESPHome running in Rasperry 3 B+ with Raspbian.
- PlatformIO version 4.0.3
- g++ version: 8.3.0

I have do this to solve an TypeError: __init__() error:
- sudo pip2 install voluptuous==0.11.5.

My code in ESPHome:
[image]

My g++ version:
[image]

My error:
[image]

Thanks for your help!!!

KarlosHit, 2019-10-13:
I am losing hope. I can complete an ESP32, but always the same error in ESP01. I've tried everything you recommended, but it doesn't work.
In my previous Raspi card, it worked for me, but I had to reinstall the entire system, and from there, always the same error: "xtensa-lx106-elf-g ++: internal compiler error: Segmentation fault (program cc1plus)".

It is a very specific error, it is always repeated, and only with ESP01, it has to have a clear cause.

This is my configuration:

ESPHome running in Rasperry 3 B+ with Raspbian.
PlatformIO version 4.0.3
g++ version: 8.3.0
I have do this to solve an TypeError: init() error:

sudo pip2 install voluptuous==0.11.5.

My code in ESPHome:
[image]

My g++ version:
[image]

My error:
[image]

Thanks for your help!!!!

KarlosHit, 2019-10-13:
I am losing hope. I can complete an ESP32, but always the same error in ESP01. I've tried everything you recommended, but it doesn't work.
In my previous Raspi card, it worked for me, but I had to reinstall the entire system, and from there, always the same error:

"**_xtensa-lx106-elf-g ++: internal compiler error: Segmentation fault (program cc1plus)_**".

It is a very specific error, it is always repeated, and only with ESP01, it has to have a clear cause.

**This is my configuration:**
- ESPHome running in Rasperry 3 B+ with Raspbian.
- PlatformIO version 4.0.3
- g++ version: 8.3.0

**I have do this to solve an TypeError: init() error:**
- sudo pip2 install voluptuous==0.11.5.

**My code in ESPHome:**
[image]

**My g++ version:**
[image]

**My error:**
[image]

Thanks for your help!!!!

glmnet, 2019-10-13:
@wojcioo you have a different issue which is solvable. Check pinned issues in this repository.

bcameron21, 2019-10-13:
Also experiencing this same error.
**"xtensa-lx106-elf-g ++: internal compiler error: Segmentation fault (program cc1plus)"**
I am using Hassio running on Raspbian, on a Raspberry Pi 4.
Successfully compiled for an ESP32. 
D1 Mini is giving me this error using only basic config.

KarlosHit, 2019-10-13:
I see that some people with the problem have Raspi 4. Since my problem started when I completely reinstalled Raspbian, could it be related to an update? Maybe going back to update an item would be fixed?
I don't know ... it's about contributing something.
Thank you!!!

patrul, 2019-10-13:
Everything works if I go back to Ubuntu VM on NUC8.

I re-installed addon on RPI4 and nothing changes

OttoWinter, 2019-10-14:
Yeah so this sounds a lot like it's a resources problem - In general RPi's are probably right on the edge of what can work. If you've got something else consuming memory running the background you will certainly face this issue.

RPi 4s (especially the larger memory ones) should work better, but as @bcameron21 mentioned there it could also have to do with firmware versions.

Anyway, for anybody facing this ("segmentation fault", not any other ld errors):

 1. Make sure you don't have any large applications running in the background

 2. Increase the swap size on your device (see also google)

patrul, 2019-10-14:
I do not have any high demanding app, letting 3GB of free memory for my RPI4

Swap file was changed to 2GB but it is 0% used because of the fact above.

S-Przybylski, 2019-10-14:
Dear @OttoWinter 
I just reinstalled a Raspi 4 with 4 GBytes (Raspian buster with latest patches) and started the Docker container of esphome project on that without any additional application. I ran into the same problem. 
Its interesting that when i delete the whole directory structure besides the yaml file, the segment fault comes later. If i retry the built, then the error raises after the first entries. When i retry that several times, the location of the error is always different! But you never get finished the compilation.

btw: If you update the docker image, could you please also update the arm version too ?

KarlosHit, 2019-10-14:
Hello, I explained before what my configuration is, but I summarize again: Raspi 3 B +, with Home Assistant and an ESPHome server running on it. **No other application, except mosquitto, and two crontab for backups and certificate updates**.

Following your advice, I expanded the Swap from 100 Mbyte to 1,000 MByte. I started again, and nothing, the same error.

The conclusion would be that ESPHome does not run on Raspberry Pi?
However, **with my previous card with Rasbpian Buster**, yes, it worked!!!

I see what happens to more people ... what is happening then?

Thank you.

[image]

[image]

**The same result:**
[image]

KarlosHit, 2019-10-14:
Could you install ESPHome on a PC with Ubunty and flash from there, and have Home Asistant running on RPI?

Or maybe it is necessary for ESPHome to run on the same server as Home Assistant?

Thank you.

KarlosHit, 2019-10-14:
Could I install ESPHome on a PC with Ubuntu and flash ESP01 from there, and have Home Asistant running on RPI?

Or maybe it is necessary for Home Assistant to run on the same server as ESPHome?

Thank you.

OttoWinter, 2019-10-15:
> Or maybe it is necessary for Home Assistant to run on the same server as ESPHome?

It is not.

As for why it still doesn't work with the swap, that does sound weird. Maybe there's a second factor here (potentially a xtensa toolchain update that platformio pushed?). Either way, I don't think we will be able to figure out what exactly it is - I haven't debugged a compiler yet and I don't think it will be easy.

Does this only happen for ESP8266 projects, or also ESP32 ones. If it's limited to ESP8266 it would point to a toolchain update.

dmitrybabeshko, 2019-10-15:
With 8266 only

glmnet, 2019-10-15:
All screenshots seems to be from people trying to compile ESPAsyncTCP with both 1.2.0 and 1.2.2 at the same time, may be this is another manifestation of the same #683 issue, but compiler failing more ugly with this error.

Can you people try [this fix](https://github.com/esphome/issues/issues/683#issuecomment-541353954)

bcameron21, 2019-10-16:
I tried the fix. It doesn't fix the issue for me. I have no "webserver" in my code.

I can confirm that ESP32 works. ESP8266 is where the error messages come.

This is my full log:

INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/d1mini1
Processing d1mini1 (framework: arduino; platform: espressif8266@1.8.0; board: d1_mini)
--------------------------------------------------------------------------------
LibraryManager: Installing id=305 @ 1.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Compiling /data/d1mini1/.pioenvs/d1mini1/src/esphome/components/api/api_message.cpp.o
Compiling /data/d1mini1/.pioenvs/d1mini1/src/esphome/components/api/api_server.cpp.o
Compiling /data/d1mini1/.pioenvs/d1mini1/src/esphome/components/api/basic_messages.cpp.o
Compiling /data/d1mini1/.pioenvs/d1mini1/src/esphome/components/api/command_messages.cpp.o
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/d1mini1/.pioenvs/d1mini1/src/esphome/components/api/api_message.cpp.o] Error 4
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/d1mini1/.pioenvs/d1mini1/src/esphome/components/api/api_server.cpp.o] Error 4
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/d1mini1/.pioenvs/d1mini1/src/esphome/components/api/basic_messages.cpp.o] Error 4
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/d1mini1/.pioenvs/d1mini1/src/esphome/components/api/command_messages.cpp.o] Error 4
========================== [ERROR] Took 7.87 seconds ==========================

bcameron21, 2019-10-16:
Update: 

I kept trying to Compile, and noticed that the error occured at a different point each time. 

So I ran compile multiple times and it worked further through the code each time. With each subsequent compile the log started from the point the error occurred, and continued. 

After running Compile about 30-40 times it eventually completed successfully.

patrul, 2019-10-16:
What happens if you try to compile again after this “forced” first compile?

(Maybe you may need to change something in the code to let the program know there are changes and consider new code)

[22 more replies]

closed 2019-10-22 by OttoWinter