esphome/issues #679: "Unexpected keyword" on windows

simonszu, 2019-09-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10, Python 2.7.16, pip 19.2.3

**ESP (ESP32/ESP8266, Board/Sonoff):**
Wemos D1 Mini with DHT22 shield

**Affected component:**

Core - https://esphome.io/components/esphome.html

**Description of problem:**
``` esphome sketch.yaml run``` creates an exception. This is the stack trace: 

```
Z:\Datengrab\Elektronik\ESP8266-Sketches\esphomelib>esphome schlafzimmerklima.yaml run
Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Python27\Scripts\esphome.exe\__main__.py", line 5, in <module>
  File "c:\python27\lib\site-packages\esphome\__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "c:\python27\lib\site-packages\esphome\writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "c:\python27\lib\site-packages\esphome\config.py", line 15, in <module>
    from esphome import core, core_config, yaml_util
  File "c:\python27\lib\site-packages\esphome\core_config.py", line 6, in <module>
    import esphome.config_validation as cv
  File "c:\python27\lib\site-packages\esphome\config_validation.py", line 1242, in <module>
    Optional(CONF_COMMAND_TOPIC): All(requires_component('mqtt'), subscribe_topic),
  File "c:\python27\lib\site-packages\esphome\voluptuous_schema.py", line 198, in extend
    ret = super(_Schema, self).extend(schema, extra=extra)
  File "c:\python27\lib\site-packages\voluptuous\schema_builder.py", line 782, in extend
    return result_cls(result, required=result_required, extra=result_extra)
TypeError: __init__() got an unexpected keyword argument 'required'
```

**Problem-relevant YAML-configuration entries:**
```
esphomeyaml:
  name: schlafzimmerklima
  platform: ESP8266
  board: d1_mini
```

**Logs (if applicable):**
See stack trace

**Additional information and things you've tried:**

glmnet, 2019-09-23:
I can't reproduce this, is that all your yaml? whats the purpose?
Seems like something is not installed properly, can you run `pip install esphome` and add output here?

DSuds, 2019-09-23:
try rolling back voluptuous.

`pip  install voluptuous==0.11.5`

Valcob, 2019-09-25:
Hey as it's says [Here](https://github.com/esphome/issues/issues/666)  just install the dev branch as master is pretty much F'uped
`pip install https://github.com/esphome/esphome/archive/dev.zip`

Hope it helps, and we can close this one as a duplicate.

brandond, 2019-09-25:
duplicate of #666

closed 2019-09-25 by brandond