esphome/issues #681: Safe Mode Fails to start if an SSID is set

alexyao2015, 2019-09-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCUv2 and Tuya bulb
**Affected component:**

OTA Safe Mode

**Description of problem:**
Safe mode unable to connect to network or setup AP if an SSID is set in the YAML. Loops with starting scan... and scan timeout!. To recreate this without a bad YAML, I used a good YAML that boots and pressed the rst button 10 times to reach safe mode. From there, it should connect to wifi or setup an AP to allow an OTA but it fails to do so. 

I originally flashed a bad YAML by setting a switch on one of the flash pins which caused it to bootloop. Same issues there.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: test
  platform: ESP8266
  board: nodemcuv2
    
wifi:
  ssid: "redacted"
  password: "redacted"
  
  ap:
   ssid: "ESPhome test"
   
ota:

# Enable logging
logger:
  level: DEBUG

web_server:
  port: 80

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[1B][0;35m[C][ota:364]: There have been 10 suspected unsuccessful boot attempts.[1B][0m
[1B][1;31m[E][ota:369]: Boot loop detected. Proceeding to safe mode.[1B][0m
[1B][0;35m[C][wifi:029]: Setting up WiFi...[1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][1;31m[E][wifi:280]: Scan timeout![1B][0m
[1B][0;35m[C][wifi:138]: Setting up AP...[1B][0m
[1B][0;35m[C][wifi:140]:   AP SSID: 'ESPhome test'[1B][0m
[1B][0;35m[C][wifi:141]:   AP Password: ''[1B][0m
[1B][0;35m[C][wifi:150]:   IP Address: 192.168.4.1[1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][1;31m[E][wifi:280]: Scan timeout![1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][1;31m[E][wifi:280]: Scan timeout![1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][1;31m[E][wifi:280]: Scan timeout![1B][0m
[1B][0;33m[W][wifi:431]: Restarting WiFi adapter...[1B][0m
[1B][0;33m[W][wifi:431]: Restarting WiFi adapter...[1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][1;31m[E][wifi:280]: Scan timeout![1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][1;31m[E][wifi:280]: Scan timeout![1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][1;31m[E][wifi:280]: Scan timeout![1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][1;31m[E][wifi:280]: Scan timeout![1B][0m
[1B][0;33m[W][wifi:431]: Restarting WiFi adapter...[1B][0m
[1B][0;33m[W][wifi:431]: Restarting WiFi adapter...[1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][1;31m[E][wifi:280]: Scan timeout![1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][1;31m[E][wifi:280]: Scan timeout![1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
```

**Additional information and things you've tried:**
Safe mode is effectively rendered useless if there is an SSID set in the config.

prabbit237, 2019-10-19:
Likewise having the same issue.
````
esphome:
  name: test1
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
   - ssid: "SSID1"
     password: "**********"
   - ssid: "SSID2"
     password: "**********"
   - ssid: "SSID3"
     password: ""

# Enable logging
logger:

# Enable Home Assistant API
api:
ota:
````

Upload and then reset 10 times:
````
[D][wifi:323]:     Channel: 11
[D][wifi:324]:     RSSI: -69 dB
[I][wifi:322]: - 'NETGEAR-NEW' (10:DA:43:90:2F:36) ▂▄▆█
[D][wifi:323]:     Channel: 6
[D][wifi:324]:     RSSI: -73 dB
[D][wifi:326]: - '' (B0:A7:37:B8:56:C1) ▂▄▆█
[I][wifi:164]: WiFi Connecting to 'OpenWrt'...
sd$▒▒|▒l▒<
▒l▒c|▒▒▒▒;▒c▒c▒▒g'▒log▒▒▒
cx▒▒l;l{lx▒n▒
▒
d
▒▒
cg▒|l▒
▒c▒▒'o▒d▒▒$`▒o$`'{▒▒▒ob▒lr▒▒'
c
▒l▒c#c
▒▒▒▒$`▒'▒[I][logger:116]: Log initialized
[C][ota:364]: There have been 10 suspected unsuccessful boot attempts.
[E][ota:369]: Boot loop detected. Proceeding to safe mode.
[C][wifi:029]: Setting up WiFi...
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[W][wifi:431]: Restarting WiFi adapter...
[W][wifi:431]: Restarting WiFi adapter...
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[W][wifi:431]: Restarting WiFi adapter...
[W][wifi:431]: Restarting WiFi adapter...
[D][wifi:272]: Starting scan...
[E][wifi:104]: Can't connect to WiFi, rebooting...
[I][app:123]: Forcing a reboot...

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
   [I][logger:116]: Log initialized
[C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[I][app:028]: Running through setup()...
[C][wifi:029]: Setting up WiFi...
[D][wifi:272]: Starting scan...
[D][wifi:287]: Found networks:
[I][wifi:322]: - 'SSID1' (7A:D2:94:56:B1:F1) ▂▄▆█
[D][wifi:323]:     Channel: 11
[D][wifi:324]:     RSSI: -53 dB
[I][wifi:322]: - 'SSSID2' (78:D2:94:56:B1:F1) ▂▄▆█
[D][wifi:323]:     Channel: 11
[D][wifi:324]:     RSSI: -54 dB
[I][wifi:322]: - 'SSID3' (10:DA:43:90:2F:36) ▂▄▆█
[D][wifi:323]:     Channel: 6
[D][wifi:324]:     RSSI: -58 dB
[D][wifi:326]: - '' (B0:A7:37:B8:56:C1) ▂▄▆█
[I][wifi:164]: WiFi Connecting to 'SSID1'...
[I][wifi:380]: WiFi connected!
[C][wifi:254]:   SSID: 'SSID1'
[C][wifi:255]:   IP Address: 192.168.1.20
[C][wifi:257]:   BSSID: 7A:D2:94:56:B1:F1
[C][wifi:258]:   Hostname: 'test1'
[C][wifi:262]:   Signal strength: -60 dB ▂▄▆█
[C][wifi:263]:   Channel: 11
[C][wifi:264]:   Subnet: 255.255.255.0
[C][wifi:265]:   Gateway: 192.168.1.1
[C][wifi:266]:   DNS1: 192.168.1.1
[C][wifi:267]:   DNS2: 0.0.0.0
[C][ota:029]: Over-The-Air Updates:
[C][ota:030]:   Address: test1.local:8266
[C][api:030]: Setting up Home Assistant API server...
[I][app:060]: setup() finished successfully!
[I][app:096]: esphome version 1.13.6 compiled on Oct 18 2019, 23:19:23
[C][wifi:372]: WiFi:
[C][wifi:254]:   SSID: 'SSID1'
[C][wifi:255]:   IP Address: 192.168.1.20
[C][wifi:257]:   BSSID: 7A:D2:94:56:B1:F1
[C][wifi:258]:   Hostname: 'test1'
[C][wifi:262]:   Signal strength: -60 dB ▂▄▆█
[C][wifi:263]:   Channel: 11
[C][wifi:264]:   Subnet: 255.255.255.0
[C][wifi:265]:   Gateway: 192.168.1.1
[C][wifi:266]:   DNS1: 192.168.1.1
[C][wifi:267]:   DNS2: 0.0.0.0
[C][logger:137]: Logger:
[C][logger:138]:   Level: DEBUG
[C][logger:139]:   Log Baud Rate: 115200
[C][logger:140]:   Hardware UART: UART0
[C][ota:029]: Over-The-Air Updates:
[C][ota:030]:   Address: test1.local:8266
[C][api:103]: API Server:
[C][api:104]:   Address: test1.local:6053
[E][api:092]: No client connected to API. Rebooting...
[I][app:123]: Forcing a reboot...
[W][wifi_esp8266:354]: Event: Disconnected ssid='SSID1' bssid=7A:D2:94:56:B1:F1 reason='Association Leave'
````

It also never goes to safe mode when it doesn't get a connection within 5 mins.

OttoWinter, 2019-10-20:
yeah I've seen this before. These are both instances of https://github.com/esphome/issues/issues/455 - closing...

closed 2019-10-20 by OttoWinter

alexyao2015, 2019-10-20:
This seems to be different from that issue since it actually does connect normally, just doesn't connect when safe mode is activated. Potentially a fix is to ignore the ssid set when safe mode is activated and directly activate the ap?

OttoWinter, 2019-10-20:
See these messages:

```text
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
[D][wifi:272]: Starting scan...
[E][wifi:280]: Scan timeout!
```

that fits into #455 (all ESP8266 connection related issues are combined there)

I understand it is working in non-safe mode. But that's probably because a device restart is sometimes causing issues (as some have noted in #455). The OTA safe mode is pretty much the same as normal mode, with the one difference that most user-facing stuff (sensors, switches etc) aren't loaded. The WiFi code is the same with safe mode and normal mode.

alexyao2015, 2019-10-20:
Is there a chance that you would be able to strip ssid and networks from the config during safe mode? 

It's strange that it would work during regular mode but not during safe mode if the wifi config remains the same. Something has to be different.