esphome/issues #683: ESPasynctcp compile error

gordonthree, 2019-09-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip install on Linux, Fedora 30

**ESP (ESP32/ESP8266, Board/Sonoff):**

Boards used; Generic ESP8266 12e and Wemos D1 Mini

**Affected component:**

ESP have nothing exciting attached, external Dallas temperature probe, transistors for controlling external relays and motors

**Description of problem:**

https://esphome.io/components/mqtt.html

Project fails to compile, appears to be downloading multiple versions of the espasynctcp library.

Removing mqtt allows project to compile, but project is useless without mqtt

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bedroom1
  platform: ESP8266
  board: esp12e

wifi:
  ssid: "Tell my WiFi I love her"
  password: "xxxxxxxxxxxxx"

# Enable logging
logger:
  level: info

# Enable Home Assistant API
api:

ota:

mqtt:
  broker: 192.168.xxx.xxx
  topic_prefix: home/bedroom1
  reboot_timeout: 300s
  
output:
  - platform: gpio
    pin: 2
    inverted: True
    id: gpio_2

  - platform: gpio
    pin: 12
    inverted: True
    id: gpio_12

  - platform: gpio
    pin: 14
    inverted: True
    id: gpio_14

switch:
  - platform: gpio
    pin: 5
    name: "Bedroom1 DOW Power"
    restore_mode: ALWAYS_ON

light:
  - platform: binary
    name: "Bedroom1 ID"
    id: idled
[... 661 lines cut ...]
Compiling .pioenvs/bedroom1/FrameworkArduino/sntp-lwip2.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs/spiffs_cache.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs/spiffs_check.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs/spiffs_gc.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs/spiffs_hydrogen.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs/spiffs_nucleus.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs_api.cpp.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs_hal.cpp.o
Compiling .pioenvs/bedroom1/FrameworkArduino/time.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/uart.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/umm_malloc/umm_malloc.c.o
Archiving .pioenvs/bedroom1/libFrameworkArduino.a
Indexing .pioenvs/bedroom1/libFrameworkArduino.a
Linking .pioenvs/bedroom1/firmware.elf
Retrieving maximum program size .pioenvs/bedroom1/firmware.elf
Checking size .pioenvs/bedroom1/firmware.elf
Building .pioenvs/bedroom1/firmware.bin
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  39.6% (used 32472 bytes from 81920 bytes)
PROGRAM: [===       ]  33.0% (used 344656 bytes from 1044464 bytes)
============================================= [SUCCESS] Took 5.80 seconds =============================================
INFO Successfully compiled program.
[gordon@taichi esphome]$

```

brandond, 2019-09-23:
espasynctcp 1.2.1 seems to contain some breaking changes. Anything that uses it will cause this error.

gordonthree, 2019-09-23:
@brandond 

thank you for the suggestions :)

tried removing the entire project folder and letting it rebuild using the yaml, still getting both versions to espasynctcp and the compile fails

it appears platformio.ini is already asking for ESPAsyncTCP@1.2.0, I added it under common just to see ... here is my file:

```
[gordon@taichi bedroom1]$ cat platformio.ini
; Auto generated code by esphome

[common]
lib_deps =
    ESPAsyncTCP@1.2.0
build_flags =
upload_flags =

; ===== DO NOT EDIT ANYTHING BELOW THIS LINE =====
; ========== AUTO GENERATED CODE BEGIN ===========
[env:bedroom1]
board = esp12e
board_build.flash_mode = dout
build_flags =
    -DESPHOME_LOG_LEVEL=ESPHOME_LOG_LEVEL_INFO
    -DUSE_STORE_LOG_STR_IN_FLASH
    -Wl,-Teagle.flash.4m.ld
    -Wno-sign-compare
    -Wno-unused-but-set-variable
    -Wno-unused-variable
    -fno-exceptions
    ${common.build_flags}
framework = arduino
lib_deps =
    ArduinoJson-esphomelib@5.13.3
    AsyncMqttClient@0.8.2
    ESP8266WiFi
    ESP8266mDNS
    ESPAsyncTCP@1.2.0
    Update
    ${common.lib_deps}
platform = espressif8266@1.8.0
upload_speed = 115200
; =========== AUTO GENERATED CODE END ============
; ========= YOU CAN EDIT AFTER THIS LINE =========

```

alexconrey, 2019-09-23:
I'm also able to confirm this. I had added my output to (apparently) the wrong issue on GH, but I can confirm ESPAsyncTCP is breaking compilation when `web_server` or `mqtt` is included. version locking the lib in `platformio.ini` was also not fixing the issue. My output observed both 1.2.0 and 1.2.1 versions being included for compilation with that file change, which leads me to believe some alternate lib that's being included is pulling the `latest` version of ESPAsyncTCP.

brandond, 2019-09-24:
That's odd. I just cleaned mine here and I can successfully compile with both web_server and mqtt. What version of platformio are you using? 

```
(python27) brandond@host:~/nodes$ platformio --version
PlatformIO, version 4.0.3
```

gordonthree, 2019-09-24:
/shrug ... just one of those things! I was building these projects fine just a few days ago, last build was Sept 14. 

```
[gordon@taichi ~]$ pio --version
PlatformIO, version 4.0.3
```
esphome version gives me an error message "too few arguments", not sure how to check where I'm at, other than pip says I'm up to date.

```
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.1
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.1
|   |-- <ESP8266WiFi> 1.0
```

@brandond  can you compile my yaml from the op?

codedmind, 2019-09-24:
The issue is happen with mqtt and webserver, both use old espasynctcp (1.2.0) and other 1.2.1

alexyao2015, 2019-09-24:
I previously had this issue but it seems like it was fixed and its not happening anymore.

gordonthree, 2019-09-24:
still broken for me

```
$ pip install --upgrade --user esphome
```
says I'm up to date

```
$ esphome bedroom1.yaml compile
(snip)
.pioenvs/bedroom1/lib35f/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x8): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/bedroom1/firmware.elf] Error 1
```

How do I check what version of esphome I have installed?

gordonthree, 2019-09-24:
Just for sanity sake, I created a new virtual machine with Fedora, and installed esphome following the documentation. Created a brand new project using mqtt. Still failing to compile. Same ESPasynctcp errors

What am I doing wrong?

alexyao2015, 2019-09-24:
I use hassio so it might be different, but I've had no problems with the hassio docker.

alexyao2015, 2019-09-24:
I take everything back. Just cleared builds and it seems to be failing to compile again. I do notice that when it fails to compile its showing `ESPAsyncTCP@1.2.1` instead of `ESPAsyncTCP@1.2.0` which is when it compiles fine.

thoemy, 2019-09-24:
It's because `ESPAsyncTCP@1.2.0` is hardcoded in the [api component](https://github.com/esphome/esphome/blob/master/esphome/components/api/__init__.py#L77) and the ESP Async WebServer v1.1.1 always uses the most recent ESPAsyncTCP version (see [library.json](https://github.com/me-no-dev/ESPAsyncWebServer/blob/6edc4310348a54d14c44d2889dcf42e7ef2c0100/library.json)). 3 days ago `ESPAsyncTCP@1.2.1` was released, which now causes a conflict because both versions are compiled. This can be seen in the various dependency graphs that were posted here.

You can fix it by editing the version to be `1.2.1` in your local `esphome/components/api/__init__.py` file.

gordonthree, 2019-09-24:
@thoemy thank you! that did the trick.

alexyao2015, 2019-09-24:
@thoemy 
I can confirm that this works! For those using hassio/docker, the file is `/opt/esphome/esphome/components/api/__init__.py`. I guess a potential fix is to update the __init__.py to use the 1.2.1 or figure out how to use an older version of ESPAsyncTCP.

For the record, it is the api component so if api: is in the yaml, nothing compiles.

alexconrey, 2019-09-25:
I am also able to confirm that's a working fix. I opened up a PR, we'll see if any changes are required or if it's good to go :)

Anonym-tsk, 2019-09-25:
Can't reproduce in dev branch.

alexconrey, 2019-09-25:
@Anonym-tsk are you able to show the dependency tree output from your build on the `dev` branch? i'm curious what the `dev` branch is requiring for library versions.

Anonym-tsk, 2019-09-25:
With config from first post:
```
Dependency Graph
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
```

It seems that [this commit](https://github.com/esphome/esphome/commit/2712c440041cb697af50168fa1f8266d16393fc6) fixed the issue.

brandond, 2019-09-25:
That might be why it's working for me. I guess I assume everyone tries the dev branch first when they have an issue.

glmnet, 2019-09-25:
I didn't expect people patching 1.13, however I did reproduce this issue with dev branch 
@Anonym-tsk 
https://github.com/esphome/issues/issues/677#issuecomment-533941690

AquaMCU, 2019-09-25:
Same issue here. I am using the docker image on my pi : esphome/esphome-armv7

when you guys hard code the next version into the code, will we not have the same issue when the lib is released the next time?

alexconrey, 2019-09-25:
@AquaMCU seems like as long as they have some sort of specification requiring global lib version locking instead of allowing components to control library dependencies, this wont be an issue going forward. my understanding is that one component was using a pinned ESPAsyncTCP library version and another component was using the `latest` ESPAsyncTCP lib version. When `latest` version deviated from the pinned version, both were pulled in resulting in the duplicate errors I/others observed.

Valcob, 2019-09-25:
As a workaround I can suggest the following simple steps:
1. From ... menu select [Clean Build] to remove everything in .pioenv folder
[image]
2. Click the Edit button to change the configuration and comment out the web_server/mqtt block(s)
[image]
3. Click save button and close then go to ... menu and click Compile
[image]
4. The moment u see in the log lines with 
```
 |   |-- <ESPAsyncTCP> 1.2.0 
 |   |   |-- <ESP8266WiFi> 1.0
```
[image]

Feel free to stop the build as you have cached the right version of the lib
[image]

5. Go back to edit and uncomment the web-server/mqtt block(s)
6. Save, Compile, Enjoy

Hope these helps :)

Anonym-tsk, 2019-09-26:
@glmnet @Valcob Compilation successful from dev branch with web_server and mqtt after clean build.

Config:
```yaml
esphome:
  name: bedroom1
  platform: ESP8266
  board: esp12e

wifi:
  ssid: "Tell my WiFi I love her"
  password: "xxxxxxxxxxxxx"

# Enable logging
logger:
  level: info

# Enable Home Assistant API
api:

ota:

mqtt:
  broker: 192.168.100.1
  topic_prefix: home/bedroom1
  reboot_timeout: 300s
  
output:
  - platform: gpio
    pin: 2
    inverted: True
    id: gpio_2

  - platform: gpio
    pin: 12
    inverted: True
    id: gpio_12

  - platform: gpio
    pin: 14
    inverted: True
    id: gpio_14

switch:
  - platform: gpio
    pin: 5
    name: "Bedroom1 DOW Power"
    restore_mode: ALWAYS_ON

light:
  - platform: binary
    name: "Bedroom1 ID"
    id: idled
    output: gpio_2

  - platform: binary
    name: "Bedroom Window Lamp"
    id: windowlamp
    output: gpio_14

fan:
  - platform: binary
    name: "Bedroom Window Fan"
    id: windowfan
    output: gpio_12

dallas:
  - pin: 4
    update_interval: 30s

[... 236 lines cut ...]
Compiling .pioenvs/bedroom1/FrameworkArduino/pgmspace.cpp.o
Compiling .pioenvs/bedroom1/FrameworkArduino/sntp-lwip2.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs/spiffs_cache.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs/spiffs_check.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs/spiffs_gc.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs/spiffs_hydrogen.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs/spiffs_nucleus.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs_api.cpp.o
Compiling .pioenvs/bedroom1/FrameworkArduino/spiffs_hal.cpp.o
Compiling .pioenvs/bedroom1/FrameworkArduino/time.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/uart.c.o
Compiling .pioenvs/bedroom1/FrameworkArduino/umm_malloc/umm_malloc.c.o
Archiving .pioenvs/bedroom1/libFrameworkArduino.a
Linking .pioenvs/bedroom1/firmware.elf
Retrieving maximum program size .pioenvs/bedroom1/firmware.elf
Building .pioenvs/bedroom1/firmware.bin
Checking size .pioenvs/bedroom1/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  49.6% (used 40612 bytes from 81920 bytes)
PROGRAM: [====      ]  42.9% (used 447764 bytes from 1044464 bytes)
================================================================================================================================================ [SUCCESS] Took 67.67 seconds ================================================================================================================================================
INFO Successfully compiled program.
```

Please provide your config.

Valcob, 2019-09-26:
@Anonym-tsk, glad to hear that. BTW nice wifi.ssid ! 10 points ))

lmamakos, 2019-09-27:
I just stumbled across this issue as I'm having the same problem, though on an ESP32 system.  Would have have also been addressed on the dev branch?  

From looking at the `api/__init__.py` library file, it seems that the ESP32 libraries have their own treatment vs. for the ESP8266.

Also, I'm not sure if this is the root cause of the problem; curiously, I don't even have the `api:` component in my ESPHome configuration, just MQTT.  So perhaps something else is dragging in a hard-coded library version?

lmamakos, 2019-09-28:
Here's my failed build

```
louie@kzin[123] $ rm -r .esphome/
louie@kzin[124] $ rm -r /Users/louie/.platformio/
.cache/        appstate.json  lib/           packages/      platforms/
louie@kzin[124] $ rm -r /Users/louie/.platformio/.cache/
louie@kzin[125] $ esphome uselessbox.yaml compile
INFO Reading configuration...
INFO Detected timezone 'EST' with UTC offset -5 and daylight savings time from 03/08/20 02:00:00 to 11/01/20 02:00:00
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting uselessbox/.pioenvs
INFO Deleting uselessbox/.piolibdeps
INFO Compiling app...
INFO Running:  platformio run -d uselessbox
Processing uselessbox (board: nodemcu-32s; framework: arduino; platform: espressif32@1.6.0)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 1.6.0 > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10001.190111 (1.0.1), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for ArduinoJson-esphomelib library in registry
Found: https://platformio.org/lib/show/3837/ArduinoJson-esphomelib
LibraryManager: Installing id=3837 @ 5.13.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
ArduinoJson-esphomelib @ 5.13.3 has been successfully installed!
Looking for AsyncMqttClient library in registry
Found: https://platformio.org/lib/show/346/AsyncMqttClient
LibraryManager: Installing id=346 @ 0.8.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
AsyncMqttClient @ 0.8.2 has been successfully installed!
Installing dependencies
Looking for ESPAsyncTCP library in registry
Found: https://platformio.org/lib/show/305/ESPAsyncTCP
LibraryManager: Installing id=305 @ ^1.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
ESPAsyncTCP @ 1.2.1 has been successfully installed!
Looking for AsyncTCP library in registry
Found: https://platformio.org/lib/show/1826/AsyncTCP
LibraryManager: Installing id=1826 @ ^1.0.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
AsyncTCP @ 1.1.0 has been successfully installed!
Looking for Hash library in registry
Warning! Library `{'requirements': None, 'name': 'Hash'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Hash'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling .pioenvs/uselessbox/src/esphome/components/debug/debug_component.cpp.o
Compiling .pioenvs/uselessbox/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/uselessbox/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/uselessbox/src/esphome/components/mqtt/custom_mqtt_device.cpp.o
Compiling .pioenvs/uselessbox/src/esphome/components/mqtt/mqtt_binary_sensor.cpp.o
[... 61 lines cut ...]
wifi:
  ssid: "xxx"
  password: "yyy"

debug:
ota:

logger:
  
time:
  - platform: sntp
    id: sntp_time
    servers: 10.200.0.1

mqtt:
  broker: "10.200.0.100"
  topic_prefix: "19916/uselessbox"
  discovery: False
  birth_message:
    topic: 19916/uselessbox/lwt
    payload: online
  will_message:
    topic: 19916/uselessbox/lwt
    payload: offline
```

glmnet, 2019-09-30:
@OttoWinter should we make a 13.x patch for this one?

brandond, 2019-09-30:
I've been waiting for Otto to resurface as it seemed like he had some other stuff he was waiting on to do a release. If he's not around we might think about fixing it in dev and then backporting to 1.13.x

I have no idea if the Docker images and hassio metadata are automated to the point where our simply tagging a release on GH will update them, or if there's some stuff he needs to pushed manually on the backend.

glmnet, 2019-09-30:
Actually I think @Anonym-tsk is right and the commit [he points](https://github.com/esphome/issues/issues/683#issuecomment-535087764) is actually fixing it in dev, so we should check if we can apply that commit on master.

Unless somebody can confirm the issue is still happening on dev

glmnet, 2019-09-30:
> automated to the point where our simply tagging a release on GH will update them

I wouldn't be surprised, although I have no idea how that can magically work

glmnet, 2019-10-04:
I did a bit of research, posted results in other [thread](https://github.com/esphome/issues/issues/713#issuecomment-538534166)

glmnet, 2019-10-04:
In order to have a successful compilation you must first compile **from a clean state** without the `web_server` component and with the `api` then later add the `web_server` component without doing a clean.

If you compile first with `web_server` then it won't work if you later add `api`

adukale, 2019-10-12:
> As a workaround I can suggest the following simple steps:
> 
> 1. From ... menu select [Clean Build] to remove everything in .pioenv folder
>    [image]
> 2. Click the Edit button to change the configuration and comment out the web_server/mqtt block(s)
>    [image]
> 3. Click save button and close then go to ... menu and click Compile
>    [image]
> 4. The moment u see in the log lines with
> 
> ```
>  |   |-- <ESPAsyncTCP> 1.2.0 
>  |   |   |-- <ESP8266WiFi> 1.0
> ```
> 
> [image]
> 
> Feel free to stop the build as you have cached the right version of the lib
> [image]
> 
> 1. Go back to edit and uncomment the web-server/mqtt block(s)
> 2. Save, Compile, Enjoy
> 
> Hope this helps :)

Works like a charm ! Thanks.

lmamakos, 2019-10-14:
This workaround is one way forward, however it's a really bad experience for a new user.  I hope that a real fix can be committed soon.  Just for clarity, in my particular case, I wasn't using the API capability at all, just MQTT.

glmnet, 2019-10-16:
@lmamakos you're right, however the way ESPHome works it really brings up this kind of issues, a development environment relying on libraries which can change is the issue, ESPHome can be (and in fact it is) configured to use a specific library version of it's dependencies, however some components like api, or web_server have dependencies which in turn have dependencies which are not pinned to a specific version, so when this components starts bringing up different version of libraries is that this problem happens over and over.
Every time you create a new esphome compilation, platformio will download the latest bits from the internet.
The only solution we know so far is to have ESPHome to rely on no dependencies, however that is a lot of maintenance work overhead.
If somebody has a good idea of how to solve this, we are listening.

lmamakos, 2019-10-16:
I suppose you could pin the integration in ESPHome to specific libraries?  It'd be great to be able to have reproduceable builds.  Especially since you have to compile unique binaries for each device.. this makes it difficult to add a new instance of some device down the road and know that it will behave the same... I've always wished that I could include, e.g., the MAC address as part of the MQTT topic name prefix, so I could just deploy on image over a set of devices..

Regardless, at the moment the situation is that some random new user is going to try to use ESPHome for the first time and have the experience fail.  As I'm not one of the developers on this product, I don't think I can make any strong suggestions on the path forward (other than maybe pinning the libraries versions in all the integrations).  I'm just reporting the broken (new) user experience that requires a convoluted workaround that you have to search github issues to discover.

exxamalte, 2019-10-16:
If there was no way to fully avoid this issue in the future, is there maybe at least a way to clearly detect the issue? For example, as part of the build process, is there maybe a way to compare all the versions of the libraries used, and print out a warning message that gives users (and developers) and hint on where to look for fixing the issue?

OttoWinter, 2019-10-16:
@exxamalte I'm working on solving the issue by just forking all the base libraries required. That will give us the control necessary to prevent this from happening in the future (I may even look into forking arduino framework and platformio package to work around the many issues we've had with those things as well if this forking goes well).

@lmamakos 
> I suppose you could pin the integration in ESPHome to specific libraries?

That is in fact already the case. However, platformio's library dependency system chose to ignore that based on an issue (see also https://github.com/platformio/platformio-core/issues/2115) in platformio.

LastedGuide, 2019-10-16:
My error on hassio on rpi3 buster

`INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esp_temp1
Processing esp_temp1 (framework: arduino; platform: espressif8266@1.8.0; board: d1_mini)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Compiling /data/esp_temp1/.pioenvs/esp_temp1/src/esphome/components/sensor/filter.cpp.o
Compiling /data/esp_temp1/.pioenvs/esp_temp1/src/esphome/components/sensor/sensor.cpp.o
Compiling /data/esp_temp1/.pioenvs/esp_temp1/src/esphome/components/wifi/wifi_component.cpp.o
Compiling /data/esp_temp1/.pioenvs/esp_temp1/src/esphome/components/wifi/wifi_component_esp32.cpp.o
Compiling /data/esp_temp1/.pioenvs/esp_temp1/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling /data/esp_temp1/.pioenvs/esp_temp1/src/esphome/core/application.cpp.o
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/esp_temp1/.pioenvs/esp_temp1/src/esphome/core/application.cpp.o] Error 4
========================== [ERROR] Took 27.15 seconds ==========================`

OttoWinter, 2019-10-17:
@LastedGuide **and to anyone else seeing "xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)": please see https://github.com/esphome/issues/issues/677 instead (external issue, atm it looks like it may have to do with the OS)

marc-gist, 2019-10-23:
I have read through this and I'm very confused as to how to fix this issue :( I have, in an python 3.6 env.  So how do you fix it?

OttoWinter, 2019-10-23:
This issue **has been fixed in the latest beta** (1.14.0b1+) and dev versions. Once 1.14.0 stable is released, it will also be there. Read the FAQ to find out how to update to beta versions.

Alternatively, you can try [this workaround](https://github.com/esphome/issues/issues/683#issuecomment-535222063) by @Valcob.

I will close this issue and lock it to make it easy to find the resolution to this problem.

closed 2019-10-23 by OttoWinter