esphome/issues #684: Branch dev: CT Clamp always return nan Value

Rihan9, 2019-09-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

espHome on hassio.
**ESP (ESP32/ESP8266, Board/Sonoff):**

Esp8266 based board: Wemos D1 mini
**Affected component:**

Ads1115 and CT Clamp

**Description of problem:**
Hi,
i need to use CT Clamp on external adc, because my probe has low sensitivity and it has an offset pretty hight (Vcc/5 when Vcc is 5.0V). So i need to use ads1115 on differential configuration. Follow my entire YAML file.

I try with a 60w load on 220V, but nothing change.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: developer_board
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: ""
  password: ""

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:
  sda: D2
  scl: D1
  scan: True
  id: bus_a
  
ads1115:
  - address: 0x48
    continuous_mode: True
  
sensor:
  - platform: ads1115
    multiplexer: 'A0_A1'
    gain: 4.096
    id: ACS712
    name: "ACS712"
    update_interval: 1s
  - platform: ct_clamp
    sensor: ACS712
    name: "Measured Current"
    update_interval: 10s
    filters:
      - offset: 0.0
      - calibrate_linear:
          # Measured value of 0 maps to 0A
          - 2.50 -> 0
          # Known load: 4.0A
          # Value shown in logs: 0.1333A
          - 2.566 -> 1.0
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```
[23:48:21][D][ads1115:163]: 'ACS712': Got Voltage=-0.000750V
[23:48:21][D][sensor:092]: 'ACS712': Sending state -0.00075 V with 3 decimals of accuracy
[23:48:22][D][ads1115:163]: 'ACS712': Got Voltage=-0.000250V
[23:48:22][D][sensor:092]: 'ACS712': Sending state -0.00025 V with 3 decimals of accuracy
[23:48:23][D][ads1115:163]: 'ACS712': Got Voltage=-0.000500V
[23:48:23][D][sensor:092]: 'ACS712': Sending state -0.00050 V with 3 decimals of accuracy
[23:48:24][D][ads1115:163]: 'ACS712': Got Voltage=-0.000375V
[23:48:24][D][sensor:092]: 'ACS712': Sending state -0.00038 V with 3 decimals of accuracy
[23:48:25][D][ads1115:163]: 'ACS712': Got Voltage=-0.000625V
[23:48:25][D][sensor:092]: 'ACS712': Sending state -0.00062 V with 3 decimals of accuracy
[23:48:26][D][ads1115:163]: 'ACS712': Got Voltage=-0.000125V
[23:48:26][D][sensor:092]: 'ACS712': Sending state -0.00013 V with 3 decimals of accuracy
[23:48:27][D][ads1115:163]: 'ACS712': Got Voltage=-0.000125V
[23:48:27][D][sensor:092]: 'ACS712': Sending state -0.00013 V with 3 decimals of accuracy
[23:48:28][D][ads1115:163]: 'ACS712': Got Voltage=-0.000250V
[23:48:28][D][sensor:092]: 'ACS712': Sending state -0.00025 V with 3 decimals of accuracy
[23:48:29][D][ads1115:163]: 'ACS712': Got Voltage=-0.000125V
[23:48:29][D][sensor:092]: 'ACS712': Sending state -0.00013 V with 3 decimals of accuracy
[23:48:30][D][ads1115:163]: 'ACS712': Got Voltage=-0.000250V
[23:48:30][D][sensor:092]: 'ACS712': Sending state -0.00025 V with 3 decimals of accuracy
[23:48:30][D][sensor:092]: 'Measured Current': Sending state nan A with 2 decimals of accuracy
**Additional information and things you've tried:**

carlosmgr, 2019-09-24:
Yes, same problem here

Rihan9, 2019-10-03:
Any news? Do you need some additional info?

daned33, 2019-10-13:
@Rihan9 can you show your physical setup? photos, wiring, diagrams, etc.

Rihan9, 2019-10-13:
ads1115:
 - A0 To ACS712 data pin (power on Vcc (approximately 5V)
 - A1 to midpoint of a multi-turn trimmer (set at 50% of Vcc)

As you can see from the log, the current reading is consistent.

In the following link a draft on the fly of the circuit
https://crcit.net/c/a7ea52a4eb8d4b2fbbcbccac70ea4fe3

daned33, 2019-10-13:
Are you using a CT Clamp or ACS712 sensor

Rihan9, 2019-10-13:
I'm using an acs712. But the output should be the same, right?

daned33, 2019-10-13:
get rid of the CT clamp stuff if you're not using it and debug using the basics.

Is your ADS1115 working correctly? with A0_GND etc and a stable voltage?

What voltage is the ACS712 giving with a load (using a multimeter)?

The CT Clamp module from what I understand expects a 0-1V signal.

Rihan9, 2019-10-14:
I'm using the CT Clamp component because seems to be the only way to measure an AC voltage (in this case, the output of acs712). 

Later, i try to provide test photos with and without load.

closed 2019-10-14 by OttoWinter