esphome/issues #686: Rotary encoder initial value

Luc3as, 2019-09-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io , ESPHOME DEV 1.13.6
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini
**Affected component:**

Rotary encoder
https://esphome.io/components/sensor/rotary_encoder.html?highlight=rotary

**Description of problem:**
Hi,
I set min_value to rotary encoder component, all is working as supposed however, it would be nice if there were initial_value or simply hardcoded to the min_value, After first start, I can turn rotary encoder to one side and it starts counting from 0 , so I have to get to 20 for example and then I can go lower.

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: rotary_encoder
    name: "Rotary Encoder"
    id: "rotary_encoder1"
    pin_a: D2
    pin_b: D3  
    min_value: 20
    max_value: 400
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-09-25:
This probably belongs in https://github.com/esphome/feature-requests/

I would probably set up an [on_boot](https://esphome.io/components/esphome.html#on-boot) action to set the rotary encoder value to 20 when the device boots.

Luc3as, 2019-09-26:
I tried this, first rotation it looks good but second it resets to zero
again. Some internal counter I think.

OttoWinter, 2019-10-20:
Closing per https://github.com/esphome/esphome/pull/747

closed 2019-10-20 by OttoWinter