esphome/issues #687: voluptuous 0.11.7 breaks something

flowolf, 2019-09-25:
I did a new install of esphome and it broke.

This is what I did:
```
flo@lade ~/Applications/esphome >>> virtualenv -p /usr/bin/python3.7 .
Running virtualenv with interpreter /usr/bin/python3.7
Using base prefix '/usr'
New python executable in /home/flo/Applications/esphome/bin/python3.7
Also creating executable in /home/flo/Applications/esphome/bin/python
Installing setuptools, pip, wheel...
done.
flo@lade ~/Applications/esphome >>> 
flo@lade ~/Applications/esphome >>> 
flo@lade ~/Applications/esphome >>> . ./bin/activate
(esphome) flo@lade ~/Applications/esphome >>> which pip
/home/flo/Applications/esphome/bin/pip
(esphome) flo@lade ~/Applications/esphome >>> which pip3
/home/flo/Applications/esphome/bin/pip3
(esphome) flo@lade ~/Applications/esphome >>> which pip2
/usr/local/bin/pip2
(esphome) flo@lade ~/Applications/esphome >>> pip install esphome 
Collecting esphome
  Using cached https://files.pythonhosted.org/packages/df/bd/a96abf8fb626b21793c42d5750a1503520d33a11fc4947408845fa7a4a24/esphome-1.13.6-py2.py3-none-any.whl
Collecting ifaddr<1,>=0.1.6 (from esphome)
Collecting platformio>=3.6.5 (from esphome)
Collecting PyYAML<6,>=5.1 (from esphome)
Collecting colorlog>=4.0.2 (from esphome)
  Using cached https://files.pythonhosted.org/packages/68/4d/892728b0c14547224f0ac40884e722a3d00cb54e7a146aea0b3186806c9e/colorlog-4.0.2-py2.py3-none-any.whl
Collecting tornado<6,>=5.1.1 (from esphome)
Collecting paho-mqtt<2,>=1.4 (from esphome)
Collecting pyserial<4,>=3.4 (from esphome)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting voluptuous<0.12,>=0.11.5 (from esphome)
Collecting esptool<3,>=2.6 (from esphome)
Collecting tzlocal>=1.5.1 (from esphome)
  Using cached https://files.pythonhosted.org/packages/ef/99/53bd1ac9349262f59c1c421d8fcc2559ae8a5eeffed9202684756b648d33/tzlocal-2.0.0-py2.py3-none-any.whl
Collecting protobuf<3.8,>=3.7 (from esphome)
  Using cached https://files.pythonhosted.org/packages/19/a5/ac51df34cdf4739574492ed4903c11dadd72a7bec4a31bb0496f4f50fc19/protobuf-3.7.1-cp37-cp37m-manylinux1_x86_64.whl
Collecting tabulate<1,>=0.8.3 (from platformio>=3.6.5->esphome)
Collecting colorama (from platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/4f/a6/728666f39bfff1719fc94c481890b2106837da9318031f71a8424b662e12/colorama-0.4.1-py2.py3-none-any.whl
Collecting requests<3,>=2.4.0 (from platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/51/bd/23c926cd341ea6b7dd0b2a00aba99ae0f828be89d72b2190f27c11d4b7fb/requests-2.22.0-py2.py3-none-any.whl
Collecting click<8,>=5 (from platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/fa/37/45185cb5abbc30d7257104c434fe0b07e5a195a6847506c074527aa599ec/Click-7.0-py2.py3-none-any.whl
Collecting bottle<0.13 (from platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/69/d1/efdd0a5584169cdf791d726264089ce5d96846a8978c44ac6e13ae234327/bottle-0.12.17-py3-none-any.whl
Collecting semantic-version<3,>=2.8.1 (from platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/0f/3b/8fee26649a86c71df159ed0ae7ac5f9ac38829bccd8a7404e116f903929b/semantic_version-2.8.2-py2.py3-none-any.whl
Collecting ecdsa (from esptool<3,>=2.6->esphome)
  Using cached https://files.pythonhosted.org/packages/23/a8/8aa68e70959e1287da9154e5164bb8bd5dd7025e41ae54e8d177b8d165c9/ecdsa-0.13.2-py2.py3-none-any.whl
Collecting pyaes (from esptool<3,>=2.6->esphome)
Collecting pytz (from tzlocal>=1.5.1->esphome)
  Using cached https://files.pythonhosted.org/packages/87/76/46d697698a143e05f77bec5a526bf4e56a0be61d63425b68f4ba553b51f2/pytz-2019.2-py2.py3-none-any.whl
Collecting six>=1.9 (from protobuf<3.8,>=3.7->esphome)
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Requirement already satisfied: setuptools in ./lib/python3.7/site-packages (from protobuf<3.8,>=3.7->esphome) (41.2.0)
Collecting idna<2.9,>=2.5 (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl
Collecting certifi>=2017.4.17 (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/18/b0/8146a4f8dd402f60744fa380bc73ca47303cccf8b9190fd16a827281eac2/certifi-2019.9.11-py2.py3-none-any.whl
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/e0/da/55f51ea951e1b7c63a579c09dd7db825bb730ec1fe9c0180fc77bfb31448/urllib3-1.25.6-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
Installing collected packages: ifaddr, tabulate, pyserial, colorama, idna, certifi, urllib3, chardet, requests, click, bottle, semantic-version, platformio, PyYAML, colorlog, tornado, paho-mqtt, voluptuous, ecdsa, pyaes, esptool, pytz, tzlocal, six, protobuf, esphome
Successfully installed PyYAML-5.1.2 bottle-0.12.17 certifi-2019.9.11 chardet-3.0.4 click-7.0 colorama-0.4.1 colorlog-4.0.2 ecdsa-0.13.2 esphome-1.13.6 esptool-2.7 idna-2.8 ifaddr-0.1.6 paho-mqtt-1.4.0 platformio-4.0.3 protobuf-3.7.1 pyaes-1.6.1 pyserial-3.4 pytz-2019.2 requests-2.22.0 semantic-version-2.8.2 six-1.12.0 tabulate-0.8.5 tornado-5.1.1 tzlocal-2.0.0 urllib3-1.25.6 voluptuous-0.11.7
(esphome) flo@lade ~/Applications/esphome >>> esphome --help
Traceback (most recent call last):
  File "/home/flo/Applications/esphome/bin/esphome", line 6, in <module>
[... 129 lines cut ...]
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Ubuntu - pip - python3

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

the comman line tool is unusable

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Valcob, 2019-09-25:
Hey, why u don't use search? 
.... as it says [Here ](https://github.com/esphome/issues/issues/666) just install the dev branch as master is pretty much F'uped
`pip install https://github.com/esphome/esphome/archive/dev.zip`

Hope it helps, and we can close this one as a duplicate.

brandond, 2019-09-25:
duplicate of #666

closed 2019-09-25 by brandond

flowolf, 2019-09-28:
I tend to search open issues, with things that are present in the CURRENT master.

I guess #666 is a fitting issue for this.