esphome/issues #690: Uptime sensor doesn't get triggered when update_interval set to a large value

yeralin, 2019-09-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Raspbian GNU/Linux 9 (stretch)
(Raspberry Pi 3B)

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 (NODEMCUV2 by espressif)

**Affected component:**
Uptime sensor

**Description of problem:**
When I set update interval to some large value (ex. `1h`), and let the ESP run 
autonomously, the automation does not get triggered. If I change the interval to a smaller value (ex. `10min`), everything will work as expected. 

Interestingly enough, this issue does not occur when ESP is connected to Hass instance on my raspberry.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: uptime
     id: uptime_sensor
     update_interval: 1h
     on_value:
       then:
         - logger.log: "Test"
```
For my fool configuration, refer to https://github.com/yeralin/IoT/blob/master/applications/smart-feeder/smart-feeder.yml

**Additional information and things you've tried:**
I don't use SNTP or HomeAssistant time bc I want to ensure that ESP will work even in case when internet is down or raspberry pi is turned off.

glmnet, 2019-10-10:
> Interestingly enough, this issue does not occur when ESP is connected and is broadcasting some values to Hass instance on my raspberry.

ahhh your esp is rebooting due to it cannot connect to ha (you have api defined, by default it reboots every 5 minutes)
so your device uptime never reaches 1 hour.

there i also the wifi's reboot time out which default to 5 minutes too. 
I don't know if your node is loosing wifi connectivity.

closed 2019-10-10 by glmnet

glmnet, 2019-10-10:
by the way, I would use interval instead of uptime sensor for doing periodic tasks.

yeralin, 2019-10-10:
@glmnet `interval` of which component?

glmnet, 2019-10-10:
this [interval](https://esphome.io/guides/automations.html#interval)

yeralin, 2019-10-11:
Hi @glmnet, thank you so much! You were totally right, as soon as I set `api.reboot_timeout=0s`, everything started working perfectly!

As per `interval` instead of `uptime_sensor`. My main issue is that you cannot change interval dynamically, something like:
```
globals:
  - id: some_interval # in hours
    type: int
    restore_value: yes
    initial_value: '5'

interval:
  - interval: $(some_interval) + 'h'
    then:
      - switch.toggle: relay_1
```

So, instead I hit `uptime_sensor` every hour, and then in condition I convert `$(uptime_sensor).state #seconds` to hours and then check for `$(uptime_sensor_in_hours) % $(some_interval) == 0`. 

That's how I mimic _dynamic interval_. If I try to replace `uptime_sensor` with `interval`, then I cannot retrieve `$(interval).state`.

The best that I can come up with is
```
sensor:
 - platform: uptime
    id: uptime_sensor

interval:
  - interval: 1h
    then:
      - if:
          condition:
            - lambda: |-
                int uptime_sensor_in_hours= id(uptime_sensor).state/3600 + 0.5;
                return (uptime_sensor_in_hours % id(some_interval)) == 0;
```

So, essentially I still rely on `uptime_sensor`. 
Is that fine? Is there a better way that you might know?
Why is it better to rely on `interval` instead of `uptime_sensor.update_interval`?

Sorry for this extra questions :)

glmnet, 2019-10-11:
It's just less hacky, so less potential issues in future changes, but if it works for you its ok then.

I don't understand what's your real goal, may be then I will have a better idea.

yeralin, 2019-10-11:
Here is my use case:
I have a pet feeder running that runs on `esphome`. It is supposed to execute `script.feed` every `feeder_interval: int` hours. `feeder_interval` is a global var that can be changed through hass.

So, in short my set up some thing similar to:
```
global:
  - id: feeder_interval # in hours
    type: int
    restore_value: yes
    initial_value: '5'

script:
 - id: feed
   then:
     ...

api:
  reboot_timeout: 0s
  services:
    - service: change_feed_interval
      variables:
        new_feed_interval: int
      then:
        - lambda: |-
            id(feeder_interval) = new_feed_interval;
```

The tricky part is that the pet feeder is supposed to work autonomously i.e. operate even if there is no `hass` instance or internet access.

Since there is neither hass instance nor internet access, I cannot use `time` component since it fetches time from outside sources.

That's why my idea was to rely on `uptime_sensor`, **check it every hour** using `update_interval: 60min`, convert uptime from seconds to hours, and then perform mod operation with `feeder_interval`. If `uptime % feeder_interval == 0`, then I fire the feeder.

```
condition:
 - lambda: |- # Rounding float seconds to the nearest int hour
     int roundToHour = id(uptime_sensor).state/3600 + 0.5;
     return (roundToHour % id(feeder_interval)) == 0;
```

Initially, my idea was to just simply set `uptime.update_interval: $(feeder_interval )` instead of doing this mod operation in lambda, but I don't think `esphome` supports that yet.

I tried similar thing with interval component i.e.
```
interval:
  - interval: $(feeder_interval) + 'h'
     ...
```
But it doesn't work either.

glmnet, 2019-10-11:
i'll use another global variable for elapsed hours, increment once every time the interval hits (every hour)

then if the global counter = global preset run the action and reset the global counter

you can even create a templated sensor to return the global variable so you know when is the next feed, or you can reset it at your wish, e.g. if you have a manual feed action, so next feed is more or less in the next x hours

yeralin, 2019-10-11:
You are amazing. Awesome idea, thank you!