esphome/issues #691: Ir transmit does not work because lambda casts uint32 as uint16

pyorama, 2019-09-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, Rasbian

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 v0.9

**Affected component:**

Lambda code generation

**Description of problem:**

Samsung ir code sending fails because generated c++ code casts uint32_t value as uint16_t.

**Problem-relevant YAML-configuration entries:**
```yaml

globals:
  - id: samsung_data
    type: uint32_t

remote_transmitter:
  pin: D4
  carrier_duty_percent: 50%

mqtt:

  on_json_message:
    - topic: irsend/samsung
      qos: 0
      then:
        - lambda: |-
            if (x.containsKey("data")) {
              id(samsung_data) = x["data"];
              ESP_LOGD("IR", "Samsung code: %lu", id(samsung_data));
            }

        - remote_transmitter.transmit_samsung:
            data: !lambda 'return id(samsung_data);'

```

**Logs (if applicable):**

```

```

**Additional information and things you've tried:**

Generated code block from main.cpp:

```
// this is ok

samsung_data = new globals::GlobalsComponent<uint32_t>();
App.register_component(samsung_data);
lambdaaction = new LambdaAction<const ArduinoJson::JsonObject &>([=](const ArduinoJson::JsonObject & x) -> void {
      if (x.containsKey("data")) {
        samsung_data->value() = x["data"];
        ESP_LOGD("IR", "Samsung code: %lu", samsung_data->value());
      } 
  });

// this should return type uint32_t instead of uint16_t

remote_base_samsungaction = new remote_base::SamsungAction<const ArduinoJson::JsonObject &>();
  remote_base_samsungaction->set_parent(remote_transmitter_remotetransmittercomponent);
  remote_base_samsungaction->set_data([=](const ArduinoJson::JsonObject & x) -> uint16_t {
      return samsung_data->value();
  });
```

glmnet, 2019-09-26:
Is it failing at compile time?

glmnet, 2019-09-27:
did you try changing to `uint32_t` [here](https://github.com/esphome/esphome/blob/d4e0e1518a7682bdafe59c57a650b7c794fd5ad3/esphome/components/remote_base/__init__.py#L643)

pyorama, 2019-09-27:
> did you try changing to `uint32_t` [here](https://github.com/esphome/esphome/blob/d4e0e1518a7682bdafe59c57a650b7c794fd5ad3/esphome/components/remote_base/__init__.py#L643)

Not yet, but I thing that is the bug. It should be uint32_t
Thanks

pyorama, 2019-09-27:
Confirmed. This simple patch fixes the issue:
```
diff --git a/esphome/components/remote_base/__init__.py b/esphome/components/remote_base/__init__.py
index f54fd1d..f3042e5 100644
--- a/esphome/components/remote_base/__init__.py
+++ b/esphome/components/remote_base/__init__.py
@@ -640,7 +640,7 @@ def samsung_dumper(var, config):
 
 @register_action('samsung', SamsungAction, SAMSUNG_SCHEMA)
 def samsung_action(var, config, args):
-    template_ = yield cg.templatable(config[CONF_DATA], args, cg.uint16)
+    template_ = yield cg.templatable(config[CONF_DATA], args, cg.uint32)
     cg.add(var.set_data(template_))
```

Simple test case:
```
  - platform: template
    name: Samsung fails
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: !lambda 'return 3772793023;'
```
This did not work earlier, but works after the patch is applied.

glmnet, 2019-09-27:
Thanks for testing, do you want to send a PR or you'd rather have me do it?

pyorama, 2019-09-27:
Please do it :)

closed 2019-09-28 by glmnet