esphome/issues #692: Light state after reboot

Rink76, 2019-09-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Basic

**Affected component:**

Restore_mode

**Description of problem:**

Initially I used a GPIO Switch to control my light. The option "restore_mode: ALWAYS_OFF" prevented the light from switching on after a reboot.

Because I wanted my light to show-up as light. in Hassio, not as switch. I changed some configuration. Configuration works fine, but on reboot the light is always switched on.

**Problem-relevant YAML-configuration entries:**
```yaml
# Generic ESP8266
esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp01_1m

#
# Binary Sensor Components
#
binary_sensor: 
 
  # GPIO Binary Sensor
  - platform: gpio
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: true
    name: "${upper_devicename} Switch"
    on_press:
      - light.toggle: bulb

#   
# Output Components
#
output:
 
  # ESP8266 Software PWM Output
  - platform: gpio
    pin:
      number: GPIO12
      inverted: false
    id: relay

#
# Light Components
#
light:
  - platform: binary
    name: "${upper_devicename} Relais"
    output: relay
    id: bulb
    restore_mode: ALWAYS_OFF

```

**Logs (if applicable):**

```

```

**Additional information and things you've tried:**

Changed inverted from false to true, now the light stays off after a reboot.

```yaml
  - platform: gpio
    pin:
      number: GPIO12
      inverted: true
    id: relay
```
But now my light switch reports "on" when the bulb (GPIO12) is "off" and "off" when the bulb is "on".

```yaml
light:
  - platform: binary
    name: "${upper_devicename} Relais"
    output: relay
    id: bulb
    restore_mode: ALWAYS_OFF
```
Is there any way to invert the light switch as well, or a different approach to keep the light switched off after reboot ?

Thanks in advance!

glmnet, 2019-09-26:
What does it do if you don’t specify restore mode. By default is restore default off

So if you reset it, reboot, update OTA etc it should keep the state. 
If you cut power wait and power on again it should be off because you are not storing state in flash (that’s another option)

Rink76, 2019-10-04:
removed "restore_mode" from the light section, but unfortunately after a reboot the light still switches on
```yaml
light: 
  - platform: binary
    name: "${upper_devicename} Relais"
    output: relay
    id: lamp

OttoWinter, 2019-10-20:
Huh, that's weird. My guess would be that the button is somehow toggling the light switch at boot. With just the light+gpio output I cannot reproduce this issue.

closed 2020-12-05 by stale[bot]