esphome/issues #694: esp32 cam snapshot creates a thumbnail instead of specified resolution image

vks007, 2019-09-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio
**ESP (ESP32 CAM AI Thinker Board):**

**Affected component:**

ESP32 Camera Component

**Description of problem:**
I have set up a ESP32 camera using ESPHome and am trying to capture snapshots via the snapshot service in HA. While I do get the snapshot the resolution of the image is not what I specify in the ESPHome config. Am I expecting the right thing here? It creates an image of ~60KB this way.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32_camera
  platform: ESP32
  board: esp-wrover-kit
wifi:
  ssid: !secret primary_ssid
  password: !secret primary_ssid_pswd
  fast_connect: True
api:
  reboot_timeout: 0s
ota:
logger:

# ESP32-CAM
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  vertical_flip: false
  resolution: 1280x1024

  name: bedroom
  # ... Frame Settings

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
light:
  - platform: binary
    output: gpio_4
    name: esp32_camera light
binary_sensor:
  - platform: gpio
    pin: GPIO13
    name: "PIR Sensor"
    device_class: motion

```
My Lovelace config to take a snapshot:

```
entities:
  - action_name: Take Snapshot
    icon: 'mdi:camera'
    name: Take Snapshot
    service: camera.snapshot
    service_data:
      entity_id: camera.bedroom
      filename: /config/www/photos/esp32_snapshot.jpg
    type: call-service
show_header_toggle: false
title: Entities card sample
type: entities
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Valcob, 2019-09-27:
It might be related to the amount of ram your board has try to lower the resolution and gradually increase it afterwards. Would say start at 800*600

vks007, 2019-09-27:
Thanks for the hint , I was not looking at the image properties but only the size. The image is indeed of the size I specify in the config , its just that the size is pretty small, I was somehow expecting a larger image size. The size of the image with resolution 1600x1200 & jpeg quality 10 is 66KB, which is surprising. The face of a person at a distance isn't very recognizable as is the use case for my security cam. Assuming this is the best image that can be captured by a esp32 cam I am closing the issue.

closed 2019-09-27 by vks007