esphome/issues #696: Esphome Wifi Connection issues

xefil, 2019-09-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Installation via Docker on Synology NAS, usage with Home Assistant

**ESP (ESP8266, Board/Sonoff):**

ESP8266 Board V2 Nodemcu-CP2102
**Affected component:**

Wifi (maybe) core component

**Description of problem:**
I've two boards installed with esphome. I'm having on both connections issues. I'm debugging one in these days and I've seen that, if the board gets connected to WiFi, until I'm having a ping opened, the connection remains stable. As soon I stop my ping to the board, it gets disconnected after some minutes. Then sometimes it gets connected again for some minutes and again disconnected for hours (or foreveruntil I reset it). Here the yaml used, but take into consideration that the Relay board is actually not connected. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: garage
  platform: ESP8266
  board: nodemcu

wifi:
  ssid: '<<SNIP>>'
  password: '<<SNIP>>'
  power_save_mode: light
  domain: .home.<<SNIP>>

logger:

api:
  password: '<<SNIP>>'

ota:
  password: '<<SNIP>>'

status_led:
  pin: D0

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

switch:
  - platform: gpio
    pin: D6
    inverted: yes
    id: garage_relay
    restore_mode: ALWAYS_OFF
  - platform: template
    name: "Garage Remote"
    icon: "mdi:garage"
    turn_on_action:
    - switch.turn_on: garage_relay
    - delay: 1500ms
    - switch.turn_off: garage_relay
    - delay: 1500ms
    - lambda: !lambda |-
        id(garage_relay).publish_state(false);
    optimistic: false
```

**Logs (if applicable):**

```
[I][logger:116]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[I][app:028]: Running through setup()...
[C][switch.gpio:011]: Setting up GPIO Switch 'garage_relay'...
[D][switch:025]: 'garage_relay' Turning OFF.
[D][switch:045]: 'garage_relay': Sending state ON
[D][switch:025]: 'garage_relay' Turning OFF.
[C][wifi:029]: Setting up WiFi...
[D][wifi:272]: Starting scan...
[D][wifi:287]: Found networks:
[I][wifi:322]: - 'MyNetworkHQ' (98:9B:CB:D2:50:2E) ▂▄▆█
[D][wifi:323]:     Channel: 11
[D][wifi:324]:     RSSI: -55 dB
[I][wifi:322]: - 'MyNetworkHQ' (E0:28:6D:7B:CE:48) ▂▄▆█
[D][wifi:323]:     Channel: 11
[D][wifi:324]:     RSSI: -64 dB
[D][wifi:326]: - '' (08:96:D7:00:4A:6A) ▂▄▆█
[D][wifi:326]: - 'MyNetworkGuest' (74:E5:43:91:FC:D1) ▂▄▆█
[D][wifi:326]: - '' (E0:28:6D:31:0C:FC) ▂▄▆█
[D][wifi:326]: - 'Home&Life SuperWiFi-001939' (78:B2:13:30:68:F7) ▂▄▆█
[D][wifi:326]: - '' (4E:EF:C0:84:3B:53) ▂▄▆█
[D][wifi:326]: - '' (02:71:47:00:99:0A) ▂▄▆█
[D][wifi:326]: - 'Vodafone-41541666' (E4:8F:34:01:18:68) ▂▄▆█
[D][wifi:326]: - 'Vodafone-WiFi' (BC:15:AC:46:E9:DB) ▂▄▆█
[D][wifi:326]: - 'Wind Y800Z_799D' (D8:E5:6D:1D:79:9D) ▂▄▆█
[D][wifi:326]: - 'Vodafone-WiFi' (E4:8F:34:01:18:6A) ▂▄▆█
[I][wifi:164]: WiFi Connecting to 'MyNetworkHQ'...
[I][wifi:380]: WiFi connected!
[C][wifi:254]:   SSID: 'MyNetworkHQ'
[C][wifi:255]:   IP Address: 192.168.3.81
[C][wifi:257]:   BSSID: 98:9B:CB:D2:50:2E
[C][wifi:258]:   Hostname: 'garage'
[C][wifi:262]:   Signal strength: -59 dB ▂▄▆█
[C][wifi:263]:   Channel: 11
[C][wifi:264]:   Subnet: 255.255.255.0
[C][wifi:265]:   Gateway: 192.168.3.1
[C][wifi:266]:   DNS1: 192.168.3.1
[C][wifi:267]:   DNS2: 0.0.0.0
[C][ota:029]: Over-The-Air Updates:
[C][ota:030]:   Address: garage.mydomain:8266
[C][ota:032]:   Using Password.
[C][api:030]: Setting up Home Assistant API server...
[I][app:060]: setup() finished successfully!
[D][sensor:092]: 'WiFi Signal Sensor': Sending state -59.00000 dB with 0 decimal                                                                                                                                                             s of accuracy
[I][app:096]: esphome version 1.13.6 compiled on Sep 26 2019, 09:43:45
[C][status_led:019]: Status LED:
[C][status_led:020]:   Pin: GPIO16 (Mode: OUTPUT)
[C][wifi:372]: WiFi:
[C][wifi:254]:   SSID: 'MyNetworkHQ'
[C][wifi:255]:   IP Address: 192.168.3.81
[C][wifi:257]:   BSSID: 98:9B:CB:D2:50:2E
[C][wifi:258]:   Hostname: 'garage'
[C][wifi:262]:   Signal strength: -59 dB ▂▄▆█
[C][wifi:263]:   Channel: 11
[C][wifi:264]:   Subnet: 255.255.255.0
[C][wifi:265]:   Gateway: 192.168.3.1
[C][wifi:266]:   DNS1: 192.168.3.1
[C][wifi:267]:   DNS2: 0.0.0.0
[C][switch.gpio:042]: GPIO Switch 'garage_relay'
[C][switch.gpio:042]:   Inverted: YES
[C][switch.gpio:043]:   Pin: GPIO12 (Mode: OUTPUT)
[C][switch.gpio:059]:   Restore Mode: Always OFF
[C][template.switch:058]: Template Switch 'Garage Remote'
[C][template.switch:058]:   Icon: 'mdi:garage'
[C][template.switch:059]:   Restore State: NO
[C][template.switch:060]:   Optimistic: NO
[C][logger:137]: Logger:
[C][logger:138]:   Level: DEBUG
[C][logger:139]:   Log Baud Rate: 115200
[C][logger:140]:   Hardware UART: UART0
[C][ota:029]: Over-The-Air Updates:
[C][ota:030]:   Address: garage.mydomain:8266
[C][ota:032]:   Using Password.
[C][api:103]: API Server:
[C][api:104]:   Address: garage.mydomain:6053
[C][wifi_signal.sensor:009]: WiFi Signal 'WiFi Signal Sensor'
[C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[D][api:573]: Client 'Home Assistant 0.97.2 (192.168.1.13)' connected successfully!
[D][sensor:092]: 'WiFi Signal Sensor': Sending state -62.00000 dB with 0 decimals of accuracy
[D][sensor:092]: 'WiFi Signal Sensor': Sending state -62.00000 dB with 0 decimals of accuracy
[I][ota:046]: Boot seems successful, resetting boot loop counter.
[D][sensor:092]: 'WiFi Signal Sensor': Sending state -59.00000 dB with 0 decimals of accuracy
[D][sensor:092]: 'WiFi Signal Sensor': Sending state -63.00000 dB with 0 decimals of accuracy
(... snip ...)
[D][sensor:092]: 'WiFi Signal Sensor': Sending state -63.00000 dB with 0 decimals of accuracy
[D][sensor:092]: 'WiFi Signal Sensor': Sending state -60.00000 dB with 0 decimals of accuracy
[D][sensor:092]: 'WiFi Signal Sensor': Sending state -60.00000 dB with 0 decimals of accuracy
[D][sensor:092]: 'WiFi Signal Sensor': Sending state -60.00000 dB with 0 decimals of accuracy
[W][wifi_esp8266:354]: Event: Disconnected ssid='MyNetworkHQ' bssid=98:9B:CB:D2:50:2E reason='Not Authenticated'
[W][wifi:089]: WiFi Connection lost... Reconnecting...
[W][wifi:431]: Restarting WiFi adapter...
[W][wifi_esp8266:354]: Event: Disconnected ssid='MyNetworkHQ' bssid=98:9B:CB:D2:50:2E reason='Authentication Failed'
[D][api:075]: Disconnecting Home Assistant 0.97.2 (192.168.1.13)
[D][wifi:272]: Starting scan...
[D][wifi:287]: Found networks:
[I][wifi:322]: - 'MyNetworkHQ' (98:9B:CB:D2:50:2E) ▂▄▆█
[D][wifi:323]:     Channel: 11
[D][wifi:324]:     RSSI: -56 dB
[I][wifi:322]: - 'MyNetworkHQ' (E0:28:6D:7B:CE:48) ▂▄▆█
[D][wifi:323]:     Channel: 11
[D][wifi:324]:     RSSI: -60 dB
[D][wifi:326]: - '' (08:96:D7:00:4A:6A) ▂▄▆█
[D][wifi:326]: - 'MyNetworkGuest' (74:E5:43:91:FC:D1) ▂▄▆█
[D][wifi:326]: - '' (E0:28:6D:31:0C:FC) ▂▄▆█
[D][wifi:326]: - '' (4E:EF:C0:84:3B:53) ▂▄▆█
[D][wifi:326]: - 'Home&Life SuperWiFi-001939' (78:B2:13:30:68:F7) ▂▄▆█
[D][wifi:326]: - '' (02:71:47:00:99:0A) ▂▄▆█
[D][wifi:326]: - 'Vodafone-41541666' (E4:8F:34:01:18:68) ▂▄▆█
[D][wifi:326]: - 'Vodafone-WiFi' (E4:8F:34:01:18:6A) ▂▄▆█
[D][wifi:326]: - 'Vodafone-WiFi' (BC:15:AC:46:E9:DB) ▂▄▆█
[D][wifi:326]: - 'Telecom-23377689' (C4:EA:1D:50:79:19) ▂▄▆█
[D][wifi:326]: - 'Wind Y800Z_799D' (D8:E5:6D:1D:79:9D) ▂▄▆█
[D][wifi:326]: - 'Vodafone-34624416' (BC:15:AC:46:E9:D9) ▂▄▆█
[I][wifi:164]: WiFi Connecting to 'MyNetworkHQ'...
[I][wifi:380]: WiFi connected!
[C][wifi:254]:   SSID: 'MyNetworkHQ'
[C][wifi:255]:   IP Address: 192.168.3.81
[C][wifi:257]:   BSSID: 98:9B:CB:D2:50:2E
[C][wifi:258]:   Hostname: 'garage'
[C][wifi:262]:   Signal strength: -61 dB ▂▄▆█
[C][wifi:263]:   Channel: 11
[C][wifi:264]:   Subnet: 255.255.255.0
[C][wifi:265]:   Gateway: 192.168.3.1
[C][wifi:266]:   DNS1: 192.168.3.1
[C][wifi:267]:   DNS2: 0.0.0.0
[D][sensor:092]: 'WiFi Signal Sensor': Sending state -61.00000 dB with 0 decimals of accuracy
[D][api:573]: Client 'Home Assistant 0.97.2 (192.168.1.13)' connected successfully!

```

**Additional information and things you've tried:**

I've different access points with same SSID (two in mesh network and one in repeater), but this shouldn't create issues.

Thanks for the support!!

Simon

xefil, 2019-09-27:
I've disabled all components, expect:

```
esphome:
wifi:
logger:
api:
ota:

```
Same behaviour.

manju-rn, 2019-09-27:
Have you tried removing the power_save_mode?  Let it default to none and check if behavior is any different

xefil, 2019-09-27:
Ok, I'll try it. I was thinking that "light" works better, not?
BTW I've uploaded an old DIY sketch (esp8266 library) I've written and it works great.
I'll let you know removing the POWER mode. Thanks!
Simon

xefil, 2019-09-29:
Thanks, my fault. Removing power LIGHT solved. BTW I'm not understanding why :-)
Thanks for the help!
Simon

closed 2019-09-29 by xefil