esphome/issues #697: Home assitant doesnt reco

Roshaan21, 2019-09-27:
[image]
[image]

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io 0.99.3

**ESP (ESP32/ESP8266, Board/Sonoff):**
NodeMcuv3

**Affected component:**
https://esphome.io/components/switch/gpio.html

**Description of problem:**

So ive been trying to add switches to my home assistant via the ESPHome add on in Home assistant. 
Im using 2 NodeMcus with an ESP8266 chip WIFI chip on them. I managed them to work properly but after a while Home assitant didnt recoginze the relais anymore that I used to turn on and off the lights with, seen in the picture, while all the sensor data was still going through.

I tried to reboot everything a few times and even used some old snaoshots. But nothing works.
It keeps telling me that it vant find any relais/switches eventhough they are in the .yaml code.

I also added some exaple code but that didnt work either.

If you need anything else, pl let me know.

Also Im sorry for any grammar mistakes.
**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: poster_licht
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "My kids took karen"
  password: "Some Random numbers and signs"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "word"

ota:
  password: "Pass"

# Example configuration entry
switch:
  - platform: gpio
    pin: D5
    name: "Living Room Dehumidifier"
  - platform: gpio
    pin: D0
    inverted: yes
    restore_mode: RESTORE_DEFAULT_ON 
    name: "Weltenlicht"
  - platform: gpio
    pin: D1
    inverted: yes
    name: "ight of Takenking"
  - platform: gpio
    pin: D2
    name: "light of Destiny 2"

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```[15:29:41][C][wifi:372]: WiFi:
[15:29:41][C][wifi:254]:   SSID: [redacted]
[15:29:41][C][wifi:255]:   IP Address: __________________
[15:29:41][C][wifi:257]:   BSSID: [redacted]
[15:29:41][C][wifi:258]:   Hostname: 'poster_licht'
[15:29:41][C][wifi:262]:   Signal strength: -64 dB ▂▄▆█
[15:29:41][C][wifi:263]:   Channel: 6
[15:29:41][C][wifi:264]:   Subnet: _________________
[15:29:41][C][wifi:265]:   Gateway: __________________
[15:29:41][C][wifi:266]:   DNS1: _____________________
[15:29:41][C][wifi:267]:   DNS2: 0______________
[15:29:41][C][switch.gpio:042]: GPIO Switch 'Living Room Dehumidifier'
[15:29:41][C][switch.gpio:043]:   Pin: GPIO14 (Mode: OUTPUT)
[15:29:41][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[15:29:41][C][switch.gpio:042]: GPIO Switch 'Weltenlicht'
[15:29:41][C][switch.gpio:042]:   Inverted: YES
[15:29:41][C][switch.gpio:043]:   Pin: GPIO16 (Mode: OUTPUT)
[15:29:41][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to ON)
[15:29:41][C][switch.gpio:042]: GPIO Switch 'ight of Takenking'
[15:29:41][C][switch.gpio:042]:   Inverted: YES
[15:29:41][C][switch.gpio:043]:   Pin: GPIO5 (Mode: OUTPUT)
[15:29:41][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[15:29:41][C][switch.gpio:042]: GPIO Switch 'light of Destiny 2'
[15:29:41][C][switch.gpio:043]:   Pin: GPIO4 (Mode: OUTPUT)
[15:29:41][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[15:29:41][C][logger:137]: Logger:
[15:29:41][C][logger:138]:   Level: DEBUG
[15:29:41][C][logger:139]:   Log Baud Rate: 115200
[15:29:41][C][logger:140]:   Hardware UART: UART0
[15:29:41][C][ota:029]: Over-The-Air Updates:
[15:29:41][C][ota:030]:   Address: poster_licht.local:8266
[15:29:41][C][ota:032]:   Using Password.
[15:29:41][C][api:103]: API Server:
[15:29:41][C][api:104]:   Address: poster_licht.local:6053

**Additional information and things you've tried:**

As said, i rebooted everything a few times, changed the cable, 
deleted and readded the ESPHome integration and used some exsample code to
see if my code eas wrong.

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

manju-rn, 2019-09-27:
Are the ESPs showing up in the integrations? I see that you have 2 notification - can you check if they are for the failed ESP integration?

Roshaan21, 2019-09-27:
I can see the ESPs in the integrations tab.

[image]

> 2019-09-27 14:19:30 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of switch. Setup failed for dependencies: group

> 2019-09-27 14:19:30 ERROR (MainThread) [homeassistant.setup] Setup failed for switch: Could not set up all dependencies.

This is the switch related error i found
[image]

glmnet, 2019-09-27:
Have you been playing around with `groups` in HA in between it stopped working?

Roshaan21, 2019-09-27:
I deleted some old switches that i used to use and replaced them with the new ones that aren't working now as well.

brandond, 2019-09-27:
You've got a ton of errors in there. Maybe try fixing everything else that's wrong with your HA install and then see if it starts working? Everything looks good on the esphome side.

yeralin, 2019-09-28:
Look at my https://github.com/esphome/issues/issues/312#issuecomment-496059755
Might be the same

Install Hass container on your laptop/PC and try to connect your ESPHome to it. See if it works

Roshaan21, 2019-09-28:
I tried to delete a few things from the groups.yaml and rebooted everything a few times. It's now working properly.

Thank you very much for your help.

closed 2019-09-28 by Roshaan21