esphome/issues #699: No such file or directory:

Spackstor, 2019-09-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Just getting started. Mosquitto and esphome on raspberry pi4
**ESP (ESP32/ESP8266, Board/Sonoff):**
/ nodemcu lolin v3

**Affected component:**
not applicable

**Description of problem:**
Yaml compiles but cannot be uploaded or downloaded because of directory

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphomeyaml:
  name: garage
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 
  password: 

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
Compiles without errors 
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d config/garage
WARNING ********************************************************************************
WARNING A new version of esphomeyaml is available: 1.13.6 (this is 1.10.1)
WARNING Changelog: https://esphomelib.com/esphomeyaml/changelog/index.html
WARNING Update Instructions: https://esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-update-to-the-latest-version
WARNING ********************************************************************************
Processing garage (board: nodemcuv2; framework: arduino; platform: espressif8266@1.8.0)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 1.8.0 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 1.40802.2 (4.8.2), tool-esptool 1.413.0 (4.13), framework-arduinoespressif8266 2.20402.4 (2.4.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphomelib> 1.10.1
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.1.3
|   |   |   |-- <ESP8266WiFi> 1.0
[... 141 lines cut ...]
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.1.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Wire> 1.0
Retrieving maximum program size .pio/build/garage/firmware.elf
Checking size .pio/build/garage/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
========================= [SUCCESS] Took 9.55 seconds =========================
INFO Successfully compiled program.
INFO Running:  esptool.py --before default_reset --after hard_reset --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 config/garage/.pioenvs/garage/firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: u'config/garage/.pioenvs/garage/firmware.bin'

brandond, 2019-09-29:
Dude

> WARNING A new version of esphomeyaml is available: 1.13.6 (this is 1.10.1)

OttoWinter, 2019-10-20:
As @brandond said: 🚨🚨🚨

> WARNING A new version of esphomeyaml is available: 1.13.6 (this is 1.10.1)

Closing until you try with a more recent version

closed 2019-10-20 by OttoWinter