esphome/issues #705: Compile error why am I getting this was ok a few week ago

davidclancy, 2019-09-28:
**Operating environment/Installation (Hass.io.):**

**ESP (ESP32/ESP8266, Board/ndemcu):**

**esp8266:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bed_led
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "y"
  password: ""

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ''

ota:
  password: ''
 
  
  
light:
  - platform: fastled_clockless
    chipset: WS2811
    pin: D8
    num_leds: 360
    rgb_order: GRB
    name: "Bed_LED"
    

    # Fireworks Effect
    effects:
      - addressable_fireworks:
      - addressable_fireworks:
          name: Fireworks Effect With Custom Values
          update_interval: 32ms
          spark_probability: 10%
          use_random_color: false
          fade_out_rate: 120
          
    # Random Twinkle
      - addressable_random_twinkle:
      - addressable_random_twinkle:
          name: Random Twinkle Effect With Custom Values
          twinkle_probability: 5%
          progress_interval: 32ms
          
    # Twinkle Effect
      - addressable_twinkle:
      - addressable_twinkle:
          name: Twinkle Effect With Custom Values
          twinkle_probability: 5%
          progress_interval: 4ms      
 
    # Color Wipe
      - addressable_color_wipe:
      - addressable_color_wipe:
          name: Color Wipe Effect With Custom Values
          colors:
            - red: 100%
[... 420 lines cut ...]
/data/bed_led/.pioenvs/bed_led/lib529/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::onPoll(std::function<void (void*, AsyncClient*)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient6onPollESt8functionIFvPvPS_EES1_+0x4): multiple definition of `AsyncClient::onPoll(std::function<void (void*, AsyncClient*)>, void*)'
/data/bed_led/.pioenvs/bed_led/lib71d/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient6onPollESt8functionIFvPvPS_EES1_+0x4): first defined here
/data/bed_led/.pioenvs/bed_led/lib529/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncServer::onClient(std::function<void (void*, AsyncClient*)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncServer8onClientESt8functionIFvPvP11AsyncClientEES1_+0x4): multiple definition of `AsyncServer::onClient(std::function<void (void*, AsyncClient*)>, void*)'
/data/bed_led/.pioenvs/bed_led/lib71d/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncServer8onClientESt8functionIFvPvP11AsyncClientEES1_+0x4): first defined here
/data/bed_led/.pioenvs/bed_led/lib529/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::onAck(std::function<void (void*, AsyncClient*, unsigned int, unsigned int)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient5onAckESt8functionIFvPvPS_jjEES1_+0x8): multiple definition of `AsyncClient::onAck(std::function<void (void*, AsyncClient*, unsigned int, unsigned int)>, void*)'
/data/bed_led/.pioenvs/bed_led/lib71d/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient5onAckESt8functionIFvPvPS_jjEES1_+0x8): first defined here
/data/bed_led/.pioenvs/bed_led/lib529/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::onData(std::function<void (void*, AsyncClient*, void*, unsigned int)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient6onDataESt8functionIFvPvPS_S1_jEES1_+0x8): multiple definition of `AsyncClient::onData(std::function<void (void*, AsyncClient*, void*, unsigned int)>, void*)'
/data/bed_led/.pioenvs/bed_led/lib71d/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient6onDataESt8functionIFvPvPS_S1_jEES1_+0x8): first defined here
/data/bed_led/.pioenvs/bed_led/lib529/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::onPacket(std::function<void (void*, AsyncClient*, pbuf*)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient8onPacketESt8functionIFvPvPS_P4pbufEES1_+0x8): multiple definition of `AsyncClient::onPacket(std::function<void (void*, AsyncClient*, pbuf*)>, void*)'
/data/bed_led/.pioenvs/bed_led/lib71d/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient8onPacketESt8functionIFvPvPS_P4pbufEES1_+0x8): first defined here
/data/bed_led/.pioenvs/bed_led/lib529/libESPAsyncTCP_ID305@1.2.0.a(ESPAsyncTCP.cpp.o): In function `AsyncClient::onTimeout(std::function<void (void*, AsyncClient*, unsigned int)>, void*)':
ESPAsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x8): multiple definition of `AsyncClient::onTimeout(std::function<void (void*, AsyncClient*, unsigned int)>, void*)'
/data/bed_led/.pioenvs/bed_led/lib71d/libESPAsyncTCP_ID305.a(ESPAsyncTCP.cpp.o):ESPAsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x8): first defined here
collect2: error: ld returned 1 exit status
*** [/data/bed_led/.pioenvs/bed_led/firmware.elf] Error 1
========================= [ERROR] Took 110.62 seconds =========================

```

**Additional information and things you've tried:**

gleepwurp, 2019-09-28:
Check this issue here, looks like the same issue you're having:

https://github.com/esphome/issues/issues/683

G.

glmnet, 2019-09-30:
@gleepwurp is right. Closing

closed 2019-09-30 by glmnet

brandond, 2019-09-30:
Duplicate of #683