esphome/issues #707: Template Question (double switches issue)

xefil, 2019-09-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Installation via Docker on Synology NAS, usage with Home Assistant

**ESP (ESP8266, Board/Sonoff):**

ESP8266 Board V2 Nodemcu-CP2102
**Affected component:**

NOPE

**Description of problem:**

Hello. I'm trying to work with two switches. The first one "cancello_relay" works as expected. As soon I send from HA a Turn_on command, the switch turns on and after 1500ms turns off. This because it's a gate button door.
Then, I've a simply other switch "stufa_relay", as soon I turn it on, it should remain on. And when I decide to turn it off, it should turn off. But in the practice, it remains ON, BUT it sends back to Home Assistant the OFF command.
But the switch remains ON. To turn it OFF, I need to turn it on again (nothing happens because it's still ON), and immediate after (before the OFF command is sent back 1500ms) I need to turn it OFF.
What I'm doing wrong?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: stufa
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'MYSSID'
  password: 'MYPWD'
  manual_ip:
    static_ip: 192.168.3.82
    gateway: 192.168.3.1
    subnet: 255.255.255.0
    dns1: 192.168.3.1
    dns2: 192.168.3.1
  domain: MYDOMAIN

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: MYPWD

# Enable Over The Air
ota:
  password: MYPWD

# Status Led
status_led:
  pin: D0

sensor:
  # DHT 11
  - platform: dht
    pin: D2
    temperature:
      name: "Stufa Temperatura"
    humidity:
      name: "Stufa Umidita"
    update_interval: 60s
    model: DHT11
  # Wifi sensor signa strength
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

switch:
  - platform: gpio
    pin: D6
    inverted: yes
    id: cancello_relay
    restore_mode: ALWAYS_OFF
  - platform: gpio
    pin: D5
    inverted: yes
    id: stufa_relay
    restore_mode: ALWAYS_OFF
  - platform: template
    name: "Cancello Remote"
    icon: "mdi:gate"
    turn_on_action:
    - switch.turn_on: cancello_relay
    - delay: 1500ms
    - switch.turn_off: cancello_relay
    - delay: 1500ms
    - lambda: !lambda |-
        id(cancello_relay).publish_state(false);
    optimistic: false
  - platform: template
    name: "Stufa Remote"
    icon: "mdi:fire"
    optimistic: false
    turn_on_action:
    - switch.turn_on: stufa_relay
    - lambda: !lambda |-
        id(stufa_relay).publish_state(false);
    turn_off_action:
    - switch.turn_off: stufa_relay
    - lambda: !lambda |-
        id(stufa_relay).publish_state(true);
```

**Logs (if applicable):**

```
NOPE
```

**Additional information and things you've tried:**

Thanks for the support!!

Simon

brandond, 2019-09-30:
In my experience, any time you start messing around with manual publish_state calls you're doing something wrong. See if you can do what you want with only templates and turn_on/delay/turn_off actions. That's much less likely to result in unwanted behavior.

xefil, 2019-09-30:
Hello,

I think I'm confusing differences between switch platform `gpio` and platform `template`....
Having first used ONLY `cancell_relay` all was working well, expect that the output returned to the HA was inverted (the relay board works inverted). So I've used the `inverted: yes` option and the `lambda` to send the inverted output state to Home Assistant. Now, adding a second switch, I'm having strange behaviours.... seems the template is applied twice... or what I'm doing wrong? Some examples or help?

My goal is:
Relay1 (cancello_relay), as soon I turn it on, it should be turned on and after 1500ms turned off. Due the relay board works inverted, it should reflect the inverted value to work.
Relay2 (stufa_relay), it's on the same relay board, so it works inverted. BTW, in this case, if I turn it on, it should remain on. And when I decide to turn it off, then it should turn it off.

Thanks for any help!

Simon

glmnet, 2019-09-30:
You need only 1 template switch, not 2.
Start with the simple code, then if it does not behave like you want we can figure out what to do.

> Relay1 (cancello_relay), as soon I turn it on, it should be turned on and after 1500ms turned off. The off state should then returned to Home Assistant, that needs to know it was turned on, and after 1500ms, turned off. Due the relay board works inverted, it should reflect the inverted value to work.

https://esphome.io/components/switch/gpio.html#momentary-switch

> Relay2 (stufa_relay), it's on the same relay board, so it works inverted. BTW, in this case, if I turn it on, it should remain on. And when I decide to turn it off, then it should turn it off.

https://esphome.io/components/switch/gpio.html#active-low-switch

xefil, 2019-09-30:
Hello @glmnet and thanks for the suggestions. This is a good point to start: `You need only 1 template switch, not 2.`
I explain why I've used it twice: AFAIK (and it's really less :-) ) I ws thinging this part is the only way to set informations like icons:

```
   icon: "mdi:fire"
```
So I've used the momentary switch as example and I've created then the first working relay switch.
Then, it was time to add a second one and so I've added a second template.
I'll give a try removing one template. I'll let you know.

Thanks!

Simon

glmnet, 2019-09-30:
`icon: ` is a [switch core](https://esphome.io/components/switch/index.html) feature, so you can use it on gpio switch too:
```yaml
switch:
  - platform: gpio
    icon: mdi:fire
  ...
```

glmnet, 2019-09-30:
You can of course do it with another template, you can do it with ten if that makes you feel smarter, however IMHO the less yaml you write the prettier (because it is easier to understand and maintain)

It takes some time to understand yaml and ESPHome, docs are very precise and complete

xefil, 2019-09-30:
Hi @glmnet ,
True, understanding te logic behind is a little complicated. BTW I'll try and learn.
Testing it from remote, so seems to work, but I need to check it on the board to see if the relay actual makes what expected:
```yaml
switch:
  - platform: gpio
    pin: D6
    inverted: yes
    id: cancello_relay
    restore_mode: ALWAYS_OFF
  - platform: gpio
    pin: D5
    inverted: yes
    id: stufa_relay
    restore_mode: ALWAYS_OFF
    name: "Stufa Remote"
    icon: "mdi:fire"
  - platform: template
    name: "Cancello Remote"
    icon: "mdi:gate"
    turn_on_action:
    - switch.turn_on: cancello_relay
    - delay: 1500ms
    - switch.turn_off: cancello_relay
    - delay: 1500ms
    - lambda: !lambda |-
        id(cancello_relay).publish_state(false);
    optimistic: false
```
I'm not really sure what the `optimistic` does.... :-(
for the rest, as soon I'll check weather the led on the relay starts or not, I'll confirm you if it's working :-)
Cheers, Simon

glmnet, 2019-09-30:
Just to notice there is two ways of using the `invert: ` there
* the way you are using, you are inverting the switch
* the other way is indenting it inside the `pin: ` so it will actually invert the `pin:` I believe they should behave the same though. I would invert the `pin:` as it's what makes more sense to me.

`optimistic:` means the component (in this case the template switch) will assume the output it gives is accepted and valid, this is used when
  - you cannot retrieve the actual state from an external source (this is using the `lambda:`), 
  - or when an external source cannot indicate the state of the device itself (this is using the `publish_state`).

In other words, consider you have a contactor with an auxiliary output, you command the contactor with an output, but the feedback comes from another input pin, so in this case you'll configure a template switch, optimistic false and will use the lambda to feedback the input pin state.

The `publish_state`, is there for more complicated actions, but if you are using lambda or `optimistic: true` then you should not use publish_state

Optimistic true means the switch has it's state on its own, it knows it's state because somebody turn it on or turn it off.

Reviewing your code:
I'll remove all the last four lines:

```
    - delay: 1500ms
    - lambda: !lambda |-
        id(cancello_relay).publish_state(false);
    optimistic: false
```

the `optimistic: false` is not needed because it is the default value
the lambda call to set the switch to false is not needed either, as this is optimistic: false, it needs feedback to be true (the lambda returning true) or a call to `publish_state(true)` since none of those are there ESPHome will not report the switch state to be true... so this works much more like a push button.
So the delay is also not needed because it would be the last action.
This is why the momentary switch example works.

In case you want to see the switch being toggle true on HA for a long (custom) time you'll need to add a `publish_state(true)` somewhere, but IMHO this is unnecessary and over complicating stuff.

xefil, 2019-09-30:
Thanks  @glmnet for the detailed anser, really!
I know jaml is really sensible to indentation, so I've overseen that I can use the `inverted: ` mode in the switch and on te pin too. Two spaces that makes the difference to me, and so it makes sense and I've understood why I've inverted it and had needed to publish it back inverted again.
Let me correct it, govea try this evening and then let you know.
Great support!!! :-)

xefil, 2019-10-01:
Thank you @glmnet all went perfectly!
I wish you a nice day!
Simon

closed 2019-10-01 by xefil