esphome/issues #713: API fails to compile for ESP8266/ESP8285

syphernl, 2019-09-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (Blitzwolf BW-SHP6)

**Affected component:**

https://esphome.io/components/api

**Description of problem:**
I am trying to build firmware for a new device, but it fails with a whole bunch of redefined classes:
```
[32mINFO Reading configuration...[0m
[32mINFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 03/29/20 02:00:00 to 10/25/20 03:00:00[0m
[32mINFO Generating C++ source...[0m
[32mINFO Compiling app...[0m
[32mINFO Running:  platformio run -d blitz02[0m
Processing blitz02 (board: esp8285; framework: arduino; platform: espressif8266@1.8.0)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8285.html
PLATFORM: Espressif 8266 1.8.0 > Generic ESP8285 Module
HARDWARE: ESP8266 80MHz, 80KB RAM, 423.98KB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), tool-esptool 1.413.0 (4.13), framework-arduinoespressif8266 2.20402.4 (2.4.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Update library in registry
Warning! Library `{'requirements': None, 'name': 'Update'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Update'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.1
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/blitz02/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/blitz02/src/esphome/components/time/real_time_clock.cpp.o
Compiling .pioenvs/blitz02/src/esphome/components/total_daily_energy/total_daily_energy.cpp.o
Compiling .pioenvs/blitz02/src/esphome/components/uptime/uptime_sensor.cpp.o
Compiling .pioenvs/blitz02/src/esphome/components/web_server/web_server.cpp.o
Compiling .pioenvs/blitz02/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/blitz02/src/esphome/components/wifi/wifi_component_esp32.cpp.o
Compiling .pioenvs/blitz02/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
In file included from src/esphome/components/api/api_server.h:9:0,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/basic_messages.h:8:7: error: redefinition of 'class esphome::api::HelloRequest'
 class HelloRequest : public APIMessage {
       ^
In file included from src/esphome/components/api/api_connection.h:5:0,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_pb2.h:52:7: error: previous definition of 'class esphome::api::HelloRequest'
 class HelloRequest : public ProtoMessage {
       ^
In file included from src/esphome/components/api/api_server.h:9:0,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/basic_messages.h:19:7: error: redefinition of 'class esphome::api::ConnectRequest'
[... 463 lines cut ...]
    name: "${plug_name}_Total Daily Energy"
    power_id: "${plug_name}_Wattage"
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
    unit_of_measurement: kWh
    icon: mdi:clock-alert
      
# Extra sensor to keep track of plug uptime
  - platform: uptime
    name: ${plug_name}_Uptime Sensor
    icon: mdi:clock-outline
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Versions: 
* Esphome: 1.13.6
* PlatformIO 4.0.3

brandond, 2019-10-01:
Have you tried the dev branch?

syphernl, 2019-10-01:
@brandond 1.14.0-dev fails as well but on different errors:
```
In file included from src/esphome/components/api/basic_messages.h:3:0,
                 from src/esphome/components/api/basic_messages.cpp:1:
src/esphome/components/api/api_message.h:75:23: error: 'APIBuffer' has not been declared
   virtual void encode(APIBuffer &buffer);
                       ^
In file included from src/esphome/components/api/basic_messages.cpp:1:0:
src/esphome/components/api/basic_messages.h:38:15: error: 'APIBuffer' has not been declared
   void encode(APIBuffer &buffer) override;
               ^
src/esphome/components/api/basic_messages.cpp: In member function 'virtual bool esphome::api::HelloRequest::decode_length_delimited(uint32_t, const uint8_t*, size_t)':
src/esphome/components/api/basic_messages.cpp:11:48: error: 'as_string' was not declared in this scope
       this->client_info_ = as_string(value, len);
                                                ^
In file included from src/esphome/components/api/command_messages.h:5:0,
                 from src/esphome/components/api/command_messages.cpp:1:
src/esphome/components/api/api_message.h:75:23: error: 'APIBuffer' has not been declared
   virtual void encode(APIBuffer &buffer);
                       ^
In file included from src/esphome/components/api/api_message.cpp:1:0:
src/esphome/components/api/api_message.h:75:23: error: 'APIBuffer' has not been declared
   virtual void encode(APIBuffer &buffer);
                       ^
src/esphome/components/api/api_message.cpp:12:25: error: variable or field 'encode' declared void
 void APIMessage::encode(APIBuffer &buffer) {}
                         ^
src/esphome/components/api/api_message.cpp:12:25: error: 'APIBuffer' was not declared in this scope
src/esphome/components/api/api_message.cpp:12:36: error: 'buffer' was not declared in this scope
 void APIMessage::encode(APIBuffer &buffer) {}
                                    ^
src/esphome/components/api/api_message.cpp: In member function 'void esphome::api::APIMessage::decode(const uint8_t*, size_t)':
src/esphome/components/api/api_message.cpp:18:72: error: 'proto_decode_varuint32' was not declared in this scope
     auto res = proto_decode_varuint32(&buffer[i], length - i, &consumed);
                                                                        ^
src/esphome/components/api/basic_messages.cpp: In member function 'virtual bool esphome::api::ConnectRequest::decode_length_delimited(uint32_t, const uint8_t*, size_t)':
src/esphome/components/api/basic_messages.cpp:25:45: error: 'as_string' was not declared in this scope
       this->password_ = as_string(value, len);
                                             ^
src/esphome/components/api/basic_messages.cpp: In member function 'virtual bool esphome::api::DisconnectRequest::decode_length_delimited(uint32_t, const uint8_t*, size_t)':
src/esphome/components/api/basic_messages.cpp:40:43: error: 'as_string' was not declared in this scope
       this->reason_ = as_string(value, len);
                                           ^
src/esphome/components/api/basic_messages.cpp: At global scope:
src/esphome/components/api/basic_messages.cpp:48:32: error: variable or field 'encode' declared void
 void DisconnectRequest::encode(APIBuffer &buffer) {
                                ^
src/esphome/components/api/basic_messages.cpp:48:32: error: 'APIBuffer' was not declared in this scope
src/esphome/components/api/basic_messages.cpp:48:43: error: 'buffer' was not declared in this scope
 void DisconnectRequest::encode(APIBuffer &buffer) {
                                           ^
*** [.pioenvs/blitz02/src/esphome/components/api/api_message.cpp.o] Error 1
*** [.pioenvs/blitz02/src/esphome/components/api/command_messages.cpp.o] Error 1
*** [.pioenvs/blitz02/src/esphome/components/api/basic_messages.cpp.o] Error 1

```

renat85, 2019-10-01:
Comment web server component in your config and try compile.

#web_server:
#  port: 80

syphernl, 2019-10-02:
> Comment web server component in your config and try compile.
> 
> #web_server:
> #port: 80

Still fails:
```
In file included from src/esphome/components/api/api_message.cpp:1:0:
src/esphome/components/api/api_message.h:75:23: error: 'APIBuffer' has not been declared
   virtual void encode(APIBuffer &buffer);
                       ^
src/esphome/components/api/api_message.cpp:12:25: error: variable or field 'encode' declared void
 void APIMessage::encode(APIBuffer &buffer) {}
                         ^
src/esphome/components/api/api_message.cpp:12:25: error: 'APIBuffer' was not declared in this scope
src/esphome/components/api/api_message.cpp:12:36: error: 'buffer' was not declared in this scope
 void APIMessage::encode(APIBuffer &buffer) {}
                                    ^
src/esphome/components/api/api_message.cpp: In member function 'void esphome::api::APIMessage::decode(const uint8_t*, size_t)':
src/esphome/components/api/api_message.cpp:18:72: error: 'proto_decode_varuint32' was not declared in this scope
     auto res = proto_decode_varuint32(&buffer[i], length - i, &consumed);
                                                                        ^
*** [.pioenvs/blitz02/src/esphome/components/api/api_message.cpp.o] Error 1
In file included from src/esphome/components/api/basic_messages.h:3:0,
                 from src/esphome/components/api/basic_messages.cpp:1:
src/esphome/components/api/api_message.h:75:23: error: 'APIBuffer' has not been declared
   virtual void encode(APIBuffer &buffer);
                       ^
In file included from src/esphome/components/api/basic_messages.cpp:1:0:
src/esphome/components/api/basic_messages.h:38:15: error: 'APIBuffer' has not been declared
   void encode(APIBuffer &buffer) override;
               ^
src/esphome/components/api/basic_messages.cpp: In member function 'virtual bool esphome::api::HelloRequest::decode_length_delimited(uint32_t, const uint8_t*, size_t)':
src/esphome/components/api/basic_messages.cpp:11:48: error: 'as_string' was not declared in this scope
       this->client_info_ = as_string(value, len);
                                                ^
src/esphome/components/api/basic_messages.cpp: In member function 'virtual bool esphome::api::ConnectRequest::decode_length_delimited(uint32_t, const uint8_t*, size_t)':
src/esphome/components/api/basic_messages.cpp:25:45: error: 'as_string' was not declared in this scope
       this->password_ = as_string(value, len);
                                             ^
src/esphome/components/api/basic_messages.cpp: In member function 'virtual bool esphome::api::DisconnectRequest::decode_length_delimited(uint32_t, const uint8_t*, size_t)':
src/esphome/components/api/basic_messages.cpp:40:43: error: 'as_string' was not declared in this scope
       this->reason_ = as_string(value, len);
                                           ^
src/esphome/components/api/basic_messages.cpp: At global scope:
src/esphome/components/api/basic_messages.cpp:48:32: error: variable or field 'encode' declared void
 void DisconnectRequest::encode(APIBuffer &buffer) {
                                ^
src/esphome/components/api/basic_messages.cpp:48:32: error: 'APIBuffer' was not declared in this scope
src/esphome/components/api/basic_messages.cpp:48:43: error: 'buffer' was not declared in this scope
 void DisconnectRequest::encode(APIBuffer &buffer) {
                                           ^
*** [.pioenvs/blitz02/src/esphome/components/api/basic_messages.cpp.o] Error 1
In file included from src/esphome/components/api/command_messages.h:5:0,
                 from src/esphome/components/api/command_messages.cpp:1:
src/esphome/components/api/api_message.h:75:23: error: 'APIBuffer' has not been declared
   virtual void encode(APIBuffer &buffer);
                       ^
*** [.pioenvs/blitz02/src/esphome/components/api/command_messages.cpp.o] Error 1```

I can only get it to compile if I disable `api` and anything relating to that.

glmnet, 2019-10-03:
Try cleaning everything, deleting the folder that has your node name `blitz02`

syphernl, 2019-10-03:
> Try cleaning everything, deleting the folder that has your node name `blitz02`

With the API stuff enabled: I have already tried that to no avail :(

glmnet, 2019-10-04:
Duplicate of #683

glmnet, 2019-10-04:
> With the API stuff enabled: I have already tried that to no avail :(

Please note is a known bug of 1.13 that api: won't compile, are you confirming you are using dev + did the cleaning on dev?

glmnet, 2019-10-04:
I executed your yaml from first post with 1.13.6 and reproduced you issue.
I switched to dev without cleaning anything the issue persists (all that follows is with dev version)

I did
`rm -rf bliz02`
then compiled again and issue persisted
I commented web_server and it compiled.
added web server back and it failed
delete again
`rm -rf bliz02`
Commented web server and it worked
Uncommented web server and it still worked

So in order to have a successful compilation you must first compile from a clean state without the web_server component and then later add the web_server component without doing a clean.

glmnet, 2019-10-04:
I'm closing this as duplicated of #683

closed 2019-10-04 by glmnet

syphernl, 2019-10-05:
> I commented web_server and it compiled.

I cannot reproduce this here. It consistently keeps on failing on the API, with or without the webserver commented out.

glmnet, 2019-10-05:
Can you try changing the node name to E.g. blitz03 (a new one) while keeping web server commented

syphernl, 2019-10-05:
> Can you try changing the node name to E.g. blitz03 (a new one) while keeping web server commented

I have copied the file to blitz03.yaml and tried to compile it but no luck either.

glmnet, 2019-10-05:
It’s not changing the file name is changing the mode name. Where it says plug_name blitz02

syphernl, 2019-10-05:
> It’s not changing the file name is changing the node name. Where it says plug_name blitz02

I changed both the filename and the variable.