esphome/issues #714: Dallas temperature sensor "Unavailable" in Home Assistant

sabeechen, 2019-10-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
Home Assistant  0.98.5
ESPHome 1.13.6 
Running on an Ubuntu 19.04 Host (not a VM)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-32S ([amazon link](https://www.amazon.com/gp/product/B0718T232Z))

**Affected component:**
[Dallas Temperature Sensor](https://esphome.io/components/sensor/dallas.html)

**Description of problem:**
The temperature shows up in the log output from the ESP32, but the sensor shows up as "unavailable" in Home Assistant.  The board has a bunch of other devices configured for it (RGB strip, 2 relays, some buttons, and a rotary encoder).  They all function correctly and have their state accurately reported to Home Assistant.  Only the temperature sensor seems to have this problem.  It is the only device on the dallas bus.  I'm able to see the correct value of the sensor from lambdas on the ESP32, so it seems like the problem is with the addon and not the microcontroller. 
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
...
dallas:
  - pin: GPIO25
    update_interval: 5s

...

sensor:
  - platform: dallas
    id: transformer_temp
    address: 0x4103049779539728
    name: "Living Room Light Temperature"
    unit_of_measurement: "°F"
...
```

**Logs (if applicable):**

```
INFO Reading configuration...
INFO Detected timezone 'MST' with UTC offset -7 and daylight savings time from 03/08/20 02:00:00 to 11/01/20 02:00:00
INFO Starting log output from ********** using esphome API
INFO Connecting to ********** (**********)
INFO Successfully connected to **********
[20:21:42][D][time:030]: Synchronized time: Tue Oct  1 20:21:42 2019
[20:21:43][I][app:096]: esphome version 1.13.6 compiled on Oct  1 2019, 20:17:28
[20:21:43][C][wifi:372]: WiFi:
[20:21:43][C][wifi:254]:   SSID: 

[redacted]
[20:21:43][C][wifi:255]:   IP Address: **********
[20:21:43][C][wifi:257]:   BSSID: 

[redacted]
[20:21:43][C][wifi:258]:   Hostname: **********
[20:21:43][C][wifi:262]:   Signal strength: -66 dB ▂▄▆█
[20:21:43][C][wifi:263]:   Channel: 7
[20:21:43][C][wifi:264]:   Subnet: 255.255.255.0
[20:21:43][C][wifi:265]:   Gateway: 192.168.1.1
[20:21:43][C][wifi:266]:   DNS1: 192.168.1.1
[20:21:43][C][wifi:267]:   DNS2: 0.0.0.0
[20:21:43][C][power_supply:017]: Power Supply:
[20:21:43][C][power_supply:018]:   Pin: GPIO21 (Mode: OUTPUT)
[20:21:43][C][power_supply:019]:   Time to enable: 20 ms
[20:21:43][C][power_supply:020]:   Keep on time: 10.0 s
[20:21:43][C][switch.gpio:042]: GPIO Switch 'Living Room Lamp White 1'
[20:21:43][C][switch.gpio:043]:   Pin: GPIO23 (Mode: OUTPUT)
[20:21:43][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[20:21:43][C][switch.gpio:042]: GPIO Switch 'Living Room Lamp White 2'
[20:21:43][C][switch.gpio:043]:   Pin: GPIO22 (Mode: OUTPUT)
[20:21:43][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[20:21:43][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Living Room Lamp Button'
[20:21:43][C][gpio.binary_sensor:016]:   Pin: GPIO19 (Mode: INPUT_PULLUP, INVERTED)
[20:21:43][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Living Room Lamp Motion'
[20:21:43][C][gpio.binary_sensor:016]:   Pin: GPIO16 (Mode: INPUT_PULLUP)
[20:21:43][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Living Room Lamp Switch'
[20:21:43][C][gpio.binary_sensor:016]:   Pin: GPIO15 (Mode: INPUT_PULLUP)
[20:21:43][D][api:075]: Disconnecting ESPHome v1.13.6 (192.168.1.149)
[20:21:43][C][fastled:019]: FastLED light:
[20:21:43][C][fastled:020]:   Num LEDs: 185
[20:21:43][C][fastled:021]:   Max refresh rate: 400
[20:21:43][C][logger:137]: Logger:
[20:21:43][C][logger:138]:   Level: DEBUG
[20:21:43][C][logger:139]:   Log Baud Rate: 115200
[20:21:43][C][logger:140]:   Hardware UART: UART0
[20:21:43][C][light:174]: Light 'Living Room Lamp RGB'
[20:21:43][C][light:176]:   Default Transition Length: 1.0s
[20:21:43][C][light:177]:   Gamma Correct: 2.80
[20:21:43][C][rotary_encoder:115]: Rotary Encoder 'Living Room Lamp Encoder'
[20:21:43][C][rotary_encoder:115]:   Unit of Measurement: 'steps'
[20:21:43][C][rotary_encoder:115]:   Accuracy Decimals: 0
[20:21:43][C][rotary_encoder:115]:   Icon: 'mdi:rotate-right'
[20:21:43][C][rotary_encoder:116]:   Pin A: GPIO18 (Mode: INPUT_PULLUP)
[20:21:43][C][rotary_encoder:117]:   Pin B: GPIO17 (Mode: INPUT_PULLUP)
[20:21:43][C][rotary_encoder:124]:   Resolution: 2 Pulses Per Cycle
[20:21:43][C][homeassistant.time:010]: Home Assistant Time:
[20:21:43][C][homeassistant.time:011]:   Timezone: *********
[20:21:43][C][dallas.sensor:070]: DallasComponent:
[20:21:43][C][dallas.sensor:071]:   Pin: GPIO25 (Mode: INPUT)
[20:21:43][C][dallas.sensor:072]:   Update Interval: 5.0s
[20:21:43][D][dallas.sensor:077]:   Found sensors:
[20:21:43][D][dallas.sensor:080]:     0x4103049779539728
[20:21:43][C][dallas.sensor:085]:   Device 'Living Room Light Temperature'
[20:21:43][C][dallas.sensor:085]:     Unit of Measurement: '°F'
[20:21:43][C][dallas.sensor:085]:     Accuracy Decimals: 1
[20:21:43][C][dallas.sensor:085]:     Icon: 'mdi:thermometer'
[20:21:43][C][dallas.sensor:093]:     Address: 0x4103049779539728
[20:21:43][C][dallas.sensor:094]:     Resolution: 12
[20:21:43][C][ota:029]: Over-The-Air Updates:
[20:21:43][C][ota:030]:   Address: **********
[20:21:43][C][ota:032]:   Using Password.
[20:21:43][C][api:103]: API Server:
[20:21:43][C][api:104]:   Address: **********
[20:21:44][D][dallas.sensor:144]: 'Living Room Light Temperature': Got Temperature=24.9°C
[20:21:44][D][sensor:092]: 'Living Room Light Temperature': Sending state 24.87500 °F with 1 decimals of accuracy
[20:21:49][D][dallas.sensor:144]: 'Living Room Light Temperature': Got Temperature=24.9°C
[20:21:49][D][sensor:092]: 'Living Room Light Temperature': Sending state 24.93750 °F with 1 decimals of accuracy
[20:21:54][D][dallas.sensor:144]: 'Living Room Light Temperature': Got Temperature=24.8°C
[20:21:54][D][sensor:092]: 'Living Room Light Temperature': Sending state 24.81250 °F with 1 decimals of accuracy
[20:21:59][D][dallas.sensor:144]: 'Living Room Light Temperature': Got Temperature=24.9°C
[20:21:59][D][sensor:092]: 'Living Room Light Temperature': Sending state 24.87500 °F with 1 decimals of accuracy
[20:22:04][D][dallas.sensor:144]: 'Living Room Light Temperature': Got Temperature=24.9°C
```
And then it keeps going with the readings (which appear to be correct).

**Additional information and things you've tried:**
- Restarting Home Assistant
- Restarting the host machine
- Restarting the ESPHome addon
- Restarting the ESP32
- Changing the unit of measurement
- Changing the sensor's name
- Deleting the sensor from the device registry (it comes back "unavailable" again after a restart) 

I've tried all the things I know how to do.  I see nothing out of the ordinary in addon logs or the supervisor logs.  I'm happy to provide any additional details you might need to help.

glmnet, 2019-10-03:
This is not MQTT right? (just to make sure)

Dig deeper:

Please enable debug logging in HA (see logger section) and post what HA is sending to the ESP.

sabeechen, 2019-10-03:
I enabled debug logging in HA, restarted, and saw this in the output:
```
homeassistant.exceptions.HomeAssistantError: Entity id already exists: sensor.living_room_light_transformer_temperature. Platform esphome does not generate unique IDs
```
Which caused me to remember I had set up a device long ago using the same dallas sensor (eg it had the same address and name) on a different device months ago that was no longer connected but still present in the HA integrations page.  I deleted the older device and renamed the temperature sensor in the new device and now it shows up!

Since Ive got this working I'd consider this issue resolved for me, but is it a problem that esphome can generate conflicting ids?  If you're interested I could try to reproduce it again, I'm not sure if it was having the same name or address (or both) that caused the collision.

brandond, 2019-10-07:
This is supposedly a feature - you can move a sensor from device to device and as long as it uses the same ID, HA will track it as it moves. It does lead to some unexpected behavior though in case like this.

sabeechen, 2019-10-07:
Ah, that makes more sense.  For posterity's sake, was it the "id" used for code generation or the "name" field that caused the name collision in Home Assistant?

glmnet, 2019-10-07:
Is it the name, that's what makes an entity public to HA (unless you set it to internal: true) if you don't set id, ESPHome will create one out of the name, if you only set the id, the entity will be hidden to HA. If you set none it fails.

sabeechen, 2019-10-07:
I could have sworn I got the same problem when I changed the name, but I'm probably just mis-remembering.  Thanks for looking into it!

closed 2019-10-07 by sabeechen