esphome/issues #716: Sonoff SV GPIO17

HFsi, 2019-10-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff SV (ESP8266)

**Affected component:**

https://next.esphome.io/devices/sonoff.html#sonoff-sv

**Description of problem:**

In the docs says that the Sonoff SV has an analog input on GPIO17, but I couldn´t find which pin of the board is.
I googled it but nothing appeared neither.
Is the doc correct?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

Not done yet

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

OttoWinter, 2019-10-20:
Yes it is - the ESP8266 has an analog input that has different names. Sometimes it's called A0, sometimes GPIO17, and sometimes also TOUT (or something similar, can't remember).

closed 2019-10-20 by OttoWinter

HFsi, 2019-10-20:
Well, I don´t see any pin labeled like that

[image]

OttoWinter, 2019-10-20:
Well it could also mean one of the pins on the chip itself - idk. Is having more info on the sonoff page an issue?