esphome/issues #718: Socket error on client deurbel-600194361796, disconnecting. - MQTT

Imperial-Guard, 2019-10-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant and ESP on seperate Docker container
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Wemos Mini D1 Pro 
**Affected component:**

https://esphome.io/components/wifi.html
https://esphome.io/components/mqtt

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: deurbel
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: light
  # manual_ip:
  #   static_ip: 192.168.1.100
  #   gateway: 192.168.1.1
  #   subnet: 255.255.255.0

# Enable logging
logger:
  # level: info

mqtt:
  broker: '192.168.1.8'
  discovery: False
  log_topic: hallway/switch/0001/doorbell/message
  birth_message:
    topic: hallway/switch/0001/doorbell/online
    payload: "yes"
  will_message:
    topic: hallway/switch/0001/doorbell/online
    payload: "no"

ota:

binary_sensor:
  - platform: gpio
    pin: 
      number: D2
      inverted: True
      mode: INPUT_PULLUP
    name: "gang_deurbel"
    state_topic: hallway/switch/0001/doorbell/state
    filters:
      - delayed_on: 25ms
      - delayed_off: 2000ms
    on_press:
      then:
        if:
          condition:
            - switch.is_on: deurbel_aan
          then:
                - switch.turn_on: relay
                - delay: 1000ms
                - switch.turn_off: relay
                
globals:
  - id: deurbel_unmute
    type: bool
    restore_value: true
    initial_value: 'true'

sensor:
  - platform: wifi_signal
    name: "gang_deurbel__wifi"
    state_topic: hallway/switch/0001/doorbell/wifi
    update_interval: 60s
    
switch:
  - platform: gpio
    name: "gang_deurbel_relais"
    state_topic: hallway/switch/0001/doorbell/relay
    command_topic: hallway/switch/0001/doorbell/relay/set
    id: relay
    internal: false
    icon: mdi:alarm-bell
    pin: 
      number: D1
      inverted: false

  - platform: restart
    name: "gang_deurbel_herstart"
    command_topic: hallway/switch/0001/doorbell/restart/set
      
  - platform: template
    name: gang_deurbel_aan
    state_topic: hallway/switch/0001/doorbell/switch
    command_topic: hallway/switch/0001/doorbell/switch/set
    id: deurbel_aan
    restore_state: false
    icon: mdi:bell-ring
    turn_on_action:
      - globals.set:
          id: deurbel_unmute
          value: 'true'
    turn_off_action:
      - globals.set:
          id: deurbel_unmute
          value: 'false'
    lambda: |-
      return id(deurbel_unmute);
      
status_led:
  pin:
    number: D4
    inverted: yes

```

**Logs (if applicable):**

```
1570165765: Client deurbel-600194361796 has exceeded timeout, disconnecting. 1570165765: Socket error on client deurbel-600194361796, disconnecting.

```

**Additional information and things you've tried:**
I added the power_save_mode and changed it to light and removed the static ip from the code

brandond, 2019-10-06:
Please connect it via USB/serial and see what it logs. Unless it's crashing and rebooting, the most likely culprit is poor wifi coverage.

Imperial-Guard, 2019-10-06:
> Please connect it via USB/serial and see what it logs. Unless it's crashing and rebooting, the most likely culprit is poor wifi coverage.

With my code made in ArduinoIDE and flashed I don't have any issues, while the ESP isn't moved. 
Wifi signal strenght is around -62.0 dB. 

The problem is while testing it on the serial I don't have any issues, but when it's placed back on it's originnal position I got those disconnecting issues.

OttoWinter, 2019-10-20:
Please also see https://github.com/esphome/issues/issues/455

Without debug logs we can't help you, sorry.

lancelon, 2020-11-17:
@Imperial-Guard did you ever sort this out?

closed 2021-03-31 by stale[bot]