esphome/issues #720: Strobe effect does't work with binary light

MaximumSU, 2019-10-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Home Assistant

**ESP (ESP32/ESP8266, Board/Sonoff):**
Sonoff (platform: ESP8266, board: esp01_1m)

esphomeyaml 1.13.6
**Affected component:**

[https://esphome.io/components/light/binary.html](url)

**Description of problem:**
Strobe effect doesn't work. Light just on and lit, no any effects.

**Problem-relevant YAML-configuration entries:**
```yaml
light:
  - platform: binary
    name: s_bath_1f_sm_led
    id: "led"
    output: blue_led
    effects:
      - strobe:
          name: blinking_500x500
          colors: 
            - state: true
              duration: 500ms
            - state: false
              duration: 500ms
```

**Logs (if applicable):**
 Log show no problems.

```
[17:42:03][D][light:261]: 's_bath_2f_sm_led' Setting:
[17:42:03][D][light:270]:   State: ON
[17:42:05][D][light:261]: 's_bath_2f_sm_led' Setting:
[17:42:05][D][light:323]:   Effect: 'blinking_500x500
```

**Additional information and things you've tried:**
In the previous releases (1.12, or 1.11, i don't remember exactly) all worked as expected - blinked with 500ms period.

glmnet, 2019-10-07:
Did you try dev branch?

MaximumSU, 2019-10-07:
No. It makes sense? This is production environment, i do not want to experiment on it. I use the lambda effect as workaround now.

closed 2019-10-17 by OttoWinter