esphome/issues #721: Automation Light Effect doesn't work

Fusseldieb, 2019-10-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

NodeMCU v3 4MB

**Affected component:**

https://esphome.io/components/light/index.html#automation-light-effect

**Description of problem:**
If I define a "automation light effect", it just doesn't work. The light stays still and doesn't do anything (Like it's set to Effect None). All other effects like Rainbow, Flicker, etc work just fine.

**Problem-relevant YAML-configuration entries:**
```yaml
light:
  - platform: fastled_clockless
    chipset: WS2811
    pin: D5
    num_leds: 150
    rgb_order: BRG
    name: "Cozinha LED Teto"
    color_correct: [100%,100%,100%]
    id: cozinha_led_teto
    effects:
      - addressable_twinkle:
      - addressable_scan:
      - addressable_color_wipe:
      - addressable_rainbow:
      - flicker:
      - strobe:
      - random:
      - automation:
          name: Custom Automation Effect
          sequence:
            - light.addressable_set:
                id: cozinha_led_teto
                red: 100%
                green: 100%
                blue: 100%
            - delay: 100ms
            - light.addressable_set:
                id: cozinha_led_teto
                range_from: 0
                range_to: 20
                red: 100%
                green: 0%
                blue: 0%
```

**Logs (if applicable):**

```
[23:46:51][D][light:261]: 'Cozinha LED Teto' Setting:
[23:46:51][D][light:323]:   Effect: 'Custom Automation Effect'
[23:46:53][D][light:261]: 'Cozinha LED Teto' Setting:
[23:46:53][D][light:323]:   Effect: 'None'
```

**Additional information and things you've tried:**
The Lambda Effect apparently works.

glmnet, 2019-10-10:
are you testing with dev version too?

OttoWinter, 2019-10-17:
Yeah, closing as the changes with #568 probably fixes this. Please reopen if not.

closed 2019-10-17 by OttoWinter