esphome/issues #723: Hyphen should be allowed in names.

dzikus, 2019-10-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** pip

**ESP (ESP32/ESP8266, Board/Sonoff):** any

**Affected component:** DNS system

**Description of problem:**

>Oh noes, "esp-dgm-test" isn't a valid name. Names can only include numbers, letters and **underscores**.

It is wrong because DNS system doesn't recommend underscore char in names. It is allowed and common in use like _tcp _sip _ldap but not recommended in normal names. Try to register domain with underscore.

Hyphen should be allowed. I'm using hyphen in my DNS and cannot use it with ESPHome without many changes. Tasmota, ESPEasy migration is also hard because of that (they don't have such strange limit)

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: esp-dgm-test
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "123"
  password: "456"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO3
      mode: INPUT_PULLUP
      inverted: true
    name: "Button"
    on_press:
      - switch.toggle: relay

sensor:
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: true
    cf_pin: 4
    cf1_pin: 5
    current:
      name: "Current"
    voltage:
      name: "Voltage"
    power:
      name: "Power"
    update_interval: 1s
    voltage_divider: 800
    current_resistor: 0.0023

switch:
  - platform: gpio
    name: "Relay"
    pin: GPIO14
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes

```

**Logs (if applicable):**

```
INFO Reading configuration...
Failed config

esphome: [source dgm-test.yaml:2]

  '-' is an invalid character for names. Valid characters are: abcdefghijklmnopqrstuvwxyz0123456789_ (lowercase, no spaces).
  name: esp-dgm-test [source dgm-test.yaml:2]
  platform: ESP8266
  board: esp01_1m
```

**Additional information and things you've tried:**

glmnet, 2019-10-16:
Please see https://github.com/esphome/feature-requests/issues/418

OttoWinter, 2019-10-17:
Yeah, moving to esphome/feature-requests#418 - I know it's a bit against DNS but it's also not really possible to change it now (there was some things that required no hyphens, but I can't recall what exactly again).

closed 2019-10-17 by OttoWinter