esphome/issues #727: Feature request: OTA with a target IP

Nornode, 2019-10-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker & Command Line

**ESP (ESP32/ESP8266, Board/Sonoff):**

Any ESP
**Affected component:**

Background: my mDNS isn't working as expected and I have two separate (v)LAN's, a Standard and one for all IoT's and ESPs.
I can reach and add the flashed ESPs to Homeassistant, all fine.
But ESPhome command line/dashboard can't see the devices so I'm unable to update them OTA/see in the dashboard if they are online.

I tried to edit the json file .esphome/config.yaml.json to have:

```
{
  "arduino_version": "espressif8266@1.8.0",
  "address": "192.168.12.51",
  "storage_version": 1,
  "esphome_version": "1.13.6",
...
...
```
However that didn't work and I still can't e.g. read logs from the dashboard... it still says:

I'd like to specify the ESPhome's local ip in the dashboard and in the command line for opening up a direct connection like:

esphome config.yaml run/logs -i 192.168.12.52

Thanks!

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
INFO Reading configuration...
WARNING Could not auto-detect timezone. Using UTC...
INFO Detected timezone 'UTC' with UTC offset 0
INFO Starting log output from 8ch_relay.local using esphome API
INFO Connecting to 8ch_relay.local:6053 (PUB-IP)

```

glmnet, 2019-10-09:
If you want to OTA using command line to a custom IP you can do:

esphome file.yaml run —upload-port <ip>

I guess this works for logs too. 

This request is too vague, please be more specific or it may be discarded

glmnet, 2019-10-09:
Also feature request belongs to the feature request repo. Please open one there

closed 2019-10-09 by glmnet