esphome/issues #728: Default example on clean Linux does not run

irgendwienet, 2019-10-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
 - Ubuntu 18.04
 - phyton 2.7
 - pip

Clean install in a virtual machine to verify the problem. 

Only base system, python und `pip install esphome` (as described in the tutorial)

**ESP (ESP32/ESP8266, Board/Sonoff):**

 /

**Affected component:**

 /

**Description of problem:**

I just ran `esphome` from bash:

```
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 7, in <module>
    from esphome.__main__ import main
  File "/usr/local/lib/python2.7/dist-packages/esphome/__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "/usr/local/lib/python2.7/dist-packages/esphome/writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "/usr/local/lib/python2.7/dist-packages/esphome/config.py", line 15, in <module>
    from esphome import core, core_config, yaml_util
  File "/usr/local/lib/python2.7/dist-packages/esphome/core_config.py", line 6, in <module>
    import esphome.config_validation as cv
  File "/usr/local/lib/python2.7/dist-packages/esphome/config_validation.py", line 1242, in <module>
    Optional(CONF_COMMAND_TOPIC): All(requires_component('mqtt'), subscribe_topic),
  File "/usr/local/lib/python2.7/dist-packages/esphome/voluptuous_schema.py", line 198, in extend
    ret = super(_Schema, self).extend(schema, extra=extra)
  File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 782, in extend
    return result_cls(result, required=result_required, extra=result_extra)
TypeError: __init__() got an unexpected keyword argument 'required'
```

**Problem-relevant YAML-configuration entries:**
```yaml
 - 

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ilyaglar, 2019-10-10:
same error

glmnet, 2019-10-10:
duplicate of #666

closed 2019-10-10 by glmnet