esphome/issues #729: Ping doesn't work outside of local domain, device shown as offline

Diaoul, 2019-10-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266

**Affected component:**
Ping in dashboard

**Description of problem:**
As the title says, the ping is failing and the device shows as offline in the dashboard when everything else work. I can actually ping my device at `myesp.mydomain` from my computer and from the esphome container when I `docker exec` in it so no DNS or network issue here.

**Problem-relevant YAML-configuration entries:**
```yaml
# api component not used
wifi:
  domain: .mydomain
```

**Logs (if applicable):**
```
Nothing in the logs
```

**Additional information and things you've tried:**
Looks like `.local` is hardcoded [here](https://github.com/esphome/esphome/blob/master/esphome/dashboard/dashboard.py#L471) instead of reusing the wifi setting from the configuration.

Diaoul, 2019-10-11:
Maybe related to #151 but I don't know if this applied to Hassio.

Diaoul, 2019-10-11:
Works with `status_use_ping`, so I guess this can be closed?

Diaoul, 2019-10-11:
It feels strange to send mDNS requests to check if a device is online or offline. Is this common practice? What are the pro/cons of this compared to a regular ping? (sorry if this is a stupid question, I'm not well versed into networking)

I've found a pure python implementation of the ping ICMP request [here](https://github.com/kyan001/ping3), that might avoid subprocess calls and improve portability.

back0nline, 2019-10-14:
mDNS its the equipment that sends a multicast saying "im here this is my name". In the same way other equipments can see what equipments are available on the network.
This is how apple airplay works.
With ping, you send a ping an wait for a replay.
So, mDNS is something similar with a DNS and ping its a command pront utilitary.

OttoWinter, 2019-10-15:
Closing as per https://github.com/esphome/issues/issues/729#issuecomment-540935273

> What are the pro/cons of this compared to a regular ping? 

1. You would need to do an mDNS lookup anyway, if you ping a .local domain the IP address is usually looked up through mDNS

2. Ability to attach TXT records, like the current version of the device.

closed 2019-10-15 by OttoWinter

Diaoul, 2019-10-15:
Makes sense. However this only work if:
1. You are on the same network or allow mDNS across multiple VLANs
2. For the .local domain

Maybe this is niche but how about fallback on ping when wifi domain option is enabled?

OttoWinter, 2019-10-17:
> Maybe this is niche but how about fallback on ping when wifi domain option is enabled?

Hmm, that may be an option, yes.