esphome/issues #734: OTA times out and fails

teeling, 2019-10-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Running through Docker, esphome/esphome and esphome/esphome:dev

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266, esp01_1m
The unit is a small smart plug from a company called Nedis. It was tuya-converted and flashed Tasmota, and after that manually uploaded a ESPhome bin, which works.

**Description of problem:**
The ESPHome bin manually uploaded from Tasmota works as expected, except I now can't OTA update.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bathroom_heater_plug
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "myssid"
  password: "anothersupersecretpasswordhere"
  
  # Optional manual IP
  manual_ip:
    static_ip: xxx.xxx.xxx.xxx
    gateway: xxx.xxx.xxx.xxx
    subnet: 255.255.255.0

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  password: "supersecretpasswordhere"

ota:
  safe_mode: True
  password: "supersecretpasswordhere"

switch:
  - platform: gpio
    name: "Bathroom heater"
    pin: GPIO14
    id: relay

```

**Logs (if applicable):**

```
Compile and upload:
Linking .pioenvs/bathroom_heater_plug/firmware.elf
Retrieving maximum program size .pioenvs/bathroom_heater_plug/firmware.elf
Building .pioenvs/bathroom_heater_plug/firmware.bin
Checking size .pioenvs/bathroom_heater_plug/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  45.2% (used 37032 bytes from 81920 bytes)
PROGRAM: [===       ]  33.5% (used 343028 bytes from 1023984 bytes)
====================================== [SUCCESS] Took 15.38 seconds ======================================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.50
INFO Uploading /config/bathroom_heater_plug/.pioenvs/bathroom_heater_plug/firmware.bin (347168 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error receiving acknowledge receive OK: timed out

Log from OTA on device:
INFO Starting log output from xxx.xxx.xxx.xxx using esphome API
INFO Connecting to xxx.xxx.xxx.xxx:6053 (xxx.xxx.xxx.xxx)
INFO Successfully connected to xxx.xxx.xxx.xxx
[19:42:26][I][app:096]: esphome version 1.13.6 compiled on Oct 13 2019, 17:04:32
[19:42:26][C][wifi:372]: WiFi:
[19:42:26][C][wifi:254]:   SSID: 'myssid'
[19:42:26][C][wifi:255]:   IP Address: xxx.xxx.xxx.xxx
[19:42:26][C][wifi:257]:   BSSID: 14:91:82:C2:65:F7
[19:42:26][C][wifi:258]:   Hostname: 'bathroom_heater'
[19:42:26][C][wifi:262]:   Signal strength: -49 dB ▂▄▆█
[19:42:26][C][wifi:263]:   Channel: 6
[19:42:26][C][wifi:264]:   Subnet: 255.255.255.0
[19:42:26][C][wifi:265]:   Gateway: xxx.xxx.xxx.xxx
[19:42:26][C][wifi:266]:   DNS1: 0.0.0.0
[19:42:26][C][wifi:267]:   DNS2: 0.0.0.0
[19:42:26][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Power Button'
[19:42:26][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[19:42:26][C][switch.gpio:042]: GPIO Switch 'red_led'
[19:42:26][C][switch.gpio:043]:   Pin: GPIO13 (Mode: OUTPUT, INVERTED)
[19:42:26][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[19:42:26][C][switch.gpio:042]: GPIO Switch 'Bathroom heater'
[19:42:26][C][switch.gpio:043]:   Pin: GPIO14 (Mode: OUTPUT)
[19:42:26][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[19:42:26][C][logger:137]: Logger:
[19:42:26][C][logger:138]:   Level: DEBUG
[19:42:26][C][logger:139]:   Log Baud Rate: 115200
[19:42:26][C][logger:140]:   Hardware UART: UART0
[19:42:26][C][status:034]: Status Binary Sensor 'Status'
[19:42:26][C][status:034]:   Device Class: 'connectivity'
[19:42:26][C][ota:029]: Over-The-Air Updates:
[19:42:26][C][ota:030]:   Address: xxx.xxx.xxx.xxx:8266
[19:42:26][C][ota:032]:   Using Password.
[19:42:26][C][api:103]: API Server:
[19:42:26][C][api:104]:   Address: xxx.xxx.xxx.xxx:6053
[19:43:10][D][ota:072]: Starting OTA Update from xxx.xxx.xxx.xxx...
[19:43:11][D][ota:243]: OTA in progress: 0.3%
[19:43:12][D][ota:243]: OTA in progress: 4.9%
[19:43:13][D][ota:243]: OTA in progress: 8.3%
[19:43:14][D][ota:243]: OTA in progress: 12.0%
[19:43:15][D][ota:243]: OTA in progress: 15.4%
[19:43:16][D][ota:243]: OTA in progress: 19.1%
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to xxx.xxx.xxx.xxx:6053 (xxx.xxx.xxx.xxx)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to xxx.xxx.xxx.xxx:6053 (xxx.xxx.xxx.xxx)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to xxx.xxx.xxx.xxx:6053 (xxx.xxx.xxx.xxx)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 2 seconds
INFO Connecting to xxx.xxx.xxx.xxx:6053 (xxx.xxx.xxx.xxx)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 3 seconds
INFO Connecting to xxx.xxx.xxx.xxx:6053 (xxx.xxx.xxx.xxx)
INFO Successfully connected to xxx.xxx.xxx.xxx

```

**Additional information and things you've tried:**

Tried to do the OTA with both the stable docker image and dev. Also through command line and through the ESPHome dashboard, all to the same results.
When I do the OTA, the device restarts, but it does not appear to have caught the new bin-file.

OttoWinter, 2019-10-14:
Please see #455

closed 2019-10-14 by OttoWinter

teeling, 2019-10-14:
@OttoWinter That does not seem like the same issue to me. In #455 they seem to not see the device on the network, while in my case, the device is working properly, and can be controlled through Home assistant, I just cant seem to upload a new firmware through the OTA function.
Am I missing something?

OttoWinter, 2019-10-14:
@teeling So the device is not dropping of the network while doing an OTA? (if it were, then it would fit into #455). 

Can you also look at the logs while trying the OTA? Does the "sender" reach 100% while the ESP itself is only at ~20%?

teeling, 2019-10-14:
Thanks for your reply! The sender, uploading progress bar, reaches 100% very quickly. Sometimes instantly, and sometimes it instantly goes to 95-ish % and then 100 after less than a second.
The receiver logs the progress for roughly between 10-20 seconds until it fails with the "WARNING Disconnected from API: Timeout while waiting for message response!"
Sometimes it reaches 30-ish % and, as in the case pasted above, only 19%. My uneducated guess would be that something goes wrong around there, making the device reboot, starting up the previous firmware. After the restart I can connect the device again, but no sign of the new firmware.

OttoWinter, 2019-10-14:
Ok, does this seem similar: https://github.com/esphome/issues/issues/114#issuecomment-475323092 ?

teeling, 2019-10-14:
The behaviour looks similar, but I don't get any error like they do in #114 :
`[15:07:59][W][ota:386]: Update end failed! Error: ERROR[0]: No Error`
or
`[15:08:33][W][ota:314]: Preparing OTA partition failed! 'ERROR[0]: No Error`

Mine just throws:
`WARNING Disconnected from API: Timeout while waiting for message response!`
and then starts to try to reconnect while the device seems to restart.

I've tried to minimise the config, and even trying to upload a minimal Tasmota bin, but get the same behaviour.

OttoWinter, 2019-10-15:
> [15:07:59][W][ota:386]: Update end failed! Error: ERROR[0]: No Error
> or
> [15:08:33][W][ota:314]: Preparing OTA partition failed! 'ERROR[0]: No Error

Those are messages from over the USB cable, you probably will see the same ones if you would connect the device via USB and view the logs that way.

> Mine just throws:
> WARNING Disconnected from API: Timeout while waiting for message response!

That would be because the device is dropping of the network, see also #455