esphome/issues #735: Problems with esp32_ble_tracker an Xiaomi CGG1

Alex9779, 2019-10-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io and pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

https://esphome.io/components/esp32_ble_tracker.html

**Description of problem:**
I just added a PR for adding a sensor for the Xiaomi CGG! bluetooth sensor.
My code works fine so far but I had troubles getting all the brodcasts messages from the device.
I had toi adjust the BLE scan_:interval and scan_window here:
https://github.com/esphome/esphome/blob/dev/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp#L161
I reduce the entries to 0x10 and 0x10 which are the defaults as the comment states and did some research in Espressif's docs. I have really no idea what the comment talks about becasue it seems not to be up-to-date, 0x200 and 0x30 is there normally but the comment mentions different defaults. I tried with other values and 0x50/0x50 worked, 0x30/0x30 too.
I also have a problem with the naming of the folloing "scan interval" which is also changeable with a YAML configuration, in Espressif's docs this value is called "duration" of the scan. This is a bit misleading when reading the ESPHOME code, you have two variables called nearly the same but doing different things.
So the point of this issue is I wanted clarification why are the values the way they are, how do my changes affect stability (in my environment the scanner works fine for a day now), maybe someone knows why the normal values do not show my any brodcasts of the device but when I set them the same value then it works and maybe some reasons to change the code, rename "scan interval" to "scan duration" and make the other two available in the ble tracker config via YAML.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-10-14:
YAML scan_interval: The interval in seconds that the BLE scan is restarted (how long `esp_ble_gap_start_scanning` is active). To be honest, I don't think that value changes anything now that I know more about BLE and is just an artifact brought in from espressif's APIs

scan_params scan_interval/scan_window: The classic BLE scan interval/window settings. window is how much of the time the device is actively listening, and interval how often active scan requests are made. Descriptions of what these values really mean are in espressif's docs.

The reason it got set this low was that higher values degraded stability of the system noticeably. Things may have changed in the last few esp-idf updates, but will require testing before we can make higher values the default.

Alex9779, 2019-10-14:
Just to get this right, 0x200 is low and 0x10 is high?

OttoWinter, 2019-10-14:
Depends on which value you mean. If I remember correctly I think it actually was the value of `scan_window/scan_interval` (so what percentage of the time the scan is active). But the esp-idf docs are probably the best reference here.

Alex9779, 2019-10-14:
As far as I can see there you configure a multiplicator for 0.625 milliseconds, so 0x10 is 16*0.625ms=10ms so small numbers are shorter time intervals, lowest possible value is 0x004 (2.5ms)...

I just find now real docs about what that really does as I see to just be able to get the broadcast message of that specific device if the values are euqal, if I have a smaller window then it does not work...

Alex9779, 2019-10-14:
@sermayoral maybe you have some experience with the device, did you have such issues? I am still wondering if that may be because of about the ten other devices sending broadcasts around my house, mostly Apple devices and other home electronic...

OttoWinter, 2019-10-15:
See also https://electronics.stackexchange.com/questions/82098/ble-scan-interval-and-window http://dev.ti.com/tirex/content/simplelink_academy_cc2640r2sdk_1_12_01_16/modules/ble_scan_adv_basic/ble_scan_adv_basic.html

AFAIK, the multiplicator itself doesn't have that much of an influence (probably mostly with packet collision avoidance I think), but the ratio of the two does. The ratio of the two defines what percentage of the time it is listening for ADV packets.

Alex9779, 2019-10-15:
I found also an old post on the Espressif forum where someone explains just as you did. The windows is the fraction of the interval where actually is being scanned and the duration is the overall time where the interval is repeated...
Will link later but I also habe no real idea what that all means and why I get the messages of that specific device just when I set them to euqal numbers.
I have some more ordered but maybe someone else can enlighten us or tell if he got problems too with the default numbers...

Alex9779, 2019-10-16:
This is the thread I mentioned: https://www.esp32.com/viewtopic.php?t=2291

Alex9779, 2019-10-16:
You have any idea in detail how the listening works and when it raises its events?
Only thing at the moment I can think of why it does not work with that device for me is that the interval is quite long (320ms) and the window just 30ms, about 10% and that the component just raises the scan results after the interval but there are too many broadcasts and it can't handle them all.
Ok but that would not explain why it works with 0x50/0x50, in 0x50*0.625ms a lot more packets must arrive than in 30ms...
I really have no idea...

Alex9779, 2019-10-16:
I did some more tests with different values and I think I have found something.
It seems that if the windows is significantly smaller than of the interval then I won't receive the messages from the Xiaomi ClearGrass CGG1.
For example:
- 0x200/0x30 (default) returns **no** messages
- 0x200/0x100 returns messages
- 0x200/0xF0 returns messages
- 0x200/0xB0 returns messages
- 0x200/0x90 returns **no** messages
- 0x40/0x30 returns messages
- 0x40/0x1F return **no** messages
- 0x10/0x10 returns messages

OttoWinter, 2019-10-16:
Exactly, as I said it's the ratio between the two. The scan_window itself probably doesn't have a big impact (though in theory long frames could cause issues), but the percentage of time the scan is active (as denoted by the ratio) is the important factor.

Alex9779, 2019-10-16:
I think it also affects my Xiaomi MiJia (LYWSDCGQ) too. The important message with humidity and temperature arrive mostly but I have/had issues with the battery message.
If I manipulate the values to get the messages from the CGG1 then I also get constantly the battery messages.

So what now? just increase the window? Lower both to what Apple has as stated in the one post (40ms/30ms)???
How can I test? Is the logger pushing out problems or errors when something with that BLE stuff fails? If so I could setup one or two test devices and monitor their stability...

sermayoral, 2019-10-16:
Same behaviour in Xiaomi Mijia, bur i dont see a problem there. I get temperature and humidity levels often. Battery level each... 30 minutes? Ok, but i think its not an impontant issue because the battery level changes once in a blue moon... Don't you think so?

Talking about LYWSDCGQ. Do you know why this device does not sent battery level in ESPHome. I saw that in Openmqttgateway the battery level is sent.

OttoWinter, 2019-10-17:
> How can I test? Is the logger pushing out problems or errors when something with that BLE stuff fails? If so I could setup one or two test devices and monitor their stability...

The main problem was that these higher values made WiFi more unstable. To test, I would recommend putting the device somewhere where it doesn't necessarily have a great WiFi signal. Then, use the status binary sensor to see how many times it drops off the network (or with what frequency)

Alex9779, 2019-10-17:
I have two test devices running for about 16 hours now with different values one with 0x40/0x30 the other with 0x100/0xC0, the last one produces more messages as expected but without Wifi and other stuff so...
Will upgrade the test to check the wifi and see how it goes.
As another test I upgraded my two production devices monitoring two different LYWSDCGQ's and each controlling a fan, both now with 0x100/0xC0, so there I will see too...

Alex9779, 2019-10-17:
> Same behaviour in Xiaomi Mijia, but i dont see a problem here. I get temperature and humidity levels often. Battery level each... 30 minutes? Ok, but i think its not a problem because the battery level changes once in a blue moon... Don't you think so?
> 
> Talking about LYWSDCGQ. Do you know why LYWSD02 model does not send battery level in ESPHome? I saw that in Openmqttgateway the battery level is sent in this device.
> 
> Last comment in [esphome/esphome#664](https://github.com/esphome/esphome/pull/664)

As far as I know from other sources the LYWSD02 does not send battery information, read that somewhere...

And looking at the logs of my running test the LYWSDCGQ send battery info about every minute while the CGG1 every 10 seconds...

sermayoral, 2019-10-17:
Perfect @Alex9779. Quite more than necessary :-)

Alex9779, 2019-10-18:
Ok i setup both trackers with wifi now, the 100/C0 I packed in a box reducing wifi strength to about -77dbm, bound both to my HA with a sensor for connection and wifi strenght and will monitor now...

Alex9779, 2019-10-18:
I have some first results, the device with 40/30 has really problems.
At first it ran about 30 minutes then the wifi disconnected and didn't come back.
After a reboot it ran longer but the BLE scan broke somehow, I still got the staring scan log messages and two messages from random devices but nothing more, it connected to the wifi and reported the signal strength...
The device with 100/C0 runs without any losses or problems for about 3 hours now... And it gets all the messages from all devices still...

Alex9779, 2019-10-18:
I am now going to run 100/C0 on the 40/30 device too to see if there is no hardware issue causing that, but so far 100/C0 looks promising...

OttoWinter, 2019-10-18:
Ok, so if it ends up working well that would be great (if you could, please also test with latest dev, I just updated the arduino framework to the latest version; hopefully esp-idf has not had a regression with BLE).

As for the values, we know that higher ratios between window and interval will yield better results. Then of course is the question what value we should choose (0x50 or 0x400 for example). There are three BLE advertising channels (37, 38, 39), and we would like each of them to be scanned at least once every few seconds (presence detection).

So I would say a scan interval of a ~300ms would be good - at most you would then see around 3*300ms= ~1s of latency from device in range to result. I don't have a rig to test, but it would be interesting to see values around 0x100-0x200 for scan interval and 0xF0 to 0x150 for scan window.

OttoWinter, 2019-10-18:
On second thought, I think I'll just add options for that to the YAML and configure the defaults now - the beta phase will leave enough time to figure out if there's a major flaw.

Alex9779, 2019-10-18:
once that is merged I will pull dev and start another test with the values you proposed starting with 0x200/0x150 i think...

closed 2019-10-18 by OttoWinter

Alex9779, 2019-10-19:
seems the default values now run stable and get also the messages from the Xiaomi CGG1 Cleargrass sensor...
One issue, with new dev I had to update my devices using the BLE tracker again with the cable, OTA didn't work anymore, after that OTA worked again, is that a known issue? I read that initially when using that component you have to flash by cable because of some resizing going on but the new dev needed that too...

OttoWinter, 2019-10-19:
> One issue, with new dev I had to update my devices using the BLE tracker again with the cable, OTA didn't work anymore

Oh that should not happen. Will have to test that migration path later. Could you post the error that it failed with here?

Alex9779, 2019-10-19:
No error, at my first tries the progress went to 2% then some data timeout error came up, later progress didn't even start, reported a timeout, one of my four tested devices worked so, no idea why...

OttoWinter, 2019-10-20:
Huh, and the device was running the old version of esphome while you were uploading the dev version?

Alex9779, 2019-10-20:
Yeah, really strange, one of the four is the test device for all others, so I test changes with that one first, I tried to upload and got the timeout. Then I tried the second one, timeout, the thrid one I flashed by cable right away, and I have no idea why I tried the fourth again with OTA first but this one worked...
If you can‘t‘ reproduce it might be some device specific stuff, they have been up for more than a week without a reboot, maybe that had an influence, I didn‘t try to reboot them and then try OTA again...

Alex9779, 2019-10-29:
Just a small addition, I run my devices now for a week without an issue, no connection losses to wifi or BLE hangs.
Not sure about the initial update problem, didn't invetigate that further... If anyone else reports something similar this was likely just a temporary problem with my devices...

diraimondo, 2019-11-01:
I just bought two Xiaomi ClearGlass and I'm trying to make them work with EspHome 1.14.0 and ble_tracker. I can't see any related reads. During the scan I can recognize two similar mac address which prefix belong to Xiaomi but no value is reported. @Alex9779 reported it is working for him with default ble tracker params. 

I'm wondering if I need to associate the sensors with the related Android app to make them broadcast values. 

Any help would be great!

Alex9779, 2019-11-01:
@diraimondo hmmm see #799, maybe it is that problem I just faced too...

diraimondo, 2019-11-02:
> @diraimondo hmmm see #799, maybe it is that problem I just faced too...

Yes, it could be: I'll check.

OttoWinter, 2019-11-05:
We might have a case of the new defaults causing some issues. Ref: https://community.home-assistant.io/t/ble-tracker-xiaomi-miflora-and-too-big-firmware/146485 :(

Will wait for confirmation from that thread though first before acting on it. And if it turns out to be true maybe just an update to the docs is fine.