esphome/issues #740: Fallback AP instable / not working / loosing connection ESP8266

SimplyRun, 2019-10-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.): Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff): ESP8266 (Tuya / Gosund SP111 flashed with ESPHome)
I use ESPHome (dev) in Hass.io: Version 1.13.6

Current config:
`
> 
esphome:
  name: p3gosund
  platform: ESP8266
  board: esp8285

wifi:
  ssid: "pidri"
  password: "12345678"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "p3gosund_Fallback_AP"
    password: "12345678"

captive_portal:

# Enable logging
logger: 
  level: DEBUG

# Enable Home Assistant API
api:
  password: "Update"

ota:
  password: "Update"
> 
`

**Affected component:
Unfortunetly I only found the issue and the implementation of it, but no documentation.

**Description of problem:

I intended to have the ap enabled with the capative_portal enabled.
Unfortunetly I only found the issue and the implementation of it, but no documentation.
Reference feature request: https://github.com/esphome/feature-requests/issues/86
Reference implementation: #624

Issue:
When connecting to the accesspoint "p3gosund_Fallback_AP" it is often not getting connected and when it connects, the IP: 192.168.4.1 is not accessible.
After some seconds the connection to the AP is lost and I have to reconnect.

1.) Is the documentation about "captive portal" already somewhere?
2.) I know the Wifi Manager is somehow doing some ugly things to get ESP stable for AP, has this implementation similiar modifications or is it maybe required?
3.) Unfortunately I am not aware how to get the logs, I have a serial connector here for the plug, but how can I get the logs of it?

Any help is highly appreciated.

Many thanks! 👍

OttoWinter, 2019-10-18:
> When connecting to the accesspoint "p3gosund_Fallback_AP" it is often not getting connected and when it connects, the IP: 192.168.4.1 is not accessible.

What device is this you're connecting with? I've heard that some devices have problems connecting to ESP8266s in general, would be good to know to diagnose.

> Is the documentation about "captive portal" already somewhere?

next.esphome.io

> I know the Wifi Manager is somehow doing some ugly things to get ESP stable for AP, has this implementation similiar modifications or is it maybe required?

Anything in particular? ESPHome does have a lot of fixes for issues in the ESP-SDK too (and works directly with the ESP-SDK, not through the arduino framework).

> Unfortunately I am not aware how to get the logs, I have a serial connector here for the plug, but how can I get the logs of it?

Depends on the device - most likely it includes opening the device up unfortunately.

SimplyRun, 2019-10-19:
> What device is this you're connecting with? I've heard that some devices have problems connecting to ESP8266s in general, would be good to know to diagnose.
Doesn't matter to be honest, I tried notebook with Windows and Android. Same behavior.
As I just started with ESPHome and Home Assistant, I flashed the TUYA Plugs.
AVATAR AWP07L and Gosund SP111
For both plugs the same behaviour.

> > Unfortunately I am not aware how to get the logs, I have a serial connector here for the plug, but how can I get the logs of it?
> 
> Depends on the device - most likely it includes opening the device up unfortunately.

This is no issue, both types have screws and for the GOSUND SP111 I have rebuild an adapter similiar to this:
[https://www.ebay.de/itm/264465731444](url)
Flashing in any case was possible, but as said, I am just starting with this stuff, so I am still very slow as I have to read and learn by reading a lot. :)

OttoWinter, 2019-10-20:
Fixed in https://github.com/esphome/esphome/pull/766

closed 2019-10-20 by OttoWinter