esphome/issues #742: Try to update bricked device with “OTA safe mode”

Adorem, 2019-10-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io HassOS 3.5
Home Assistant 0.100.2
ESPHome 1.13.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Arilux AL-LC11

**Affected component:**

https://esphome.io/components/remote_receiver.html
https://esphome.io/components/ota.html

**Description of problem:**
For test purpose I tried to increase the buffer_size of the component remote_receiver.
After this my device is still available on the network (reply to ping) but I can't access to log or upload new version via OTA.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: led_xxx
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "xxx"
  password: "xxx"

sensor:
  - platform: wifi_signal
    name: "Signal WiFi LED xxx"
    update_interval: 60s
  - platform: uptime
    name: Uptime LED xxx

binary_sensor:
  - platform: status
    name: "Status LED xxx"

remote_receiver:
  pin:
    number: GPIO04
    inverted: True
  dump: all
  buffer_size: 10000

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

light:
  - platform: rgb
    name: "LED xxx"
    red: output_component1
    green: output_component2
    blue: output_component3
    id: led_strip

output:
  - platform: esp8266_pwm
    id: output_component1
    pin: GPIO5
  - platform: esp8266_pwm
    id: output_component2
    pin: GPIO13
  - platform: esp8266_pwm
    id: output_component3
    pin: GPIO12
```

**Logs (if applicable):**

**Compile And Upload led_xxx.yaml**
```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/led_xxx
Processing led_xxx (framework: arduino; platform: espressif8266@1.8.0; board: esp01_1m)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Retrieving maximum program size /data/led_xxx/.pioenvs/led_xxx/firmware.elf
Checking size /data/led_xxx/.pioenvs/led_xxx/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  40.7% (used 33320 bytes from 81920 bytes)
PROGRAM: [===       ]  34.7% (used 354824 bytes from 1023984 bytes)
========================= [SUCCESS] Took 10.42 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of led_xxx.local
INFO  -> 192.168.1.13
INFO Uploading /data/led_xxx/.pioenvs/led_xxx/firmware.bin (358976 bytes)
ERROR Error receiving acknowledge version: timed out
```
**Show Logs led_xxx.yaml**
```
INFO Reading configuration...
INFO Starting log output from led_xxx.local using esphome API
INFO Connecting to led_xxx.local:6053 (192.168.1.13)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 1 seconds
INFO Connecting to led_xxx.local:6053 (192.168.1.13)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 2 seconds
INFO Connecting to led_xxx.local:6053 (192.168.1.13)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 4 seconds
```

During these two steps upload and show the device is answering to ping command on the network.

**Additional information and things you've tried:**
If possible I would like to avoid to have to update via serial port and then do it via OTA.
I see that OTA offer an “OTA safe mode”. Is there a way to for my device to enter in this mode?
I try to power on the device only for some seconds and then power off and run this cycle more that 10 times but I do not success to enter in “OTA safe mode”.
How can I update my device?

Thanks,
Christophe

OttoWinter, 2019-10-18:
> ERROR Error receiving acknowledge version: timed out

Huh, that means the ESP is not sending any data back on that socket.

> During these two steps upload and show the device is answering to ping command on the network.

Can you make sure it's the device you're pinging? If you've got two devices with the same IP on the network I would suspect something similar happening (each one disconnecting each other)

> Is there a way to for my device to enter in this mode?

Yes, if you let it stay there for a while it will automatically go in that mode. It triggers when the device:
 a) Reboots 10 times in a row (what happens when code crashes)
 b) Home Assistant doesn't connect to the device within 5 minutes for a few times in a row.

So basically, the main way to get it into that mode is to let it stay there for a while. If you have a `status_led` component, you will also see it blinking rapidly in that mode.

> I try to power on the device only for some seconds and then power off and run this cycle more that 10 times but I do not success to enter in “OTA safe mode”.

No, power cycling will not work - the safe mode counter will be cleared each time you unpower it.

Adorem, 2019-10-18:
Thank you for your reply.

> Can you make sure it's the device you're pinging? If you've got two devices with the same IP on the network I would suspect something similar happening (each one disconnecting each other)

Yes ping stop if I power off the device and restart after I re-power on.

Thanks for the details how the OTA safe mode works. But then here for example the device was powered on all the night and it is not in safe mode. When I'm in the ESPHom Dashboard I see the green button telling the device is on:
[image]

(here the second device is working correctly)

When you tell this:
> b) Home Assistant doesn't connect to the device within 5 minutes for a few times in a row.

Does it means this sequence could work:
- switch off device and Home Assistant
- switch on device
- wait enough time (how many?) for device to enter in safe mode
- switch on Home Assistant
- try to upload new version for my device
?

Adorem, 2019-10-22:
Here are the logs from the serial port:
```
Exception (0):
epc1=0x40212008 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
          sp: 3fff4690 end: 3fff4900 offset: 01a0

>>>stack>>>
           3fff4830:  3fff2f48 3fff2abc 3fff2c48 40204743
3fff4840:  3fff2ab3 3fff2abc 3fff2c48 4020b099
3fff4850:  6c697241 4c207875 00333043 00000000
3fff4860:  2f000000 00000021 23000000 00535322
3fff4870:  303a3000 30303a30 00000020 402041f1
3fff4880:  00000031 3fff1304 4022e6b4 0000001b
3fff4890:  3fff5bc4 3fff2c48 3fff2a90 3fff2c48
3fff48a0:  3fff3048 3ffe8748 3fff2a90 4020fb7d
3fff48b0:  feefeffe feefeffe feefeffe feefeffe
3fff48c0:  feefeffe feefeffe feefeffe feefeffe
3fff48d0:  feefeffe feefeffe feefeffe 3fff38d0
3fff48e0:  3fffdad0 00000000 3fff38c9 4022e6f8
3fff48f0:  feefeffe feefeffe 3fff38e0 40100700
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

Exception (0):
epc1=0x40212008 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
          sp: 3fff4690 end: 3fff4900 offset: 01a0

>>>stack>>>
```

MelvinGr, 2020-02-20:
Did you ever find a solution to this? I have the exact same problem. However i can't access the serial as the device is a light bulb.

Adorem, 2020-03-21:
Sorry for my late reply and no I do not found any solution other than update through serial line.

iMike78, 2020-06-06:
Hi! 
It's not related to the topic, but the LED driver...
Is the RF remote working as it should if I flashing with ESPHome?
Cheers

Adorem, 2020-06-06:
Yes it is. I'm using this config:
```yaml
remote_receiver:
  pin:
    number: GPIO04
    inverted: True
  dump: nec
  on_nec:
    then:
      - lambda: |-
          if( 0xFF == x.address  )
          {
            switch( x.command )
            {
              case 0x906F: // dimmer +
                {
                  auto call = id(led_strip).turn_on();
                  float brightness;
                  id(led_strip).remote_values.as_brightness( &brightness );
                  float new_brightness = clamp(brightness + 0.2f, 0.0f, 1.0f);
                  ESP_LOGD( "dimmer +", "brightness: %f, new_brightness: %f", brightness, new_brightness );
                  call.set_state( new_brightness != 0.0f );
                  call.set_brightness( new_brightness );
                  call.set_transition_length(1000); // in ms
                  call.perform();
                  id(led_strip).current_values_as_brightness( &brightness );
                  ESP_LOGD( "dimmer +", "brightness: %f", brightness );
                }
                break;
              case 0xB847: // dimmer -
                {
                  auto call = id(led_strip).turn_on();
                  float brightness;
                  id(led_strip).remote_values.as_brightness( &brightness );
                  float new_brightness = clamp(brightness - 0.2f, 0.0f, 1.0f);
                  ESP_LOGD( "dimmer -", "brightness: %f, new_brightness: %f", brightness, new_brightness );
                  call.set_state( new_brightness != 0.0f );
                  call.set_brightness( new_brightness );
                  call.set_transition_length(1000); // in ms
                  call.perform();
                  id(led_strip).current_values_as_brightness( &brightness );
                  ESP_LOGD( "dimmer -", "brightness: %f", brightness );
                }
                break;
              case 0xF807: // power off
                {
                  auto call = id(led_strip).turn_off();
                  call.perform();
                }
                break;
              case 0xB04F: // power on
                {
                  auto call = id(led_strip).turn_on();
                  // set parameters (optional)
                  // call.set_transition_length(1000); // in ms
                  call.set_brightness(1.0); // 1.0 is full brightness
                  call.set_rgb(1.0, 1.0, 1.0); // color, 1.0 is fully lit
                  // call.set_effect("The Effect");
                  // perform action:
                  call.perform();
                }
                break;

              case 0x9867: // color2 1
                {
                  auto call = id(led_strip).turn_on();
                  call.set_rgb(1.0, 0.0, 0.0);
                  call.perform();
                }
                break;
[... 108 lines cut ...]
                {
                  auto call = id(led_strip).turn_on();
                  call.set_rgb(0.0, 1.0, 1.0);
                  call.perform();
                }
                break;
              case 0xF00F: // color2 FF00FF
                {
                  auto call = id(led_strip).turn_on();
                  call.set_rgb(1.0, 0.0, 1.0);
                  call.perform();
                }
                break;
              case 0x30CF: // Scheme 3
                {
                }
                break;

              default:
                break;
            }
          }
```

For the moment all is not working like I expect but yes the receiver works.

iMike78, 2020-06-07:
@Adorem Thank you for the quick answer, I'll try this out :)

iMike78, 2020-06-13:
@Adorem It doesn't work for me. Are you sure it's for the RF version? It seems to be an IR code (NEC)

Adorem, 2020-06-14:
Oops yes I'm using IR. Then I don't know for RF.

closed 2020-12-05 by stale[bot]