esphome/issues #744: Provide valid data range filter

borpin, 2019-10-18:
Follow up to Issue #741 (sorry).

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio, docker, ESPHome add-in 1.13.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

WEMOS D1 R3
**Affected component:**

Sensor / BME680

**Description of problem:**
I am getting occasional bad data from the I2C interface. I cannot filter it out **largely** because the filter_out filter works on the full precision of the value and the 'delta' filter posts data when it exceeds the data not the inverse.

Whilst the filter added by PR https://github.com/esphome/esphome/pull/765 allows for filtering of a rounded value, the bad data is *not consistent* enough (i.e. not always the same wrong value) for this to completely filter bad data out.  The data value is though, outside an expected range or the difference between it and the last good value is too great.

I suggest one of these filters would solve the problem;

* A DeltaMax filter - specify a maximum difference between the last good value and the new value (rate-of -change limit). For instance, you might only expect a temperature to change by a small amount, certainly not 10 degrees inside a house in 60s, so a limit of change of 10 would filter out a  bad value as it is usually an extreme value.  This also accounts for occasions where a value could be at either end of the range.
* A within-range filter - only post data if within a specified range. if measuring the temperature of the house, you can estimate a valid range. If outside the range discard. This is less useful I'd suggest.

**Problem-relevant YAML-configuration entries:**
please see #741
```yaml
sensor:
  - platform: bme680
    temperature:
      name: "BME680 Temperature"
      state_topic: emon/bme680_1/Temperature
      retain: false
      discovery: false
      oversampling: 16x
      accuracy_decimals: 1
      filters:
        - filter_out: 195.8
```

**Logs (if applicable):**

```
[21:14:33][D][bme680.sensor:314]: Got temperature=195.8°C pressure=2005.6hPa humidity=0.0% gas_resistance=nanΩ
```

**Additional information and things you've tried:**

borpin, 2019-10-19:
Here is an example - the bad data is obvious;

[image]

These spikes could be filtered out by a max change filter.

Here is the raw data - the changes in each case are significant.

```
"Unix timestamp", "bme680_1:Humidity", "bme680_1:Temperature", "bme680_1:Pressure"
1571481960, 97.7, 14.5, 980.0
1571482020, 88.2, 14.8, 979.9
1571482080, 95.9, 14.5, 980.0
```

```
"Unix timestamp", "bme680_1:Humidity", "bme680_1:Temperature", "bme680_1:Pressure"
1571497610, 85.1, 15.3, 981.7
1571497670, 84.9, 15.3, 981.7
1571497730, 84.8, 15.3, 981.6
1571497790, 98.9, -131.5, 761.5
1571497850, 84.9, 15.3, 981.7
1571497910, 84.9, 15.2, 981.7
```

OttoWinter, 2019-10-20:
Well yes, adding new filters would be an option, but we should first evaluate what method would work best for most use cases (I don't want a filter for each type of possible fault).

For that (and until then), I would suggest using a lambda filter - both filters you mentioned would be 2-3 lines in a lambda filter and can be used to evaluate.

borpin, 2019-10-20:
Ok thanks. Not sure how to implement them though.

OttoWinter, 2019-10-20:
"DeltaMax":

```yaml
sensor:
  - platform: bme680
    temperature:
      name: "BME680 Temperature"
 	  filters:
        - lambda: |-
			float MAX_DIFFERENCE = 1.0;  // adjust this!
			static last_value = NAN;
			if (isnan(last_value) || std::abs(x - last_value) < MAX_DIFFERENCE)
              return x;
            else
		      return {};
```

"within-range":

```yaml
sensor:
  - platform: bme680
    temperature:
      name: "BME680 Temperature"
 	  filters:
        - lambda: |-
			float MIN_VALUE = 20.0;
			float MAX_VALUE = 20.0;
			if (MIN_VALUE <= x && x <= MAX_VALUE) return x;
		    else return {};
```

borpin, 2019-10-20:
Brilliant. I'll have a go 😀

borpin, 2019-10-20:
Getting an error

```
src/main.cpp:338:14: error: 'last_value' does not name a type
       static last_value = NAN;
```

OttoWinter, 2019-10-20:
Whoops, updated the example

borpin, 2019-10-20:
Many thanks, one other question, are filters sequential so if I filter_out NAN first, does that stop the lambda filter?

borpin, 2019-10-20:
Sorry me again. Is it possible to send a custom debug message to MQTT when the lambda filters out some data?

borpin, 2019-10-21:
Hi @OttoWinter, thanks for your help.

My only question is, does each lambda need unique `static` variables?

I ended up with this (and for anyone finding this via Google :grin);

```yaml
        - lambda: |-
            float MAX_DIFFERENCE = 2.0;  // adjust this!
            static float last_value_t = NAN;
            static int count_missed_t = 0;
            if (count_missed_t == 3) last_value_t = x;
            if (isnan(last_value_t) || std::abs(x - last_value_t) < MAX_DIFFERENCE) {
              count_missed_t = 0;
              return last_value_t = x;
            } else {
              count_missed_t += 1;
              ESP_LOGW("main", "Missed Temperature Data %d", count_missed_t);
              return last_value_t;
            }
```

It returns the last value 3 times then reverts to current value.

Accounts for larger than expected jumps and also always returns a value.

Many thanks for your help - truly an amazing thing you have built!

Cheers, Brian

OttoWinter, 2019-10-21:
> My only question is, does each lambda need unique static variables?

I don't think so - the static variables visibility should be limited to the lambda body. If it compiles, that means it's ok.

borpin, 2019-10-21:
Many thanks.

closed 2019-10-21 by borpin